MichaIng/DietPi #4984: NanoPi K2 cannot login using debug serial port

asesidaa, 2021-11-22:
I use a USB to TTL converter to connect my K2 to computer and establish a serial connection. 
On uboot session, all works fine, I can get into uboot command line and type in commands. However, after boot into kernel, I can only get output, but cannot type anything to input and stuck at `press <return> to login`.
Moreover, this issue only happens when I am using the debug port, which corresponds to /dev/ttyAML0, if I connect to another UART port, and change console to /dev/ttyAML2 in `boot.cmd`, I can successfully login.
Can anyone help me debug this issue? I believe it is related to some config in the system.

MichaIng, 2021-11-22:
There are two places to configure a serial console:
- Enable boot/kernel message output via kernel command line argument, which is done in `boot.cmd`.
- Enable a login getty on a serial device. If the debug port is on `/dev/ttyAML0`, then try the following to get a login prompt:
    ```sh
    systemctl enable --now serial-getty@ttyAML0
    ```

There is this `console=display/serial/both` variable which can be set in `/boot/armbianEnv.txt`, but using `serial` or `both` toggles kernel/boot messages on a hardcoded serial device, in this case on `ttyAML0`. If for some reason you want to use `ttyAML2` without recompiling `boot.scr`, safest is to apply/add the desired serial device via `extraargs`, adding `console=ttyAML2`.

asesidaa, 2021-11-24:
Thank you for reply! 
It seems that for some reason, ttyAML0 was masked and symlinked to /dev/null, so I cannot input anything. I tried to manually unmask it and enable it, now it works just fine.

closed 2021-11-24 by asesidaa

MichaIng, 2021-11-24:
Ah okay great. Hmm masks should only be there when such serial console was disabled manually (e.g. via `dietpi-config` > Advanced Options > Serial/UART), via `CONFIG_SERIAL_CONSOLE_ENABLE=0` `dietpi.txt` setting or by choosing so on first login. But probably the image wasn't created cleanly.

Do you think `ttyAML0` is the better default serial console on NanoPi K2? If so, I'll repack the image accordingly.