MichaIng/DietPi #4987: Synapse Matrix Installation Failed

N7-BADA, 2021-11-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
`G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'`

- Distro Version | buster 1
- Kernel version | `Linux PiSeedbox 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`

#### Additional Information (if applicable)
- Software title | Synapse matrix
- Fresh install

#### Steps to reproduce
1. Install Synapse Matrix and wait.

#### Expected behaviour
- Successful install

#### Actual behaviour
- installation fais

#### Extra details
- only other software installed is qBittorrent

#### Details:
- Date           | Mon Nov 22 21:11:53 -01 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux PiSeedbox 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `pip3 install --no-cache-dir -U matrix-synapse psycopg2`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting matrix-synapse
  Downloading https://www.piwheels.org/simple/matrix-synapse/matrix_synapse-1.47.0-py3-none-any.whl (2.1 MB)
Collecting psycopg2
  Downloading https://www.piwheels.org/simple/psycopg2/psycopg2-2.9.2-cp37-cp37m-linux_armv7l.whl (505 kB)
Collecting unpaddedbase64>=1.1.0
  Downloading https://www.piwheels.org/simple/unpaddedbase64/unpaddedbase64-2.1.0-py3-none-any.whl (6.1 kB)
Collecting idna>=2.5
  Downloading https://www.piwheels.org/simple/idna/idna-3.3-py3-none-any.whl (64 kB)
Collecting cryptography>=3.4.7
  Downloading cryptography-36.0.0.tar.gz (571 kB)
[... 253 lines cut ...]
      pip: 21.3.1
      setuptools: 59.2.0
      setuptools_rust: 1.0.0
      =============================DEBUG ASSISTANCE=============================

  error: can't find Rust compiler

  If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.

  To update pip, run:

      pip install --upgrade pip

  and then retry package installation.

  If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.

  This package requires Rust >=1.41.0.
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

Joulinar, 2021-11-22:
Hi,

issue is with `cryptography` module. The version 36 your system is trying to download is now available as pre-build wheel for Buster on https://www.piwheels.org/project/cryptography/

You could try to install version 35.

```
sudo pip3 install cryptography==35.0.0
```

afterwords try to install Synapse again

N7-BADA, 2021-11-22:
> Hi,
> 
> issue is with `cryptography` module. The version 36 your system is trying to download, is now available as pre-build wheel for Buster on https://www.piwheels.org/project/cryptography/
> 
> You could try to install version 35.
> 
> ```
> sudo pip3 install cryptography==35.0.0
> ```
> 
> Hope this satisfy Synapse and you could try to install again.

Thank you for the fast reply! that didnthe trick indeed. it successfully installed now.

Joulinar, 2021-11-22:
ok perfect

@MichaIng 
not sure if we need to do anything on this. I guess `cryptography` version 35 will be last one on Buster (Python 3.7)

Joulinar, 2021-11-23:
ok going to close this topic as original issue has been fixed. Feel free to reopen if needed.

closed 2021-11-23 by Joulinar