MichaIng/DietPi #4989: DietPi on Bookworm has no ssh client preinstalled

StephanStS, 2021-11-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=-1
G_GITBRANCH='dev'
G_GITOWNER='MichaIng'

- Distro version | bookworm 0
- Kernel version | Linux Pizero2 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
- SBC model | RPi Zero 2 W (armv7l)

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated?
  - Bookworm was updated via PREP script

#### Steps to reproduce
1. Install Bullseye (I only installed the DietPi dashboard additionally)
2. Upgrade to Bookworm via PREP script
3. Try to run ssh to connect to other system: `ssh root@bratwurst`

#### Expected behaviour
System 'bratwurst' can be accessed via ssh (ssh login appears).

#### Actual behaviour
No ssh login appears.
Output: `-bash: ssh: Kommando nicht gefunden.`

#### Extra details
With installed OpenSSH client a connection works (`apt install openssh-client`).

Joulinar, 2021-11-24:
@StephanStS 
but should be same behaviour on Buster as `/usr/bin/ssh` is part of `openssh-client` package https://packages.debian.org/search?suite=bullseye&arch=any&mode=exactfilename&searchon=contents&keywords=ssh

StephanStS, 2021-11-24:
@Joulinar: Also tested it on Bullseye (aarch64) and can confirm your description.

Joulinar, 2021-11-24:
ok going to close the issue. Feel free to reopen if needed.

closed 2021-11-24 by Joulinar

MichaIng, 2021-11-24:
Since Dropbear is pre-installed on DietPi, the Dropbear client is:
```sh
dbclient root@thueringer
```

StephanStS, 2021-11-24:
> Since Dropbear is pre-installed on DietPi, the Dropbear client is:
> 
> ```shell
> dbclient root@thueringer
> ```

Oh, I missed that. On all other devices I activated OpenSSH. So I have to learn to use the dbclient.
An option could be to have a symlink from /usr/bin/ssh to /usr/bin/dbclient in case of dropbear to make life easier for those who search for ssh like me. :-)

MichaIng, 2021-11-24:
I thought about such a symlink, but we need to verify that:
1. OpenSSH does not fail to install if that symlink is present
2. OpenSSH does not overwrite the `dbclient` binary (symlink target) on install but replaces the actual symlink

Joulinar, 2021-11-24:
But is this really a case we need to implement? Until now we did not had such a report and most people's using it as server and not as a client or have been switched to the full OpenSSH package. Just thinking...

MichaIng, 2021-11-24:
Check out the high OpenSSH client install count, which is significantly higher than OpenSSH server installs: https://dietpi.com/survey/#software

I'm pretty sure that a notable amount of those is that users do not find `ssh` and then install the OpenSSH client while the Dropbear client would have worked just fine.

So it's nothing urgent, but it may prevent some unnecessary installs and makes the Dropbear client more popular, respectively makes users aware that it exists and is pre-installed by default. It cannot be installed standalone, so we cannot make it an own software option.

Instead of the symlink, we could also create the file as shell wrapper to print an info that the Dropbear client is currently used and how to call it directly. That would eliminate issue 2.

Joulinar, 2021-11-24:
Personally I'm using `openssh-client` package to be able to do SCP.

MichaIng, 2021-11-24:
There may be other reasons, but also alternatives, like `gesftpserver`. It's just that the `ssh` is used pretty often in guides and HowTos like it was always available. On DietPi it is not OOTB and when users use search machines about it they will find that the OpenSSH client package needs to be installed. We could prevent that lookup and unnecessary install by placing a symlink or wrapper. But it must be ruled out that it can cause any issues.