MichaIng/DietPi #4990: Chromium does not run on Kiosk mode.

hkayann, 2021-11-25:
### Creating a bug report/issue

#### Required Information
- DietPi version |  G_DIETPI_VERSION_CORE=7
                            G_DIETPI_VERSION_SUB=8
                            G_DIETPI_VERSION_RC=2

- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v8+
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Official Pi Adapter
- SDcard used | Kingston Canvas Select Plus 32GB

#### Additional Information
- Software title | Chromium
- Software installed freshly.
- Issue can be replicated.
- 
#### Steps to reproduce

1. `dietpi-sofware`
2. Install Chromium.
3. Run Chromium via `chromium-browser`.

#### Expected behaviour

- Chromium should run on Kiosk mode.

#### Actual behaviour

- Generating errors.

#### Extra details

- The output of  `chromium-browser`:
```
/usr/bin/chromium-browser: line 134: lsb_release: command not found
/usr/bin/chromium-browser: line 135: lsb_release: command not found
dpkg-query: no packages found matching bluealsa
Trace/breakpoint trap
```

Joulinar, 2021-11-25:
I guess it's you as well who opened the issue on our forum? Pls try to stick to one platform. It dosn't make sense to open an issue twice. This will just double our efforts to mange both. thx for understanding.

https://dietpi.com/phpbb/viewtopic.php?t=9614

hkayann, 2021-11-25:
Hi @Joulinar,

You are definitely right, sorry for the inconvenience. I guess, it is better if we can continue from here.

MichaIng, 2021-11-25:
You cannot run Chromium without an X server. Do this:
```sh
xinit "$(command -v chromium-browser)"
```

The `lsb_release` and `bluealsa` messages are btw not related, the first is irrelevant and the second is a test where the output is not muted as it should be.

Joulinar, 2021-11-25:
or use out autostart script

```
/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
```

hkayann, 2021-11-25:
@MichaIng I installed via running `dietpi-software`. I searched for a `chromium`, ticked it, and then installed.

After I run `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh`

It opens DietPi homepage, and I also get following errors:

```
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux DietPi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:59:0A:E3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=PARTUUID=8b0ac6f5-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=tty1
Build Date: 05 August 2021  11:35:55AM
xorg-server 2:1.20.11-1+rpt1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Nov 25 13:45:56 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
/usr/bin/chromium-browser: line 134: lsb_release: command not found
/usr/bin/chromium-browser: line 135: lsb_release: command not found
dpkg-query: no packages found matching bluealsa
libEGL warning: DRI2: failed to authenticate
[2394:2460:1125/134600.219310:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2394:2460:1125/134600.219459:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2450:2450:1125/134600.604349:ERROR:gpu_init.cc(441)] Passthrough is not supported, GL is egl
[2450:2450:1125/134600.867273:ERROR:viz_main_impl.cc(162)] Exiting GPU process due to errors during initialization
[2394:2460:1125/134601.306730:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2394:2460:1125/134601.307190:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2491:2491:1125/134601.736944:ERROR:gpu_init.cc(441)] Passthrough is not supported, GL is swiftshader
[2394:2516:1125/134601.902233:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[2394:2516:1125/134601.902915:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[2394:2516:1125/134601.904452:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
```
The only thing I require is accessing Node-RED server (something like, 192.1##.###.###:1880), I appreciate if you can guide me regarding this.

MichaIng, 2021-11-25:
Probably the following mutes the EGL warning:
```sh
echo ' --use-gl=egl' >> /etc/chromium.d/custom_flags
echo ' --use-gl=egl' >> /root/.chromium-browser.init
```
Assure to use a the KMS display driver, if not done yet.

Here is the line to change the website when using the autostart script: https://github.com/MichaIng/DietPi/blob/733d81e/dietpi.txt#L252

hkayann, 2021-11-25:
> Assure to use a the KMS display driver, if not done yet.

It seems like not.
```
root@DietPi:~# ls -l /dev/fb* /dev/dri
ls: cannot access '/dev/dri': No such file or directory
crw-rw---- 1 root video 29, 0 Nov 24 16:59 /dev/fb0
```
I had a look at [DietPi tools](https://dietpi.com/phpbb/viewtopic.php?t=5)  but could not locate an option to install such a driver.

Joulinar, 2021-11-25:
But Chromium is starting correct? You can use dietpi-autostart to select Chromium to start right after boot. You will be ask to specify starting page

hkayann, 2021-11-25:
Yes, It starts. I changed to homepage to `www.google.com`. I can use it as expected.

However, it lacks browser bar (the one on the top), is there any way to add such a bar so I can have more than 1 Tab open etc?

Joulinar, 2021-11-25:
Chromium will start in Kiosk Mode if configured as autostart and thinks like menus are disabled by purpose. Just have a look to the autostart script. There you can find options we have set.

hkayann, 2021-11-25:
Here is the bash. I can't see such a thing.
```
#!/bin/bash
# Autostart run script for Kiosk mode, based on @AYapejian https://github.com/MichaIng/DietPi/issues/1737#issue-318697621
# - Please see /root/.chromium-browser.init (and /etc/chromium.d/custom_flags) for additional egl/gl init options

# Command line switches https://peter.sh/experiments/chromium-command-line-switches/
# --test-type gets rid of some of the chromium warnings that you may or may not care about in kiosk on a LAN
# --pull-to-refresh=1
# --ash-host-window-bounds="400,300"

# Resolution to use for kiosk mode, should ideally match current system resolution
RES_X=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_RES_X=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)
RES_Y=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_RES_Y=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)

CHROMIUM_OPTS="--kiosk --test-type --window-size=$RES_X,$RES_Y --start-fullscreen --start-maximized --window-position=0,0"
# If you want tablet mode, uncomment the next line.
#CHROMIUM_OPTS+=' --force-tablet-mode --tablet-ui'

# Add URL for first run:
URL=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_AUTOSTART_URL=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)
CHROMIUM_OPTS+=" --homepage $URL"

# Find absolute filepath location of Chromium binary.
FP_CHROMIUM=$(command -v chromium)
if [[ ! $FP_CHROMIUM ]]; then

        # Assume RPi
        FP_CHROMIUM="$(command -v chromium-browser)"

fi

xinit $FP_CHROMIUM $CHROMIUM_OPTS
```

MichaIng, 2021-11-25:
You wrote "Kiosk mode" in the title, so I thought you know what it is? If you want to start Chromium the regular interactive way:
```sh
xinit "$(command -v chromium-browser)"
```
For better performance, I recommend to enable the KMS driver:
```sh
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d
reboot
```

hkayann, 2021-11-25:
Oh, I see. 

I have thought the Kiosk mode is the mode when we do not have a desktop installed. Now it is all clear.

Thanks for all the help @MichaIng  and @Joulinar.

MichaIng, 2021-11-25:
A desktop is not required at all for Chromium, regardless which mode, generally not for any X application, as they can run fine in standalone mode via `xinit`. The kiosk mode is basically a fullscreen mode like when you hit `F11` to hide all GUI elements but have the website in fullscreen, but still with mouse and keyboard inputs so that you can browse the website/links.

hkayann, 2021-11-25:
Yes. 

I am totally aware what it is now. I have no idea why I had such an impression.

MichaIng, 2021-12-11:
Okay I'll mark this as closed then. Feel free to reopen when required.

closed 2021-12-11 by MichaIng