MichaIng/DietPi #4991: Some proccess case to cpu overload

Jernel-dotcom, 2021-11-25:
#### Required Information
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

echo $G_DISTRO_NAME $G_RASPBIAN

uname -a

echo $G_HW_MODEL_NAME

- Power supply used 5V 1A RAVpower
- 
- SDcard used SanDisk ultra

#### Additional Information (if applicable)
- Software title Nextcloud
- Was the software title installed freshly or updated/migrated? yes
- Can this issue be replicated on a fresh installation of DietPi? yes, however the issue appear randomly after a few months of use

d761d528-9860-473e-8b59-11e3d10774e1

#### Steps to reproduce
1. Install dietpi 
2. Install nextcloud
3. Use it for a few weeks months 
4. Than the issue appear randomly 

#### Expected behaviour
The server should running normally 

#### Actual behaviour
The server experience high load on CPU, when it unclear what causing such behavior.
I already face this issue a few months ago, when eventually I fresh install it and the issue appear again 

#### Extra details
Restarting/rebooting didn't help.

DietPi v7.8.2

root@DietPi:/dev# cat /etc/*release*
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian

root@DietPi:/dev# uname -r
5.10.63-v8+

top - 15:46:03 up 31 min,  2 users,  load average: 13.02, 11.07, 6.64
Tasks: 122 total,   1 running, 121 sleeping,   0 stopped,   0 zombie
%Cpu(s):  4.3 us,  1.4 sy,  0.0 ni, 94.2 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   3844.9 total,   3316.7 free,    276.3 used,    252.0 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.   3470.2 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                      
  288 root      20   0  337540   1760   1540 S   6.7   0.0   1:48.67 pigpiod                                                                                                      
13211 root      20   0   10640   3084   2632 R   6.7   0.1   0:00.01 top                                                                                                          
    1 root      20   0  166088   9428   7096 S   0.0   0.2   0:11.64 systemd                                                                                                      
    2 root      20   0       0      0      0 S   0.0   0.0   0:00.01 kthreadd

MichaIng, 2021-11-25:
Many thanks for your reprot.

Use `htop` to monitor running processes and their CPU usage in real time.

From the `top` it looks like `pigpiod` is the culprit, but I don't know this process, what it does, any it isn't installed by any of the `dietpi-software` options, hence was installed manually by you? Looks like the package and service have the same name. To to test whether it helps, and of course if you do not actively use it for whatever GPIO usage:
```sh
systemctl disable --now pigpiod
# or even
apt purge pigpio
```

Joulinar, 2021-11-25:
probably this one https://packages.debian.org/bullseye/libpigpiod-if-dev

MichaIng, 2021-11-25:
Nope, this ist just the development headers and library, but not the daemon. The daemon package does not exist on Debian, but only on the Raspberry Pi repository: https://archive.raspberrypi.org/debian/pool/main/p/pigpio/

Jernel-dotcom, 2021-12-15:
Hi,
I can turn off this processes however my fan on raspberry hat will turn off completely.

Jernel-dotcom, 2021-12-15:
I installed the deb http://ftp.de.debian.org/debian bullseye main to include the right packages.
Didn't help.

Nextcloud page is still unavailable and CPU is still peaking up to 4 (of 4 cpus)

MichaIng, 2021-12-15:
But it is the one responsible for high CPU load? As said, I do not know this package or what it does, obviously it controls your GPIO attached fan. However, it comes from the RPi repository, so please report this at: https://github.com/RPi-Distro/repo/issues

> I installed the deb http://ftp.de.debian.org/debian bullseye main to include the right packages.

This was installed already, no need to add it manually (else you'll see a warning about doubled packages when doing `apt update`). And as said, the Debian package seems to be only part of what you need and doesn't ship that service which seems to be required for your fan to work, so the Debian package is wrong the and one you had installed already was right. It just causes a high CPU load for unknown reasons, but probably the RPi devs can help or replicate.

MichaIng, 2022-01-04:
Were you able to identify `pigpio` as the CPU load driver?

MichaIng, 2022-02-06:
Marking as closed due to outstanding reply. Feel free to reopen when required.

closed 2022-02-06 by MichaIng