MichaIng/DietPi #4992: DietPi-Software | RealVNC not starting due to missing password

Joulinar, 2021-11-25:
Based on https://github.com/MichaIng/DietPi/pull/4679#issuecomment-908276307 we already fixed the missing password issue on RealVNC with v7.6

However it seems issue is back again and setting the password during installation did not work anymore. 

```
Nov 25 13:53:24 DietPi3 vncserver[5915]: [  OK  ] RealVNC detected
Nov 25 13:53:24 DietPi3 vncserver[5915]: [ INFO ] Starting virtual desktop at display :1 in 1280x720x16
Nov 25 13:53:24 DietPi3 vncserver[5915]: Error: Setting password failed [NoVncPasswd]
```

I tried as well to apply the password multiple times during install but without success.

```
[  OK  ] DietPi-Software | chmod +x /usr/local/bin/vncserver
Setting "Password" VNC parameter for Virtual Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/Xvnc
Successfully set "Password" VNC parameter in
/root/.vnc/config.d/vncserver-x11-virtual
Setting "Password" VNC parameter for Virtual Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/Xvnc
Successfully set "Password" VNC parameter in
/root/.vnc/config.d/vncserver-x11-virtual
Setting "Password" VNC parameter for Virtual Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/Xvnc
Successfully set "Password" VNC parameter in
/root/.vnc/config.d/vncserver-x11-virtual
Setting "Password" VNC parameter for Virtual Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/Xvnc
Successfully set "Password" VNC parameter in
/root/.vnc/config.d/vncserver-x11-virtual
Setting "Password" VNC parameter for Service Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/vncserver-x11

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Finalising install
```

the `Xvnc` file stays empty

```
root@DietPi3:~# ls -l /root/.vnc/config.d
total 8
-rw-r--r-- 1 root root  0 Nov 25 13:57 Xvnc
-rw-r--r-- 1 root root 26 Nov 25 13:57 vncserver-x11
-rw-r--r-- 1 root root 26 Nov 25 13:57 vncserver-x11-virtual
root@DietPi3:~#
```

Workaround is to apply the password manually. 🤔 

```
root@DietPi3:~# vncpasswd -virtual <<< "dietpi
dietpi"
Setting "Password" VNC parameter for Virtual Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/Xvnc
Successfully set "Password" VNC parameter in
/root/.vnc/config.d/vncserver-x11-virtual
root@DietPi3:~# ls -l /root/.vnc/config.d
total 12
-rw-r--r-- 1 root root 26 Nov 25 14:04 Xvnc
-rw-r--r-- 1 root root 26 Nov 25 13:57 vncserver-x11
-rw-r--r-- 1 root root 26 Nov 25 14:04 vncserver-x11-virtual
root@DietPi3:~#
```

Now service is starting fine.

MichaIng, 2021-11-25:
Nasty that this is still an issue. We do already call this command two times in `dietpi-software`, then check whether the file exists and if not copy from the other one (same content anyway) which does exist. I have no idea why it is not always (randomly) created as it should, even that the command output says it was, or otherwise why it is removed again.

I think we need to further debug this with the help of some debugging code to check where it is create or not or removed again. Mysterious.

MichaIng, 2021-12-11:
Hopefully finally fixed with: https://github.com/MichaIng/DietPi/commit/d0ace4f
Changelog: https://github.com/MichaIng/DietPi/commit/93dbb0f

closed 2021-12-11 by MichaIng