MichaIng/DietPi #4993: Add remote logging as an automated setup option

ckuethe, 2021-11-25:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
I have a couple dozen machines running dietpi and one of the first things I do is install rsyslog set to send all my logs to my network syslog receiver. That gets old fast, though I've got a "ckuethe-ize" script to install my usual set of packages, ssh keys, hosts files, other configs ...

#### Describe the solution you'd like:
I'd like to have a network log option that installs rsyslog, and then sends all logs to a specified syslog receiver rather than disk like full logging does.

#### Describe alternatives you've considered:
I did RTFM (making sure to read https://dietpi.com/docs/software/log_system/) to make sure this wasn't already implemented. My current alternative is copying in my local rsyslog.conf that does this.

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2021-11-25:
Many thanks for your request.

As I plan to remove rsyslog and logrotate completely (from `dietpi-software`, of course it can always be installed via `apt`), replacing it with `mkdir /var/log/journal` for persistent system logs right from the journal (instead of the round trip through rsyslog into a plain text file), maybe `systemd-journald` can be configured to send logs to a remote receiver directly. I'll have a look.

ckuethe, 2021-11-25:
I'm not attached to rsyslog; it would be entirely satisfactory if `systemd-journald` could be told to just ship all the logs to `syslog.example.com` - especially if that could be specified in `dietpi.txt` at install time.

ckuethe, 2022-04-07:
**Barf**!

According to `journald.conf(5)`, `systemd-journald` can't directly forward to a syslog server and it `systemd-journald-remote` isn't as portable as regular syslog (and it's also not installed by default). 

https://www.digitalocean.com/community/tutorials/how-to-centralize-logs-with-journald-on-debian-10

Just filed a bug on systemd... https://github.com/systemd/systemd/issues/23013

MichaIng, 2022-04-08:
`systemd-journal-remote` seems to be exactly what you are looking for:
- https://packages.debian.org/bullseye/systemd-journal-remote
- https://manpages.debian.org/bullseye/systemd-journal-remote/systemd-journal-remote.8.en.html

What do you mean by "isn't as portable as regular syslog"?

Your issue at systemd sounds like you don't know about `systemd-journald-remote`, so I suggest you adjust it to mention the exact (portability) issue you face with it, instead of requesting to have remote syslog support integrated into basic systemd-journald (which I would be against, systemd is sufficiently bloated already, keep it modular where it is already).

MichaIng, 2023-12-19:
Hmm, netlogd seems to be not (natively) available on Debian at all. Interestingly I do not find any info about what the difference of both is. Do you have a reference about "systemd (no capital "s" and "d" 😉) maintainer team recommended"?

MichaIng, 2023-12-20:
Ah, so this is the difference, thanks for clarification. In case journal format is sufficient, at least currently it is easier to install `systemd-journal-remote` just via APT, without compiling. But it makes sense to have a standard syslog distribution as well.

Btw, generally you can have the same with `rsyslog`, isn't it? Sure by default it adds the local plain text files, which might not be wanted, but AFAIK this can be disabled so that it serves just as "r"(=remote)syslog daemon it is named after. But I guess `systemd-netlogd` is simpler to setup (after having it compiled already), being more tightly developed for this purpose.

AJolly, 2024-05-03:
So whats the current best way to have a dietpi system remotely syslog (and dont store logs locally on the sd card?)

MichaIng, 2024-05-11:
By default, logs are not stored locally on the SD card on DietPi, but in tmpfs/RAM log.

For remote logging, depending on what you need (which remote logging format you require), all options are listed above:
- `systemd-journal-remote` for systemd journald format
- `systemd-netlogd` for classic syslog format, but it still requires compiling from source
- `rsyslog` as well for classic syslog format, but it requires configuration to disable local logging and enable remote logging.

However, I did never use any of these (for remote logging), so test them, and feel free to share your experience.

AJolly, 2024-05-12:
Was able to set up rsyslog without too much hassle, actually now have the diet pie being my local log collector. I'm having it store logs in ram, for me I usually don't need any real log retention, more of useful for troubleshooting when something else on the network isn't behaving.

About the only downside is it seems a bit slow to flush the log files to "disk".