MichaIng/DietPi #4994: Kickstart script fails when trying to remove non-existent file.

Mausy5043, 2021-11-26:
#### Details:
- Date           | Fri Nov 26 19:16:18 CET 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=0)
- Kernel version | `Linux rbian 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `rm -R /etc/systemd/system/multi-user.target.wants/sshswitch.service`
- Exit code      | 1
- Software title | DietPi-PREP
#### Steps to reproduce:

1. Start from an image of Foodeas/raspberrypi-ua-netinst
2. Run kickstart script
#### Expected behaviour:

- Error-free transition to DietPi system
- IMHO, trying to remove non-existent files should not lead to errors

#### Actual behaviour:

- See error above.
- `rm` command fails due to file/directory not existing.

#### Extra details:

- Raspberry Pi 3 Model B Rev 1.2

#### Additional logs:
```
rm: cannot remove '/etc/systemd/system/multi-user.target.wants/sshswitch.service': No such file or directory
```

Joulinar, 2021-11-26:
Question: why are you not using our ready to use image for Raspberry Pi?

Mausy5043, 2021-11-26:
> Question: why are you not using our ready to use image for Raspberry Pi?

First of all, I really hate having to (repeatedly) unplug the Pi, take out the SD-card, reformat the SD-card, download an image to my computer, "burn" the image to the SD-card, unmount the SD-card, put the SD-card back in the Pi and power-on the Pi. 

I've been using `raspbian-ua-netinst` and `raspberrypi-ua-netinst` repositories for a long time. I'm used to make an (re-)install happen on a live system, automatically building the installer then rebooting and waiting for a mail telling me that the system is up and running. All with minimal manual intervention. So, I thought I'd give DietPi a go and tried to use the [PREP script](https://raw.githubusercontent.com/MichaIng/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh), as I'm kind of used to, on the live system. I only needed to `sed` the script to prevent uninstalling of  the `\*e2scrub\*` packages as they are not installed on an f2fs-based system. I would have done the same with the  offending`rm` command but that is not inside the script. So, that's where it ended.

Nevertheless, I think that `rm` failing to remove a file should not lead to aborting an install.

> To skip this step you could use the `change command` option and replace the command with `true`

I have no idea what you mean. Can you refer me to some documentation/manual explaining this?

Joulinar, 2021-11-26:
I thought you are running the script interactively. Usually there should be an error prompt where you could select different options.

MichaIng, 2021-11-26:
Many thanks for your report.

Issue is already solved on the `dev` branch: https://github.com/MichaIng/DietPi/commit/fcf399c
You can use the PREP script from the `dev` branch and still select `master` as target branch for the DietPi system: https://github.com/MichaIng/DietPi/blob/dev/PREP_SYSTEM_FOR_DIETPI.sh

Mausy5043, 2021-11-26:
Thanks for your quick response. I'll give it a try. Feel free to close this issue. I assume I can reopen it if I have anything to add.

MichaIng, 2021-11-26:
Interesting that the image does not ship with the `e2fsprogs` package. This is one that is marked as "required" and hence would never be autoremoved and when trying to remove it, a prompt appears where you must enter a confirmation text that you know what you are doing. Even on an F2FS image, tools and libraries to manage ext4 filesystems are highly expected by a lot of other packages and, scripts and installers (not ours). Not sure about the source of these net installers, but probably a reasonable suggestion to not actively remove this package from the even minimal debootstrap package list.

Makes sense that we add the package to the list in PREP then, if images exist that do not ship it 🙂.

Mausy5043, 2021-11-26:
Just checked again. None of the `e2fs` nor any of the `btrfs` packages is installed.

MichaIng, 2021-11-27:
Btrfs packages are not a problem at all, they are also not preinstalled on any other of our images. Only when PREP detects that any partition on the root drive contains a Btrfs filesystem, those are installed.

But actually I changed my mind. We do not ship F2FS images yet, and when we do, they will be in beta for a while. So actually this is a good change to test the importance of the `e2fsprogs` under real condition: I'll change it so that PREP purges `e2fsprogs` is no ext filesystem is present, and adjust `dietpi-drive_manager` to assure that it will be installed once an ext filesystem is detected. If this causes issues in too many cases, we can change it again. Means, would be great if you could report when facing any issue due to missing ext filesystem utils. I'm not thinking so much about `mkfs.ext4`, `fsck.ext4`, `tune2fs`, which are explicitly called on ext filesystems only, but e.g. `lsattr`/`chattr` where it is not so obvious and which may be called on other filesystems as well, though without valid output.

Mausy5043, 2021-11-27:
> We do not ship F2FS images yet

Remarkable.
Ext4 used as root the filesystem is inefficient, since ext4's journaling wears down the SD card, and ext4 wasn't developed for flash usage. I have very good experience with the use of F2FS on SD cards. The cards have been very stable for years now.

Mausy5043, 2021-11-27:
> I thought you are running the script interactively. Usually there should be an error prompt where you could select different options.

@Joulinar : No, I was feeding the script all the options it could need and indeed trying to run it interactively.

Mausy5043, 2021-11-27:
Come to think of it, I *could* start things off by initially running a DietPi image, but I'd want to end up at the point where a simple script would revert the entire machine back to basics and reinstall everything from fresh.   
I would then not need to have physical access to the Pi. Just log in over SSH start the script and check back when later to find the server up and running again.

MichaIng, 2021-11-27:
> Remarkable.

Not so much. There are reasons why Debian, Raspberry Pi OS, Armbian and Ubuntu ship with an ext4 root filesystem, and even Arch Linux ARM clearly instructs to format it as ext4, without even mentioning F2FS or any other alternative 😉. The difference in performance and disk I/O is there, but it is not significant. But the reduced chance of unrecoverable filesystem corruption is significantly lower on ext4 due to journaling and inode checksum (for some reason it is actively disabled on Raspberry Pi OS, not sure why). Also there are practical filesystem features, like that you are able to not only increase a filesystem's size but can also shrink it, interesting for creating images, also the reserved blocks to prevent an unprivileged process from filling the disk completely, and others.

F2FS is definitely the right choice for SD cards and also SSDs in most cases, used as additional data drives or for moving data from one system to another, digital cameras, mobile phone extensions etc, but not unquestionable as root filesystem.

However, it is not pretty simple for us to crease F2FS images for Raspberry Pi's. so I'll do that and then we'll see how it turns out in the fields 🙂.

closed 2021-12-11 by MichaIng