MichaIng/DietPi #4995: RPI Monitor - Custom Addon Titles cannot be hidden

hackslikeus, 2021-11-26:
### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
buster 1

- Kernel version | `uname -a`
Linux DietPiQAudio1 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 3 Model A+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2.54 A

- SDcard used | (EG: SanDisk ultra)
Microcenter class 10 - 16gb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
RPI Monitor

- Was the software title installed freshly or updated/migrated?
Installed November 15 (Bullseye), updated, bug noticed, Uninstalled 11/25, Reinstalled 11/26
apt-update etc. with no impact on issue.

- Can this issue be replicated on a fresh installation of DietPi?
Probably
 N/A

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...Install Software #66 - RPIMonitor
2. ...Using either sample addons provided, or create a custom addon, configure in .conf file,  attempt to hide the custom
addon page titleby using "web.addons.X.showTitle=0" or web.addons.showTitle=false" in configuration settings.
restart RPIMonitor Service, and using the menu option, select the adjusted addon from the menu. The page
still includes the Addon in the upper left as a page tite, even though it should be hidden.

#### Expected behaviour

- ... once addon configuration is set to either 0 (buster) or "false" as stated in example, page title
should no longer display custom addon name in upper left corner.

#### Actual behaviour

- ...Custom Addon Title appears in upper left corner (always) regardles of web.addons.X.showTitle = 0 or false or "false"

Joulinar, 2021-11-26:
similar to the other issue you reported, might be a challenge of RPi-Monitor themselves.

MichaIng, 2021-11-27:
Yes, please report this upstream, like you did with another issue already: https://github.com/XavierBerger/RPi-Monitor/issues/388
We simply install the latest (soon 4 years old) v2.13 beta6 Debian package from here: https://github.com/XavierBerger/RPi-Monitor-deb

Sadly RPi-Monitor is pretty unmaintained, so I think there won't happen much until someone with JavaScript and perl experience opens an actual PR to fix it.

hackslikeus, 2021-11-27:
seems odd that configuration options in dietpi are organized differently:  For example, the DietPi data.conf file only lists the individual includes to configuration files found in the templates folder, which is much cleaner compared to the same file
using the installation instructions found on the RPI Monitor readme file. 
The data.conf file from the official repository includes a hodge podge of text instructions, alert diagram, and the configuration options for Friends, Header, and most importantly - Addons, not to mention the includes listing. 
I was hoping I could pickup something that was missed during the cleanup, as some of the examples files show settings of "false" rather than 0 for false.  But after exhausting all local remedies, including uninstall from Dietpi software uninstall, and re-install using the github instructions, the only different outcome was that the ever present page title changed from the Addon name to "undefined", neither of which is the desired outcome.
 Will port this issue to the RPI Monitor site as a bug.

closed 2021-11-27 by hackslikeus

MichaIng, 2021-11-27:
Interesting, I wasn't aware that there is an official repository: https://giteduberger.fr/rpimonitor/
So you say that the outcome of installing the package from that repository is different than what `dietpi-software` ships? That is strange since the above link points as alternative to the exact same package source where we pull the same "latest" package from, so I'd have expected that it is indeed the exact same package.

hackslikeus, 2021-11-27:
to clarify my earlier statement re. repository - earlier today I pulled from:  [here ]( https://xavierberger.github.io/RPi-Monitor-docs/11_installation.html)and followed installation instructions, and indeed the data.conf file includes alot of text before the config begins.  Just now, I installed from Dietpi software, and the file is concise
[Raw_Dietpi_data_conf.txt](https://github.com/MichaIng/DietPi/files/7612433/Raw_Dietpi_data_conf.txt)
.
[X_working_data_conf.txt](https://github.com/MichaIng/DietPi/files/7612448/X_working_data_conf.txt)

MichaIng, 2021-11-27:
Can you verify that it is v2.12 when following the instruction (installing from official repository)?

hackslikeus, 2021-11-27:
Last lines of the set log shows it was set up as rpimonitor (2.12.-r0) and then shows processing triggers for libc-bin (2.31 -13+rpt2+rpi1) before completed.

MichaIng, 2021-11-27:
Okay, then the cleaned up configuration file is part of v2.13. Although I have to say that I generally like config files which contain all available settings and comments to explain them, so one does not need to check documentations elsewhere 😄. But it's fine, good when you like it more the slim way.

hackslikeus, 2021-11-28:
Looks like dietpi dashboard will soon release and thus render rpi monitor as obsolete.  I very much enjoy the add-ons component as I hacked the monitor to become a whole home audio command center using squeezelite and my homegrown internet radio app, using the rpi monitor's friends component. Looking forward to using dietpi dashboard, and thanks for your help.