MichaIng/DietPi #4996: Menu Navigation on RPI Monitor only from Status Page?

hackslikeus, 2021-11-26:
### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
buster 1

- Kernel version | `uname -a`
Linux DietPiQAudio1 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 3 Model A+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2.54 A

- SDcard used | (EG: SanDisk ultra)
Microcenter class 10 - 16gb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
RPI Monitor

- Was the software title installed freshly or updated/migrated?
Installed November 15 (Bullseye), updated, bug noticed, Uninstalled 11/25, Reinstalled 11/26
apt-update etc. with no impact on issue.

- Can this issue be replicated on a fresh installation of DietPi?
Probably
 N/A
#### Steps to reproduce

1. ... Install  software #66 - RPI Monitor
2. ... Either Create more than one Custom Addons or Activate Friends Menu option by configuring at least 2 Friends
3. ... Once one option is selected from the drop down menu, the menu bar does not drop down again.

#### Expected behaviour

- ...If more than one choice is available in the drop down selection, after choosing one option, the menu options should be available.

#### Actual behaviour

- ...hovering over the menu title highlights the choice, but drop down does not occur on click, user has to select menu option of status as the only option available.  Once at the status page, the selections are available again from dropdown, so in short, all navigation has to be performed from the home base of status page.

#### Extra details

- ...

hackslikeus, 2021-11-26:
when more than one option is available, the first option seems to be OK, but selecting anything other than the first option freezes the menu bars drop down list.

Joulinar, 2021-11-26:
hmm this sounds more like an issue of the application themselves. From our side we install the Debian package and create one template fur USB HDD overview only.

Not sure if it make sense to report to RPi-Monitor developer as RPi-Monitor has not been updated for 4 years now. https://github.com/XavierBerger/RPi-Monitor

hackslikeus, 2021-11-27:
Closing this issue on this board as I was able to replicate issue on Raspbian Os - therefore it has been moved to the RPI monitor hub since it seems to be platform independent.

closed 2021-11-27 by hackslikeus