MichaIng/DietPi #4997: SABnzbd reinstallation issue

bensp, 2021-11-27:
#### Details:
- Date           | Fri Nov 26 18:12:17 GMT 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | pip3 install -Ur requirements.txt
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. SABnzbd was out of date, so I ran `dietpi-software reinstall 139`
2. Reinstallation failed due to the error below: it appears to be due to crytography?
3. I restored to a five-month old backup and it worked - great!
4. I tried to reinstall again, thinking perhaps I had changed something: it failed however it created a backup point that overwrote my working backup copy.. :( 
5. One of the recommended routes was upgrading pip.  It doesn't seem to want me to do this.
6. Now, SABnzbd won't start as it's not installed
7. I've researched here, online, to no joy.  Can you help please?

Error:
` copying src/cryptography/hazmat/backends/openssl/x509.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/x448.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/x25519.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/utils.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/rsa.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/poly1305.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/hmac.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/hashes.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/encode_asn1.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ed448.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ed25519.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ec.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/dsa.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/dh.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/decode_asn1.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/cmac.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/ciphers.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/backend.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/aead.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  copying src/cryptography/hazmat/backends/openssl/__init__.py -> build/lib.linux-armv7l-3.7/cryptography/hazmat/backends/openssl
  running egg_info
  writing src/cryptography.egg-info/PKG-INFO
  writing dependency_links to src/cryptography.egg-info/dependency_links.txt
  writing requirements to src/cryptography.egg-info/requires.txt
  writing top-level names to src/cryptography.egg-info/top_level.txt
  reading manifest file 'src/cryptography.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  no previously-included directories found matching 'docs/_build'
  warning: no previously-included files found matching 'vectors'
  warning: no previously-included files matching '*' found under directory 'vectors'
  warning: no previously-included files matching '*' found under directory '.github'
  warning: no previously-included files found matching 'release.py'
  warning: no previously-included files found matching '.coveragerc'
  warning: no previously-included files found matching 'codecov.yml'
  warning: no previously-included files found matching '.readthedocs.yml'
  warning: no previously-included files found matching 'dev-requirements.txt'
  warning: no previously-included files found matching 'tox.ini'
  warning: no previously-included files found matching 'mypy.ini'
  warning: no previously-included files matching '*' found under directory '.zuul.d'
  warning: no previously-included files matching '*' found under directory '.zuul.playbooks'
  writing manifest file 'src/cryptography.egg-info/SOURCES.txt'
  copying src/cryptography/py.typed -> build/lib.linux-armv7l-3.7/cryptography
  creating build/lib.linux-armv7l-3.7/cryptography/hazmat/bindings/_rust
  copying src/cryptography/hazmat/bindings/_rust/__init__.pyi -> build/lib.linux-armv7l-3.7/cryptography/hazmat/bindings/_rust
  copying src/cryptography/hazmat/bindings/_rust/asn1.pyi -> build/lib.linux-armv7l-3.7/cryptography/hazmat/bindings/_rust
  copying src/cryptography/hazmat/bindings/_rust/ocsp.pyi -> build/lib.linux-armv7l-3.7/cryptography/hazmat/bindings/_rust
  copying src/cryptography/hazmat/bindings/_rust/x509.pyi -> build/lib.linux-armv7l-3.7/cryptography/hazmat/bindings/_rust
  running build_ext
[... 259 lines cut ...]
      Python: 3.7.3
      platform: Linux-5.10.63-v7l+-armv7l-with-debian-10.11
      pip: 21.3.1
      setuptools: 50.3.0
      setuptools_rust: 1.0.0
      =============================DEBUG ASSISTANCE=============================
  
  error: can't find Rust compiler
  
  If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
  
  To update pip, run:
  
      pip install --upgrade pip
  
  and then retry package installation.
  
  If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
  
  This package requires Rust >=1.41.0.
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
```

Joulinar, 2021-11-27:
Hi,

issue is with `cryptography` module. The version 36 your system is trying to download, is now available as pre-build wheel for Buster on https://www.piwheels.org/project/cryptography/

You could try to install version 35.

```
sudo pip3 install cryptography==35.0.0
```

Hope this satisfy Synapse and you could try to install again.

bensp, 2021-11-27:
Thanks for your prompt response.

It didn't work, sadly.

Here is the log.

```
root@DietPi:~# sudo pip3 install cryptography==35.0.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting cryptography==35.0.0
  Downloading https://www.piwheels.org/simple/cryptography/cryptography-35.0.0-cp37-cp37m-linux_armv7l.whl (2.1 MB)
     |████████████████████████████████| 2.1 MB 2.9 MB/s            
Requirement already satisfied: cffi>=1.12 in /usr/local/lib/python3.7/dist-packages (from cryptography==35.0.0) (1.14.3)
Requirement already satisfied: pycparser in /usr/local/lib/python3.7/dist-packages (from cffi>=1.12->cryptography==35.0.0) (2.20)
Installing collected packages: cryptography
  Attempting uninstall: cryptography
    Found existing installation: cryptography 3.4.7
    Uninstalling cryptography-3.4.7:
      Successfully uninstalled cryptography-3.4.7
Successfully installed cryptography-35.0.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
root@DietPi:~# dietpi-software reinstall 139
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated reinstall

[  OK  ] DietPi-Software | Reinstalling SABnzbd: nzb download manager
[ SUB1 ] DietPi-Backup > Backup
[  OK  ] DietPi-Backup | mkdir -p /mnt/dietpi-backup/data
[ SUB2 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : sabnzbd
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : avahi-daemon
[ INFO ] DietPi-Backup | Checking for required APT packages: rsync
[  OK  ] DietPi-Backup | All required APT packages are already installed.
[  OK  ] DietPi-Backup | Free space check: path=/mnt/dietpi-backup/data | available=52626 MiB | required=528 MiB
[ INFO ] DietPi-Backup | Backup to /mnt/dietpi-backup in progress, please wait...
sending incremental file list
deleting boot/dietpi-CHANGELOG.txt
deleting boot/dietpi/boot
deleting boot/dietpi/func/obtain_network_details
      1,469,456   2%   29.19MB/s    0:00:00 (xfr#52, ir-chk=1003/1763)
deleting etc/apt/apt.conf.d/99-dietpi-force-ipv4
    697,199,074  82%    5.05MB/s    0:02:11 (xfr#65, ir-chk=1004/2874)
deleting etc/systemd/system/dietpi-boot.service
deleting etc/systemd/system/multi-user.target.wants/dietpi-boot.service
    705,037,929  52%    4.44MB/s    0:02:31 (xfr#449, ir-chk=1001/15202)
deleting root/.kodi/addons/service.xbmc.versioncheck/changelog.txt
deleting root/.kodi/temp/archive_cache/10025-e6f083c5.fi
deleting root/.kodi/temp/archive_cache/10025-1140787e.fi
    708,901,805  28%    4.41MB/s    0:02:33 (xfr#573, ir-chk=1035/26023)
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/top_level.txt
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/WHEEL
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/REQUESTED
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/RECORD
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/METADATA
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/LICENSE.PSF
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/LICENSE.BSD
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/LICENSE.APACHE
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/LICENSE
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/INSTALLER
deleting usr/local/lib/python3.7/dist-packages/cryptography-3.4.7.dist-info/
    708,902,468  28%    4.40MB/s    0:02:33 (xfr#576, ir-chk=1002/27025)
deleting usr/local/lib/python3.7/dist-packages/cryptography/hazmat/_types.py
deleting usr/local/lib/python3.7/dist-packages/cryptography/hazmat/_der.py
[... 320 lines cut ...]
      platform: Linux-5.10.63-v7l+-armv7l-with-debian-10.11
      pip: 21.3.1
      setuptools: 50.3.0
      setuptools_rust: 1.0.0
      =============================DEBUG ASSISTANCE=============================
  
  error: can't find Rust compiler
  
  If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
  
  To update pip, run:
  
      pip install --upgrade pip
  
  and then retry package installation.
  
  If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
  
  This package requires Rust >=1.41.0.
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
[FAILED] DietPi-Software | pip3 install --no-cache-dir -Ur requirements.txt
```

MichaIng, 2021-11-27:
It keeps failing because of the pip `-U` flag so that it keeps trying to update the module.

I btw reported the failing build to piwheels now: https://github.com/piwheels/packages/issues/264
Hopefully this buys us some time. But earlier or later we need to implement Rust anyway. They btw decided to use Debians `cargo` instead of `rustup` exactly because of the issue that is cannot be natively installed as system package/compiler but only on user level: https://github.com/piwheels/packages/issues/188#issuecomment-778829130

If the wheel is not fixed until next DietPi release, I'll follow this and implement the either `rustc ` or `cargo` package as dependency to all software options which pull the Python `cryptography` module for Stretch and Buster ARMv6 and ARMv7 systems.

@bensp 
You can do the same. Please give this a try first:
```sh
apt install rustc
dietpi-software reinstall 139
```
If it still fails:
```sh
apt install cargo
dietpi-software reinstall 139
```

bensp, 2021-11-27:
No joy I'm afraid.

Error log after installing Rust:

```
root@DietPi:~# apt install rustc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libllvm7 libstd-rust-1.41 libstd-rust-dev
Suggested packages:
  rust-doc rust-src lld-7
Recommended packages:
  cargo rust-gdb | rust-lldb
The following NEW packages will be installed:
  libllvm7 libstd-rust-1.41 libstd-rust-dev rustc
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 49.1 MB of archives.
After this operation, 176 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive-bm.raspbian.org/raspbian buster/main armhf libllvm7 armhf 1:7.0.1-8+rpi3+deb10u2 [11.4 MB]
Get:2 http://archive-bm.raspbian.org/raspbian buster/main armhf libstd-rust-1.41 armhf 1.41.1+dfsg1-1~deb10u1+rpi1 [12.0 MB]
Get:3 http://archive-bm.raspbian.org/raspbian buster/main armhf libstd-rust-dev armhf 1.41.1+dfsg1-1~deb10u1+rpi1 [24.0 MB]
Get:4 http://archive-bm.raspbian.org/raspbian buster/main armhf rustc armhf 1.41.1+dfsg1-1~deb10u1+rpi1 [1703 kB]                                                                                                                                                       
Fetched 49.1 MB in 12s (4255 kB/s)                                                                                                                                                                                                                                      
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package libllvm7:armhf.
(Reading database ... 37181 files and directories currently installed.)
Preparing to unpack .../libllvm7_1%3a7.0.1-8+rpi3+deb10u2_armhf.deb ...
Unpacking libllvm7:armhf (1:7.0.1-8+rpi3+deb10u2) ...
Selecting previously unselected package libstd-rust-1.41:armhf.
Preparing to unpack .../libstd-rust-1.41_1.41.1+dfsg1-1~deb10u1+rpi1_armhf.deb ...
Unpacking libstd-rust-1.41:armhf (1.41.1+dfsg1-1~deb10u1+rpi1) ...
Selecting previously unselected package libstd-rust-dev:armhf.
Preparing to unpack .../libstd-rust-dev_1.41.1+dfsg1-1~deb10u1+rpi1_armhf.deb ...
Unpacking libstd-rust-dev:armhf (1.41.1+dfsg1-1~deb10u1+rpi1) ...
Selecting previously unselected package rustc.
Preparing to unpack .../rustc_1.41.1+dfsg1-1~deb10u1+rpi1_armhf.deb ...
Unpacking rustc (1.41.1+dfsg1-1~deb10u1+rpi1) ...
Setting up libllvm7:armhf (1:7.0.1-8+rpi3+deb10u2) ...
Setting up libstd-rust-1.41:armhf (1.41.1+dfsg1-1~deb10u1+rpi1) ...
Setting up libstd-rust-dev:armhf (1.41.1+dfsg1-1~deb10u1+rpi1) ...
Setting up rustc (1.41.1+dfsg1-1~deb10u1+rpi1) ...
Processing triggers for libc-bin (2.28-10+rpt2+rpi1) ...
root@DietPi:~# dietpi-software reinstall 139
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated reinstall

[  OK  ] DietPi-Software | Reinstalling SABnzbd: nzb download manager
[ SUB1 ] DietPi-Backup > Backup
[  OK  ] DietPi-Backup | mkdir -p /mnt/dietpi-backup/data
[ SUB2 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : sabnzbd
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : avahi-daemon
[ INFO ] DietPi-Backup | Checking for required APT packages: rsync
[  OK  ] DietPi-Backup | All required APT packages are already installed.
[  OK  ] DietPi-Backup | Free space check: path=/mnt/dietpi-backup/data | available=52456 MiB | required=688 MiB
[ INFO ] DietPi-Backup | Backup to /mnt/dietpi-backup in progress, please wait...
sending incremental file list
    190,031,063   6%    7.21MB/s    0:00:25 (xfr#924, to-chk=0/50078)   

sent 191,342,282 bytes  received 25,624 bytes  7,504,623.76 bytes/sec
[... 696 lines cut ...]
      self.distribution.run_command(command)
    File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/tmp/pip-build-env-uecppy4q/overlay/lib/python3.7/site-packages/setuptools_rust/setuptools_ext.py", line 123, in run
      build_rust.run()
    File "/tmp/pip-build-env-uecppy4q/overlay/lib/python3.7/site-packages/setuptools_rust/command.py", line 72, in run
      self.run_for_extension(ext)
    File "/tmp/pip-build-env-uecppy4q/overlay/lib/python3.7/site-packages/setuptools_rust/build.py", line 163, in run_for_extension
      dylib_paths = self.build_extension(ext)
    File "/tmp/pip-build-env-uecppy4q/overlay/lib/python3.7/site-packages/setuptools_rust/build.py", line 211, in build_extension
      metadata = json.loads(check_output(metadata_command))
    File "/usr/lib/python3.7/subprocess.py", line 395, in check_output
      **kwargs).stdout
    File "/usr/lib/python3.7/subprocess.py", line 487, in run
      output=stdout, stderr=stderr)
  subprocess.CalledProcessError: Command '['cargo', 'metadata', '--manifest-path', 'src/rust/Cargo.toml', '--format-version', '1']' died with <Signals.SIGSEGV: 11>.
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
[FAILED] DietPi-Software | pip3 install --no-cache-dir -Ur requirements.txt
root@DietPi:/etc/sabnzbd#
```

Any ideas?

MichaIng, 2021-11-27:
This matches the error which the piwheels guys face as well, and which it the reason why the is no pre-compiled wheel available: https://github.com/piwheels/packages/issues/264#issuecomment-980632142

Can you try:
```sh
sudo pip3 install -U pip setuptools wheel
sudo pip3 install -U cryptography
```
Let's skip the overhead of `dietpi-software` for now and concentrate on getting the module compiled. I'll also spin up my Buster VM now.

MichaIng, 2021-11-27:
Okay, nope does not work with the cargo, rustc from Debian, regardless which Python module versions are used. No way around rustup:
```sh
# Do this as root user!
apt purge --autoremove cargo rustc
curl -sSfL 'https://sh.rustup.rs' | dash -s -- --profile minimal
. ~/.cargo/env
pip3 install -U cryptography
```

MichaIng, 2021-11-27:
Ah, awesome, the Python 3.7 wheel has just been uploaded, so all above is not required anymore and `dietpi-software reinstall 139` will now succeed 🥳.

MichaIng, 2021-11-27:
Issue persists on Stretch, but I'm not gonna implement the whole rustup block only for a Debian version where we gonna drop support for in ~1 month 😉. Obviously the amount of users still running Stretch is sufficiently low by now.

bensp, 2021-11-27:
Thanks for your amazing responsiveness, technical knowledge and the time you've invested: issue resolved!

Micha, I made a small donation, thank you again.

closed 2021-11-27 by bensp

MichaIng, 2021-11-27:
Many thanks, highly appreciated 🙂.

MichaIng, 2021-11-27:
Fixed for Stretch as well, simply by defining the cryptography version explicitly that is still pre-compiled for Stretch (Python 3.5): #4999

Build dependencies shouldn't be required on any architecture or distro version anymore, file logging disabled in favour of journal logs.