MichaIng/DietPi #4998: RPi4 (Bullseye update) Kodi error: "Your hardware does not support AE_FMT_FLOAT"

zedrdave, 2021-11-27:
### Creating a bug report/issue

#### Required Information
- DietPi version |
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
```

- Distro version | bullseye 1
- Kernel version | Linux ichigo 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)
- SDcard used: SanDisk ultra

#### Additional Information (if applicable)

- Software title | Kodi Matrix freshly reinstalled after upgrading: `Version: 3:19.1-dietpi3`
  
#### Steps to reproduce
1. Play H264 1080 video

#### Expected behaviour
1. Video plays normally

#### Actual behaviour
- video/audio desynchronising
- Kodi logs contains:
```
2021-11-27 11:56:41.341 T:710     ERROR <general>: Control 55 in window 10025 has been asked to focus, but it can't
2021-11-27 11:56:41.433 T:913      INFO <general>: CAESinkALSA::Initialize - Attempting to open device "sysdefault:CARD=vc4hdmi0"
2021-11-27 11:56:41.438 T:913      INFO <general>: CAESinkALSA::Initialize - Opened device "sysdefault:CARD=vc4hdmi0"
2021-11-27 11:56:41.439 T:913      INFO <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2021-11-27 11:56:41.439 T:913      INFO <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE3
```

From what I can fin about this, it seems like RPi4 should be supporting AE_FMT_FLOAT format… Am I missing something?

MichaIng, 2021-11-27:
Many thanks for your report.

You use the KMS drivers HDMI sound card, which may cause the issue here, not not necessarily. Generally if there is a chance, I aim to configure KMS to allow keep using the regular HDMI sound card. We played a bit around in another case. Could you try the following:
```sh
sed -i -e 's/vc4-kms-v3d/vc4-kms-v3d,noaudio/' -e '/dtoverlay=dietpi-disable_hdmi_audio/d' /boot/config.txt
reboot
# after reboot, to check
aplay -l
```
And if there is no sound OOTB, change the output device in Kodi settings to the newly found HDMI sound card.

zedrdave, 2021-11-27:
Had a look in the `/boot/config` file, and while `vc4-kms-v3d` was enabled, there was no `dietpi-disable_hdmi_audio` in the config.

output of `aplay -l` (before rebooting):

```
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

I tried adding `,noaudio` to the dtoverlay, and upon rebooting, audio was working, display went blank when Kodi was launched. Web interface was still responsive and everything else was working. But I couldn't find any option in the settings to change the video out options.

After reboot:

```
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 3/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
```

Kodi had this in the logs (along with the same `AE_FMT_FLOAT` error mentioned above):

```
2021-11-27 23:13:55.648 T:822      INFO <general>: CAESinkALSA::Initialize - Attempting to open device "sysdefault:CARD=vc4hdmi0"
2021-11-27 23:13:55.652 T:822      INFO <general>: CAESinkALSA - Unable to open device "sysdefault:CARD=vc4hdmi0" for playback
2021-11-27 23:13:55.652 T:822     ERROR <general>: CAESinkALSA::Initialize - failed to initialize device "sysdefault:CARD=vc4hdmi0"
2021-11-27 23:13:55.652 T:822      INFO <general>: CAESinkALSA::Initialize - Attempting to open device "default"
```

MichaIng, 2021-11-27:
The Kodi errors are expected as you defined `vc4hdmi0` explicitly which is disabled now. `default` should be fine as match the HDMI device, i.e. `/etc/asound.conf` contains `card 0`, right?

Removing the `dietpi-disable_hdmi_audio` overlay was just a failsafe step, in case you enabled 3.5mm audio before.

Okay so adding `noaudio` disables this KMS specific HDMI sound card and re-enables the regular one, that is what I would like to have as implementing use of the KMS audio means a complicated code logic... I still do not understand why KMS even has an own audio device, what the purpose of it is. But adding `noaudio` broke video output, that is bad and a bug in the firmware, I'd say. But to rule it out, can you try to explicitly change the audio utput device via web interface to `default` or `bcm2835 HDMI 1`?

zedrdave, 2021-11-28:
Just tried and: same thing. Except for a weird glitch: when trying to quit Kodi from the web interface (after changing the audio), it briefly displayed video (but appeared frozen). Not sure if that has anything to do with it, or just some buffer oddity. In any case, video was still blank after a restart.

MichaIng, 2021-11-28:
I just checked, and now the RPi Kodi build is finally available on the stable repository. Can you test whether this makes a difference. First of all, assure that it is really available in your case:
```sh
apt update
apt policy kodi
```
It should show this one:
```
     2:19.3-1~bullseye 500
        500 https://archive.raspberrypi.org/debian bullseye/main armhf Packages
```
If so install it, overriding our package:
```sh
apt install --reinstall --allow-downgrades kodì
```

zedrdave, 2021-11-28:
Indeed, it seems to be there! but now I'm getting a weird error:

```
root@ichigo:~# apt policy kodi
kodi:
  Installed: 3:19.1-dietpi3
  Candidate: 3:19.1-dietpi3
  Version table:
 *** 3:19.1-dietpi3 100
        100 /var/lib/dpkg/status
     2:19.3-1~bullseye 500
        500 https://archive.raspberrypi.org/debian bullseye/main armhf Packages
root@ichigo:~# apt install --reinstall --allow-downgrades kodì
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package kodì
```

Trying without `--allow-downgrades` etc, doesn't install as would be expected:

```
root@ichigo:~# apt install kodi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
kodi is already the newest version (3:19.1-dietpi3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ichigo:~# apt install --allow-downgrades kodi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
kodi is already the newest version (3:19.1-dietpi3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ichigo:~# apt install --reinstall kodi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reinstallation of kodi is not possible, it cannot be downloaded.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Joulinar, 2021-11-28:
There is a typo on the name `kodi`

```
apt install --reinstall --allow-downgrades kodì
```

But it should be

```
apt install --reinstall --allow-downgrades kodi
```

Check the `i`

zedrdave, 2021-11-28:
@Joulinar Ha. Good catch! I did retry again (using the specific version and therefore typing instead of copy-pasting) and it's now installing…

zedrdave, 2021-11-28:
OK. Done and, while everything seems to work OK, I still get that error in the log…

```
 CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2021-11-28 08:57:18.822 T:871      INFO <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE3
```

Any reason to re-install the dietpi package, or are they mostly the same at this point?

MichaIng, 2021-11-28:
So video works now while you use the `default`/`bcm2835 HDMI 1` sound device in Kodi? And `aplay -l` does not show the `vc4hdmi` sound devices?

The warning may be related to the fact that it is not the KMS HDMI sound card, but not sure. I see this quite often in search machines, probably it's even expected on RPi. Needs more testing.

zedrdave, 2021-11-29:
Not really… 

If I restart with the `vc4-kms-v3d,noaudio`, I get the same Kodi issue with no video getting displayed. And without it, I have no `bcm2835 HDMI 1`, just `vc4hdmi`:

root@ichigo:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

So… Weird thing about "default", is that Alsa does not seem to have a default device set up, despite the config file looking in order (no idea what's going on there, I'm far from an Alsa guru).

Not sure if there's much more to try. But I'm OK living with it as it is. Seems to be playing everything fine, despite complaining about missing format support.

In a possibly unrelated problem, I can't seem to get Airport working in Kodi (log says server is up and running but nothing shows up on the network), and when installing Shairport manually, I need to explicitly set a device (otherwise it complains that Alsa default does not exist)… at which point is complains that the device is already busy (presumably because Kodi is using it). Guess I would need to be using an Alsa mixer, but that's a bit beyond my expertise…

MichaIng, 2021-11-29:
The `default` device is defined in `/etc/asound.conf` as `card` index. The problem with these `vc4hdmi` devices is that depending on settings they get different indices assigned. The beauty about the common sound devices is that it can be easily assured to have `card 0` always correct (disabling the other one via device tree overlay) or having 3.5mm as automatic fallback when no capable HDMI device is attached. The `vc4hdmi` mess with all of this as the get indices higher than the 3.5mm jack, so one needs to manually change the default/used sound card device and cannot simply keep using default/`card 0`.

With older firmware, the `vc4hdmi` were always created with `card 0` and (RPi 4) `card 1`, so one could count on simply incrementing the default card  index by +1 (+2 on RPi 4) whenever KMS is enabled. Now it got very complicated as it disables the regular HDMI device, I saw them leaving `card 1` empty then, in your case using `card 1`, it's unclear whether e.g. USB DAC get a higher or a lower index and all that. Very unpredictable now, so as long as these are spawned, we have not really a way to guarantee that your `dietpi-config` sound card selection will work at all, still after reboot, break as fast as you detach HDMI etc etc.

MichaIng, 2021-12-08:
Despite the error message the issue is solved, isn't it? The problem with the sound card selection is not that big: If HDMI audio is wanted, then one can select `rpi-bcm2835-hdmi`, only the "auto" mode doesn't work as expected but sends output to 3.5mm instead. And of course the sound device can be selected manually within Kodi. I'll have a closer look into this, why `noaudio` breaks KMS and how to implement a reliable auto mode (HDMI playback when attached, else fallback to 3.5mm).

closed 2021-12-08 by MichaIng

zedrdave, 2021-12-14:
Sorry for the late reply: yes, audio seems to work fine despite format error messages in the logs.

Only remaining mystery/issue, is getting Airplay to work (Kodi's built-in airplay stopped working with the upgrade, and installing Shairplay separately conflicts with Kodi's use of the channel)… but i think that's a separate topic :-|

MichaIng, 2021-12-14:
Probably Kodi's AirPlay feature can be disabled to use an external one?

MichaIng, 2021-12-14:
Ah makes sense. You can try to use the ALSA dmix plugin which should allow concurrent audio device access, mixing input together in case. To apply it correctly for the "default" ALSA device, can you show your `cat /etc/asound.conf`, please?

zedrdave, 2021-12-14:
Current `asound.conf` is:
```
cat /etc/asound.conf
pcm.!default {
	type hw
	card 0 
	device 0
}

ctl.!default {
	type hw
	card 0
}
```

(can't remember if I already tried to modify it to get alsa working) 😅

MichaIng, 2021-12-14:
Can you try to change:
```
pcm.!default {
	type hw
	card 0 
	device 0
}
```
to
```
pcm.!default "dmix:0,0"
```
Quick test with `speaker-text`.

zedrdave, 2021-12-14:
Getting error messages (even after quitting Kodi, to ensure no interference)… Honestly too much of an Alsa noob to have any idea what might be wrong here:

```
root@ichigo:~# speaker-test -t wav

speaker-test 1.2.4

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
WAV file(s)
ALSA lib pcm_direct.c:1206:(snd1_pcm_direct_initialize_slave) requested or auto-format is not available
ALSA lib pcm_dmix.c:1087:(snd_pcm_dmix_open) unable to initialize slave
Playback open error: -22,Invalid argument
root@ichigo:~# speaker-test -t wav -Ddmix

speaker-test 1.2.4

Playback device is dmix
Stream parameters are 48000Hz, S16_LE, 1 channels
WAV file(s)
ALSA lib pcm_direct.c:1206:(snd1_pcm_direct_initialize_slave) requested or auto-format is not available
ALSA lib pcm_dmix.c:1087:(snd_pcm_dmix_open) unable to initialize slave
Playback open error: -22,Invalid argument
root@ichigo:~# speaker-test -t wav -Dhw

speaker-test 1.2.4

Playback device is hw
Stream parameters are 48000Hz, S16_LE, 1 channels
WAV file(s)
Sample format not available for playback: Invalid argument
Setting of hwparams failed: Invalid argument
```

MichaIng, 2021-12-14:
Okay it seems to be required to make it a "plug" plugin slave. Try to replace the line with:
```
pcm.!default {
	type plug
	slave.pcm "dmix:0,0"
}
```

zedrdave, 2021-12-14:
Still getting errors:
```
root@ichigo:~# speaker-test -t wav

speaker-test 1.2.4

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
WAV file(s)
ALSA lib pcm_direct.c:1206:(snd1_pcm_direct_initialize_slave) requested or auto-format is not available
ALSA lib pcm_dmix.c:1087:(snd_pcm_dmix_open) unable to initialize slave
Playback open error: -22,Invalid argument
```

zedrdave, 2021-12-14:
This *seems* like it might be [a Raspberry-specific issue](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/157)?

MichaIng, 2021-12-14:
Indeed, it seems to be specifically an issue with the `dmix` plugin, since at that time at least it was used by default with the vc4hdmi audio:
```
# default with dmix
vc4-hdmi.pcm.default {
        @args [ CARD ]
        @args.CARD {
                type string
        }
        type asym
        playback.pcm {
                type plug
                slave.pcm {
                        @func concat
                        strings [ "dmix:" $CARD ]
                }
        }
}
```
Although the alternative config which popcornmix suggested uses `plug` only and the user still had the same issue 🤔. Aside of splitting off capture (microphone) this basically matches my last suggestion. Of course the vc4hdmi or HDMI in general has no microphone embedded, so it generally makes sense to split it for playback only via `type asym`, but not doing so doesn't hurt either. I just checked the current `libasound2-data` package from the RPi repository, and indeed it has been changed to not use the `dmix` plugin anymore, which may be the reason it generally works, even when not defining `/etc/asound.conf` explicitly.