MichaIng/DietPi #5000: RPi Cam Control does not work on Raspberry Pi Zero W with the ARMv6 32-bit image

x0tester0x, 2021-11-27:
I tried it with Apache2 and Lighttpd, but everytime the website shows this:

[image]

I am also using a HQ Camera mount and the latest DietPi version 7.8.2

MichaIng, 2021-11-27:
Many thanks for your report.

RPi-Monitor has an internal webserver and hence does not require an external one like Lighttpd or Apache2. However, having a webserver installed shouldn't interfere. Or do you use it as proxy?

x0tester0x, 2021-11-28:
Yes, but when I install it, by selecting 59 or with the command, it will install also a Webserver. So, when I select in the DietPi settings Lighttpd as my preferred one, the installer will install Lighttpd and if I select Apache2 as my preferred then it will install Apache2.
No I do not use a Proxy.

Joulinar, 2021-11-28:
@MichaIng probably a misunderstanding. The issue is not talking aber `RPi-Monitor`. It's about `RPi Cam Control`. :wink:

MichaIng, 2021-11-28:
Whoosie, thanks, I think I had a little too much time invested in looking into RPi-Monitor yesterday, because of another issue 😄.

I'm no specialist of RPi Cam Control, but I guess the placeholder aims to contain the camera image, right?

So probably raspimjpeg is not running or has an issue:
```sh
journalctl -u raspimjpeg
```

MichaIng, 2021-11-28:
Are you running a Buster system? As a general recommendation when doing bug reports on any GitHub: Use and fill the bug report/issue templates that are provided as usually they are there for a reason to prevent additional Q&A loops.

I recognised that there is a dedicated Buster binary for `raspimjpeg` now: https://github.com/silvanmelchior/RPi_Cam_Web_Interface/tree/master/bin
So I guess the one without suffix is for Bullseye only and does not work on Buster. I've just pushed a commit: https://github.com/MichaIng/DietPi/commit/b7e3d38bc773f09e235b9b3e56767baddaae477a

So if you are running a Buster image and the above `journalctl -u raspimjpeg` indicates that there is an issue with that service, please try to replace the bianry:
```sh
curl -sSfL 'https://raw.githubusercontent.com/silvanmelchior/RPi_Cam_Web_Interface/master/bin/raspimjpeg-buster' -o /usr/local/bin/raspimjpeg
chmod +x /usr/local/bin/raspimjpeg
systemctl restart raspimjpeg
sleep 5
systemctl status raspimjpeg
```

x0tester0x, 2021-11-28:
I am using the latest DietPi release, an that is bullseye:
> I am also using a HQ Camera mount and the latest DietPi version 7.8.2

MichaIng, 2021-11-28:
Ah lol the old issue and this binary needs to be recompiled. Strange only that during the update to v7.8.2 there should have been a workaround applied. Can you please show the following output:
```sh
while read -r line
do
ls -l "$line" "${line%.0}"
done < <(dpkg -L 'libraspberrypi0' | grep '^/usr/lib/arm-linux-gnueabihf/.*\.so.0$')
```

The DietPi version btw has nothing to do with the Debian version, i.e. you can run DietPi v7.8.2 on a Debian Buster or even Debian Stretch system.

x0tester0x, 2021-11-28:
**while read -r line; do ls -l "$line" "${line%.0}"; done < <(dpkg -L 'libraspberrypi0' | grep '^/usr/lib/arm-linux-gnueabihf/.*\.so.0$'):**
```
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libbcm_host.so': No such file or directory
-rw-r--r-- 1 root root 75576 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libbcm_host.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libcontainers.so': No such file or directory
-rw-r--r-- 1 root root 80100 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libcontainers.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libdebug_sym.so': No such file or directory
-rw-r--r-- 1 root root 13724 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libdebug_sym.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libdtovl.so': No such file or directory
-rw-r--r-- 1 root root 50956 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libdtovl.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libmmal.so': No such file or directory
-rw-r--r-- 1 root root 13732 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libmmal.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libmmal_components.so': No such file or directory
-rw-r--r-- 1 root root 46776 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libmmal_components.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libmmal_core.so': No such file or directory
-rw-r--r-- 1 root root 59228 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libmmal_core.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libmmal_util.so': No such file or directory
-rw-r--r-- 1 root root 59228 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libmmal_util.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libmmal_vc_client.so': No such file or directory
-rw-r--r-- 1 root root 46804 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libmmal_vc_client.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libvchiq_arm.so': No such file or directory
-rw-r--r-- 1 root root 26064 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libvchiq_arm.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libvcos.so': No such file or directory
-rw-r--r-- 1 root root 46952 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libvcos.so.0
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libvcsm.so': No such file or directory
-rw-r--r-- 1 root root 30264 Oct 12 17:30 /usr/lib/arm-linux-gnueabihf/libvcsm.so.0
```

MichaIng, 2021-11-28:
Strange. Can you please rerun the last update and paste the full output here:
```sh
dietpi-update -1
```

x0tester0x, 2021-11-28:
**dietpi-update -1:**
```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/Die                                                                                                                                                                                                                         tPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 7.8.2
[ INFO ] DietPi-Update | Repatch was requested: Subversion reduced to "7" to reapply the last update
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v7.7.0
[ INFO ] DietPi-Update | Latest version  : v7.8.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=25871 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : raspimjpeg
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
[  OK  ] DietPi-Update | APT upgrade

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Installing new DietPi code

[  OK  ] DietPi-Update | Downloading update archive
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Installing new DietPi scripts
[  OK  ] DietPi-Update | Installing new DietPi system files
[  OK  ] DietPi-Update | Setting execute permissions for DietPi scripts
[ SUB1 ] DietPi-Set_software > verify_dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] DietPi-Set_software | Comment in /boot/dietpi.txt converted to setting: AUTO_SETUP_GLOBAL_PASSWORD=dietpi
[  OK  ] verify_dietpi.txt  | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches

[ INFO ] DietPi-Update | Current version : v7.7.0
[ INFO ] DietPi-Update | Latest version  : v7.8.2
[ INFO ] DietPi-Patch | Patching to DietPi v7.8...
[  OK  ] DietPi-Patch | sed -i /CONFIG_PREFER_IPV4/d /boot/dietpi.txt
[... 30 lines cut ...]
[  OK  ] DietPi-Update | Incremental patching to v7.8.2 completed
[ INFO ] DietPi-Update | Checking for new available live patches
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[0]='not applied' to end of file /boot/dietpi/.version
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[1]='not applied' to end of file /boot/dietpi/.version
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[2]='not applied' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Applying live patch 0
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[0]='applied'
[ INFO ] DietPi-Update | Applying live patch 1
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[1]='applied'
[ INFO ] DietPi-Update | Applying live patch 2
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[2]='applied'

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Completed

[ INFO ] DietPi-Update | Current version : v7.8.2
[ INFO ] DietPi-Update | Latest version  : v7.8.2
[  OK  ] DietPi-Survey | Purging survey data
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : php7.4-fpm
[  OK  ] DietPi-Services | restart : lighttpd
[  OK  ] DietPi-Services | restart : raspimjpeg
[  OK  ] DietPi-Services | restart : cron
```

MichaIng, 2021-11-28:
Ah yes that makes sense, I think we forgot to add it to the image creation script but added it to the update script only. I'll update the images to contain these.

MichaIng, 2021-11-28:
Applied same workaround in PREP: https://github.com/MichaIng/DietPi/commit/b39bd7b

x0tester0x, 2021-11-28:
Perfect.
Thanks a lot
I will close the issue now.

closed 2021-11-28 by x0tester0x

MichaIng, 2021-11-28:
All RPi ARMv6 and ARMv7 images repacked to contain the links. I also moved the patch on update forward so that it is reapplied on next DietPi update for cases like yours: https://github.com/MichaIng/DietPi/commit/2bfcaee83dfa5a64be47cc5e160184c9e911b272

hellvesper, 2025-01-30:
Hi! I have exactly same issue with fresh installed DietPi v9.9.0 on RPi 1B (armhf) and RPi Cam Web Interface installed through dietpi-software 

```bash
root@DietPi:~# journalctl -u raspimjpeg
Jan 31 00:57:03 DietPi systemd[1]: Started raspimjpeg.service - RaspiMJPEG (DietPi).
Jan 31 00:57:04 DietPi sudo[184]:     root : PWD=/ ; USER=www-data ; COMMAND=/usr/local/bin/raspimjpeg
Jan 31 00:57:04 DietPi sudo[184]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Jan 31 00:57:04 DietPi raspimjpeg.sh[192]: raspimjpeg: error while loading shared libraries: libmmal_core.so: cannot open shared object file: No such file or directory
Jan 31 00:57:04 DietPi sudo[184]: pam_unix(sudo:session): session closed for user www-data
Jan 31 00:57:08 DietPi sudo[287]:     root : PWD=/ ; USER=www-data ; COMMAND=/usr/bin/php /var/www/rpicam/schedule.php
Jan 31 00:57:08 DietPi sudo[287]: pam_unix(sudo:session): session opened for user www-data(uid=33) by (uid=0)
Jan 31 00:57:11 DietPi raspimjpeg.sh[294]: PHP Warning:  filemtime(): stat failed for /var/www/rpicam/status_mjpeg.txt in /var/www/rpicam/schedule.php on line 717
```

```bash
root@DietPi:~# while read -r line
do
ls -l "$line" "${line%.0}"
done < <(dpkg -L 'libraspberrypi0' | grep '^/usr/lib/arm-linux-gnueabihf/.*\.so.0$')
lrwxrwxrwx 1 root root    16 Dec 23 13:52 /usr/lib/arm-linux-gnueabihf/libbcm_host.so -> libbcm_host.so.0
-rw-r--r-- 1 root root 67368 Oct 19  2023 /usr/lib/arm-linux-gnueabihf/libbcm_host.so.0
lrwxrwxrwx 1 root root    17 Dec 23 13:52 /usr/lib/arm-linux-gnueabihf/libdebug_sym.so -> libdebug_sym.so.0
-rw-r--r-- 1 root root 13728 Oct 19  2023 /usr/lib/arm-linux-gnueabihf/libdebug_sym.so.0
lrwxrwxrwx 1 root root    13 Dec 23 13:52 /usr/lib/arm-linux-gnueabihf/libdtovl.so -> libdtovl.so.0
-rw-r--r-- 1 root root 50948 Oct 19  2023 /usr/lib/arm-linux-gnueabihf/libdtovl.so.0
lrwxrwxrwx 1 root root    17 Dec 23 13:52 /usr/lib/arm-linux-gnueabihf/libvchiq_arm.so -> libvchiq_arm.so.0
-rw-r--r-- 1 root root 26064 Oct 19  2023 /usr/lib/arm-linux-gnueabihf/libvchiq_arm.so.0
lrwxrwxrwx 1 root root    12 Dec 23 13:52 /usr/lib/arm-linux-gnueabihf/libvcos.so -> libvcos.so.0
-rw-r--r-- 1 root root 38752 Oct 19  2023 /usr/lib/arm-linux-gnueabihf/libvcos.so.0
```

```bash
root@DietPi:~# raspimjpeg 
raspimjpeg: error while loading shared libraries: libmmal_core.so: cannot open shared object file: No such file or directory
```