MichaIng/DietPi #5002: Home Assistant missing http dependency

kerryland, 2021-11-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.7.3
- Distro version | bullseye 0
- Kernel version | 5.10.63-v8+
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Official Pi Power Supply
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Home Assistant
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Untested

#### Steps to reproduce
1. Make sure you have no webstack installed, although I had my 'Webserver Preference' set to Lighttp.
2. Install 157 Home Assistant

#### Expected behaviour
Home Assistant should be available on port 8123

#### Actual behaviour
Installation fails. Logs mention "Setup failed for dependencies: http".

So I uninstalled it, then installed LLSP (81), and reinstalled Home Assistant. This new installation worked fine. What an excellent piece of software it is too.

#### Extra details
Log excerpts:
```
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Home Assistant: open-source home automation platform

[ INFO ] DietPi-Software | Home Assistant user:  homeassistant
[ INFO ] DietPi-Software | Home Assistant home:  /home/homeassistant
[ INFO ] DietPi-Software | pyenv activation:     ". /home/homeassistant/pyenv-ac                                                                                                             tivate.sh"
[ INFO ] DietPi-Software | pyenv Python version: 3.9.7
[  OK  ] DietPi-Software | Verifying user "homeassistant" to run Home Assistant
[  OK  ] DietPi-Software | useradd -rMU -G dialout,gpio,i2c -d /home/homeassista   
```
```
021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of auth. Setup failed for dependencies: http
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Setup failed for auth: Could not set up all dependencies.
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of webhook. Setup failed for dependencies: http
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Setup failed for webhook: Could not set up all dependencies.
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of config. Setup failed for dependencies: http
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Setup failed for config: Could not set up all dependencies.
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of image. Setup failed for dependencies: http
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Setup failed for image: Could not set up all dependencies.
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of websocket_api. Setup failed for dependencies: http
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Setup failed for websocket_api: Could not set up all dependencies.
2021-11-27 20:47:13 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of cloud. Setup failed for dependencies: http, webhook
```

kerryland, 2021-11-27:
I also think there was a SQL error (but I can't find it) which made me think there was a SQL dependency too, which LLSP (81) would've also fixed, I guess.

Joulinar, 2021-11-27:
Did you restart the service `systemctl restart home-assistant` as described on our online docs? https://dietpi.com/docs/software/home_automation/#home-assistant

kerryland, 2021-11-28:
No I didn't. I confess I read _some_ of that page, so I was aware it would take a while.

I have just uninstalled lighttp and HA is working just fine without it :-)

Sorry for wasting your time. Serves me right for jumping to conclusions,

closed 2021-11-28 by kerryland