MichaIng/DietPi #5005: DietPi fails on freshinstall

29039, 2021-11-28:
#### Details:
- Date           | Sun Nov 28 11:20:24 UTC 2021
- Bug report     | 18375e75-a840-44dd-b4d4-27cee2ae70fc
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPi-1 5.10.0-8-amd64 #1 SMP Debian 5.10.46-4 (2021-08-03) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -qq --allow-change-held-packages install intel-microcode`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Download DietPi_NativePC-BIOS-x86_64-Bullseye.7z (The Native PC BIOS - Direct flash image version) and extract
2. Confirmed that DietPi_NativePC-BIOS-x86_64-Bullseye.img matches SHA256 6c8e7a9650d4feb4409abf884bba2580cd88b968d3632f9dfb7a40d458f331a6 as per hash.txt
3. Use Rufus v3.17.1846 (portable) to write DietPi_NativePC-BIOS-x86_64-Bullseye.img to a 16GB USB Stick with the "List USB Hard Drives" option
4. I made a few tweaks to the dietpi.txt nothing fancy, kept the default one there as dietpi.txt.default. Using RAMlog #1 option.
5. Boot from the 16GB USB Stick on a Core 2 Duo PC (network connected, about 3GB usable RAM out of 4GB) and let it do its thing. I did notice that the ext4 partition got extended to use all the available space automatically.
6. Log in as root via. SSH and let it do its thing, 
7. Don't change much, only the region/keyboard/timezone,  Also did a benchmark as well. But went ahead to install without any extra packages.
8. Let it do its thing, it seems to mostly want to do updates.
9. Error as below
10. Tried to retry, it took a long time on the "update-initramfs: Generating /boot/initrd.img-5.10.0-9-amd64" step but still errored out again (also note that the USB stick I'm using is also pretty slow). Also, I did the built-in bug report :-).
11. I am not sure what I pressed after that, I was trying all the ALT+F1-F12 keys to see if I could get a second shell instance to poke around the problem but eventually it just backed me out into the shell and gave me this template to log here
#### Expected behaviour:

- DietPi should install, considering it's fresh
#### Actual behaviour:

- Incomplete install due to this error
#### Extra details:

- Note: I am by no means an expert, I am just going off ideas from similar threads like https://dietpi.com/phpbb/viewtopic.php?p=40008 , I don't actually know what I'm doing that well.
- I do have a sneaky suspicion that that maybe the /boot partition isn't actually large enough by default for whatever it is that it's trying to do.
- The solution from that topic seems to be:
```apt purge linux-image-5.10.0-8-amd64
reboot
````
But given that this system has never been fully installed, I don't have confidence that the system is actually installed properly. Even if I do this, how do I re-run the setup? At this point I prefer to start the install again with the issue resolved in advance to keep it clean.
#### Additional logs:
```
root@DietPi-1:~# lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
NAME   FSTYPE LABEL  SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
sda                 14.5G  0 disk
├─sda1 vfat          128M  0 part /boot      5e6dffe7-01                          C3E6-D424
└─sda2 ext4         14.3G  0 part /          5e6dffe7-02                          4d7e9d38-16ca-4131-aa23-216cc12c9412
root@DietPi-1:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            1.5G     0  1.5G   0% /dev
tmpfs           299M  8.1M  291M   3% /run
/dev/sda2        15G  1.3G   13G  10% /
tmpfs           1.5G     0  1.5G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           1.5G     0  1.5G   0% /tmp
tmpfs            50M   60K   50M   1% /var/log
/dev/sda1       127M   83M   45M  66% /boot
root@DietPi-1:/# ls -la /boot
total 70908
drwxr-xr-x  5 root root     2048 Jan  1  1970  .
drwxr-xr-x 18 root root     4096 Nov 28 12:02  ..
drwxr-xr-x  2 root root      512 Nov 28  2021 'System Volume Information'
-rwxr-xr-x  1 root root       83 Sep 23 20:35  System.map-5.10.0-8-amd64
-rwxr-xr-x  1 root root       83 Sep 30 19:36  System.map-5.10.0-9-amd64
-rwxr-xr-x  1 root root   236220 Sep 23 20:35  config-5.10.0-8-amd64
-rwxr-xr-x  1 root root   236055 Sep 30 19:36  config-5.10.0-9-amd64
drwxr-xr-x  4 root root     3072 Nov 28 10:44  dietpi
-rwxr-xr-x  1 root root    18092 Aug 14 15:24  dietpi-LICENSE.txt
-rwxr-xr-x  1 root root    14229 Aug 14 15:24  dietpi-README.md
-rwxr-xr-x  1 root root     2955 Aug 15 19:44  dietpi-wifi.txt
-rwxr-xr-x  1 root root    14996 Nov 28 10:44  dietpi.txt
-rwxr-xr-x  1 root root    15102 Aug 15 19:44  dietpi.txt.default
drwxr-xr-x  5 root root      512 Nov 28 10:43  grub
-rwxr-xr-x  1 root root 29199314 Nov 28 10:32  initrd.img-5.10.0-8-amd64
-rwxr-xr-x  1 root root 29204214 Nov 28 10:42  initrd.img-5.10.0-9-amd64
-rwxr-xr-x  1 root root  6820832 Sep 23 20:35  vmlinuz-5.10.0-8-amd64
-rwxr-xr-x  1 root root  6833568 Sep 30 19:36  vmlinuz-5.10.0-9-amd64
```
```
Setting up initramfs-tools (0.140) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.140) ...
ln: failed to create hard link '/boot/initrd.img-5.10.0-9-amd64.dpkg-bak' => '/boot/initrd.img-5.10.0-9-amd64': Operation not permitted
update-initramfs: Generating /boot/initrd.img-5.10.0-9-amd64

gzip: stdout: No space left on device
E: mkinitramfs failure gzip 1
update-initramfs: failed for /boot/initrd.img-5.10.0-9-amd64 with 1.
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2021-11-28:
I guess same issue like this https://dietpi.com/phpbb/viewtopic.php?t=9681

Probably workaround could be https://dietpi.com/phpbb/viewtopic.php?p=40037#p40037

MichaIng, 2021-11-28:
I rebuilt our native PC images to not contain the dedicated boot FAT partitions anymore. The idea was good, but the implied problems are bad. Also, via WSL2 it has become easier to mount an ext partition on Windows, and we should instead add instructions to our online docs about how to do that.

29039, 2021-11-28:
Thanks for your work on this project, most invaluable. It's there a Dev build already available which is all ext?

Joulinar, 2021-11-28:
Should be the regular image as it has been updated today https://dietpi.com/downloads/images/?C=M;O=D

MichaIng, 2021-11-28:
Yes exactly, also Cloudflare cache got cleared, so should be all done.

29039, 2021-11-29:
Wow thanks for fixing it so fast! I tested it out and it's working great!

Joulinar, 2021-11-29:
ok, going to close this issue for now. Feel free to reopen if needed.

closed 2021-11-29 by Joulinar