MichaIng/DietPi #5006: rebooting after PREP-script

Mausy5043, 2021-11-28:
Trying something different today: I started off with the **ARMv8-Bullseye** image on one of my Raspberry Pi's (Model 3B+ rev.1.3) and a custom version of the [`dietpi.txt`](https://github.com/Mausy5043/mod-dietpi/blob/main/machines/rbair11/dietpi.txt). My goal is to eventually be able to fully reinstall a headless DietPi-based machine with all the required packages/applications installed without any manual intervention other than starting a single script and rebooting. 

So after the initial install was complete I ran the `PREP_SYSTEM_FOR_DIETPI.sh`-script (from the `dev` branch) using [this bash script](https://github.com/Mausy5043/mod-dietpi/blob/main/mod-dietpi.sh)

This runs to completion just fine. However, when I then want to reboot using `systemctl reboot` or `shutdown -r now` (`shutdown -r 1` doesn't work; "`Failed to connect to bus`") or `init 6` I invariably end up with a Pi stuck at some intermediate state.   
This machine is calling itself "DietPi" (not as set by me); it has no network connection and I notice that I can login as root with the original (also not as set by me) password `dietpi`.   
After logging in I see an error: `[FAILED] DietPi-Login | Failed to load DietPi-Globals. Skipping DietPi login scripts...`. 
The mountpoint `/boot` is not mounted. So, that explains the error.   
I have not investigated furter; another `reboot` seems to do the trick. But I have to be physically at the machine to power-cycle it. Obviously this is not what I want. I *am* puzzled by the fact that rebooting seems not to reboot the board. Is this by design?

How do I get the Pi to reboot properly?

MichaIng, 2021-11-28:
Use the `reboot` command to reboot, and assure that it is called with root privileges. `DietPi` is the default hostname for DietPi images, `root` and `dietpi` users with password `dietpi`, all as expected, and you require a reboot after PREP before the system is usable, also as expected.

Mausy5043, 2021-11-28:
Having logged in as `root` [here is the transcript of the session](https://gist.github.com/Mausy5043/27aad533d2684659d7379960a8cdfa7d) and as you can see using the command `reboot` to re-start the machine.   
This results in a stuck Pi that needs a power-cycle to proceed with the intended re-install.

MichaIng, 2021-11-28:
How do you know that it stuck when the SSH connection was closed?

Mausy5043, 2021-11-28:
I have a monitor and keyboard connected for testing purposes ;-)

MichaIng, 2021-11-28:
And what is the output that you see on the screen when it hangs?

Mausy5043, 2021-11-28:
This is the screen after the reboot:   
[image]

Logging in as root:   
[image]

MichaIng, 2021-11-28:
I mean the screen output when it hangs, so the last service, systemd and kernel messages before it gets stuck, so that we can get an idea why it gets stuck.

At this stage, can you check:
```sh
fdisk -l /dev/mmcblk0
df
cat /etc/fstab
mount /dev/mmcblk0p1 /boot
ls -l /boot
journalctl -u boot.mount
```

Generally to me it looks like the boot partition is not mounted for some reason. The PREP script should not break the partitioning, it doesn't even touch it. But it adjusts `/etc/fstab` based on the mounted root and boot partitions. So if the above commands indicate that the boot mount is missing in `/etc/fstab` as some debug steps, you could check:
1. Is it present after booting the fresh image from our website and after finishing firstrun setup.
2. Is it still present after PREP has finished

Mausy5043, 2021-11-28:
I'll do some more investigating the coming weeks.

> I mean the screen output when it hangs, 

Not sure I understand you. In the transcript I linked above you can see that at the end I enter `reboot` and the SSH connection is close as a result. When I then look at the monitor I see the machine reboot and am presented with the first image. Just before that I can see just a **few lines** of text flashing by and a red line of text. I'm assuming that is the red text also shown in the second image.

Only when I reboot/power-cycle the usual dmesg-like output and install logging shows up.

MichaIng, 2021-11-28:
> P.S.: Should I delete that directory before rebooting? 🤔

No, only if there are very old kernel module directories left, but this is only the single up-to-date required one. I'll rephrase that. Basically it is informational only to see which kernels are installed and whether this is expected or not, mainly helpful when using 3rd party images with kernels manually downloaded/copied onto the disk, rather than via APT package.

> When I then look at the monitor I see the machine reboot and am presented with the first image

So it does not hang!? Okay so here is what happened most likely:
1. You call `reboot` which did a successful reboot, no hang, all as expected
2. The boot partition fails to mount, which leads to 90 seconds delay/timeout at an early boot stage while the system waits for the boot partition to become mountable. This is why you thought it hangs.
3. After the 90 seconds, boot proceeds, but with limited services started. However, you are presented with a login prompt, which is your first image.
4. When you login, as the boot partition was not mounted, the DietPi scripts fail to execute, which is then pretty expected.

What I do not understand at which stage and why you did a power cycle? At the login prompt? The boot has finished already when you see the login prompt, so no kernel/boot messages are expected anymore. For this you need to attach a screen before you call `reboot` in the first place.

> I could try to capture the `dmesg` output from this state.

More importantly, please run the commands I provided above, so we can identify what I think is that actual issue: https://github.com/MichaIng/DietPi/issues/5006#issuecomment-981122089

Mausy5043, 2021-12-04:
Trying to exclude hardware causes I used a different Pi and SD-card. I connected it to a keyboard and monitor. The same problem seems to happen. :-(

Here's what I do:
1. Starting with the `DietPi_RPi-ARMv8-Bullseye` image and my [custom `dietpi.txt` file](https://github.com/Mausy5043/mod-dietpi/blob/main/machines/rbian/dietpi.txt) I installed DietPi.
2. I cloned and executed my [`re-install`-script](https://github.com/Mausy5043/mod-dietpi/blob/main/mod-dietpi.sh) which effectively downloads and runs the `PREP_SYSTEM_FOR_DIETPI.sh`-script.   
[image]
As far as I could see, the script runs to completion without problems.
[image]

3. `reboot` then drops me into the "intermediate" state. I managed to capture the bootmessages entering into the intermediate state.
[image]

No amount of waiting changes this. So, at this point I need to login with the default `root/dietpi`. Network is not available. So, I had to redirect the output of the commands you requested.

```(bash)
$ fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x0140a7f0

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *      8192   270335   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      270336 15523839 15253504  7.3G 83 Linux

$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.2G  409M  6.5G   6% /
devtmpfs        447M     0  447M   0% /dev
tmpfs           480M     0  480M   0% /dev/shm
tmpfs           192M  2.9M  190M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           480M  4.0K  480M   1% /tmp
tmpfs            50M  8.0K   50M   1% /var/log

$ cat /etc/fstab
# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=0140a7f0-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=0140a7f0-01 /boot vfat noatime,lazytime,rw 0 2

$ mount /dev/mmcblk0p1 /boot
# no output

$ ls -l /boot
total 29766
-rwxr-xr-x 1 root root   18693 Dec  4 08:25 COPYING.linux
-rwxr-xr-x 1 root root    1594 Dec  4 08:26 LICENCE.broadcom
-rwxr-xr-x 1 root root   27441 Dec  4 08:25 bcm2710-rpi-2-b.dtb
-rwxr-xr-x 1 root root   29558 Dec  4 08:25 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root   28939 Dec  4 08:25 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   27429 Dec  4 08:25 bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root   28100 Dec  4 08:25 bcm2710-rpi-zero-2.dtb
-rwxr-xr-x 1 root root   49833 Dec  4 08:25 bcm2711-rpi-4-b.dtb
-rwxr-xr-x 1 root root   49877 Dec  4 08:25 bcm2711-rpi-400.dtb
-rwxr-xr-x 1 root root   50555 Dec  4 08:25 bcm2711-rpi-cm4.dtb
-rwxr-xr-x 1 root root   52456 Dec  4 08:26 bootcode.bin
-rwxr-xr-x 1 root root     135 Dec  4 11:39 cmdline.txt
-rwxr-xr-x 1 root root    3206 Dec  4 11:39 config.txt
drwxr-xr-x 4 root root    2560 Dec  4 11:39 dietpi
-rwxr-xr-x 1 root root   18092 Nov 19 17:23 dietpi-LICENSE.txt
-rwxr-xr-x 1 root root   14566 Nov 19 17:23 dietpi-README.md
-rwxr-xr-x 1 root root   15394 Dec  4 11:39 dietpi.txt
-rwxr-xr-x 1 root root    7279 Dec  4 08:26 fixup.dat
-rwxr-xr-x 1 root root    5411 Dec  4 08:26 fixup4.dat
-rwxr-xr-x 1 root root    3213 Dec  4 08:26 fixup4cd.dat
-rwxr-xr-x 1 root root    8426 Dec  4 08:26 fixup4db.dat
-rwxr-xr-x 1 root root    8420 Dec  4 08:26 fixup4x.dat
-rwxr-xr-x 1 root root    3213 Dec  4 08:26 fixup_cd.dat
-rwxr-xr-x 1 root root   10269 Dec  4 08:26 fixup_db.dat
-rwxr-xr-x 1 root root   10269 Dec  4 08:26 fixup_x.dat
-rwxr-xr-x 1 root root 7906048 Dec  4 08:25 kernel8.img
drwxr-xr-x 2 root root   20992 Dec  4 08:25 overlays
-rwxr-xr-x 1 root root 2965696 Dec  4 08:25 start.elf
-rwxr-xr-x 1 root root 2241504 Dec  4 08:26 start4.elf
-rwxr-xr-x 1 root root  801564 Dec  4 08:26 start4cd.elf
-rwxr-xr-x 1 root root 3737928 Dec  4 08:26 start4db.elf
-rwxr-xr-x 1 root root 2993544 Dec  4 08:26 start4x.elf
-rwxr-xr-x 1 root root  801564 Dec  4 08:25 start_cd.elf
-rwxr-xr-x 1 root root 4809800 Dec  4 08:25 start_db.elf
-rwxr-xr-x 1 root root 3717256 Dec  4 08:26 start_x.elf

$ journalctl -u boot.mount
# no output

$ journalctl
# output held

$ dmesg
# output held

```
I've also saved the output from `journalctl` and from `dmesg`. Not including them here now as they are quite large. Just let me know if you'd like to inspect them and I can pastebin them later.

[image]

4. `reboot` then results in what I would have expected to have happened after point 2.  
[image]

System up and running.
Hoping this helps you in reproducing this problem.

MichaIng, 2021-12-04:
All our RPi images are created with that script, so we obviously were not able to replicate such the last years 😉. So indeed the boot partition fails to mount for some reason while the `fstab` entry is correct, it can be manually mounted without issues and mounts automatically on next boot 🤔.

Can you please retry and from the intermediate state paste the output of:
```
systemctl status boot.mount
journalctl
```
Since `journalctl -u boot.mount` doesn't show anything, it must have been failed on device level already so that this service was skipped in the first place.

Mausy5043, 2021-12-04:
`journalctl` output is here: https://pastebin.com/BkLCjVJu   
`dmesg` output is here: https://pastebin.com/5CDWkdC3

> Can you please retry and from the intermediate state paste the output of:

OK. Will do that tomorrow.

Mausy5043, 2021-12-05:
BTW: Is there a safe place (where I can store e.g. logfiles) that will survive a re-install?

MichaIng, 2021-12-05:
```
Dec 04 11:54:45 DietPi systemd[1]: Found device /dev/disk/by-partuuid/0140a7f0-01.
Dec 04 11:54:46 DietPi systemd[1]: dietpi-fs_partition_resize.service: Succeeded.
Dec 04 11:54:46 DietPi systemd[1]: Finished DietPi-FS_partition_resize.
Dec 04 11:54:46 DietPi systemd[1]: Reached target Local File Systems (Pre).
Dec 04 11:54:46 DietPi systemd[1]: Reached target Swap.
Dec 04 11:54:46 DietPi systemd[1]: Mounting /tmp...
Dec 04 11:54:46 DietPi systemd[1]: Mounting /var/log...
Dec 04 11:54:46 DietPi systemd[1]: Starting File System Check on /dev/disk/by-partuuid/0140a7f0-01...
Dec 04 11:54:46 DietPi systemd[1]: Mounted /tmp.
Dec 04 11:54:46 DietPi systemd[1]: Mounted /var/log.
Dec 04 11:54:46 DietPi systemd[1]: Starting Flush Journal to Persistent Storage...
Dec 04 11:54:46 DietPi systemd[1]: Starting Set console font and keymap...
Dec 04 11:54:46 DietPi systemd[1]: Condition check resulted in Store a System Token in an EFI Variable being skipped.
Dec 04 11:54:46 DietPi systemd[1]: Condition check resulted in Commit a transient machine-id on disk being skipped.
Dec 04 11:54:46 DietPi systemd[1]: systemd-fsck@dev-disk-by\x2dpartuuid-0140a7f0\x2d01.service: Main process exited, code=killed, status=15/TERM
Dec 04 11:54:46 DietPi systemd-fsck[327]: fsck: Warning... fsck.vfat for device /dev/mmcblk0p1 exited with signal 15.
Dec 04 11:54:46 DietPi systemd[1]: systemd-fsck@dev-disk-by\x2dpartuuid-0140a7f0\x2d01.service: Failed with result 'signal'.
Dec 04 11:54:46 DietPi systemd[1]: Stopped File System Check on /dev/disk/by-partuuid/0140a7f0-01.
```
So the boot partition/device is detected, the filesystem check on it started, but then killed with SIGTERM 🤔. If that fsck is running, it is a requirement for its mounting, so it is not mounted because fsck is killed, the question is just why.

Doing an fsck manually works, doesn't it?
```sh
umount /boot
fack.vfat /dev/mmcblk0p1
mount /boot
```

Mausy5043, 2021-12-05:
```
$ fsck.vfat /dev/mmcblk0p1
fsck.vfat 4.2 (2021-01-31)
/dev/mmcblk0p1: 429 files, 69848/258077 clusters
```

Still not convinced this isn't a hardware problem I tested with another SD-card. But the problem persists. So, 3 cards and 2 RPis tested.

MichaIng, 2021-12-05:
Do you have any peripherals attached? In case please try to detach everything but the essential (screen/keyboard). Otherwise I'm also out of ideas, there is really not a single hint what kills the fsck and why. Might be systemd, but no idea why, it didn't timeout or something. And that it works on another reboot is then even stranger.

Something I missed to ask is why you actually use PREP as we have an ARMv8 image based on the official RPi OS 64-bit beta image available (where no on reported such a boot failure with)? The custom `dietpi.txt` (which looks fine) can be added easily to the FAT partition.

I had another look into your build script. When you start it, please assure that `/boot` is really mounted, so that you do not copy your custom files into an empty mount point. Also you are writing into a log file in `/boot` exactly until reboot. Can you try to change the last script lines to:
```
  echo "Rebooting in 60 seconds."
} | tee /boot/.log/mod-dietpi.log

sync
sleep 60
reboot
```
This assures that the log file is closed first, all filesystem changes are synced to disk, doing the sleeping afterwards.
And are there any other files than `dietpi.txt` in `/boot/.bin`?

Mausy5043, 2021-12-05:
No periperals attached.

These files are in `/boot/.bin`:
```
Preparing configuration...
'/tmp/mod-dietpi/machines/rbian/Automation_Custom_PreScript.sh' -> '/boot/.bin/Automation_Custom_PreScript.sh'
'/tmp/mod-dietpi/machines/rbian/Automation_Custom_Script.sh' -> '/boot/.bin/Automation_Custom_Script.sh'
'/tmp/mod-dietpi/machines/rbian/cmdline.txt' -> '/boot/.bin/cmdline.txt'
'/tmp/mod-dietpi/machines/rbian/dietpi.txt' -> '/boot/.bin/dietpi.txt'
```
The files may be inspected [here](https://github.com/Mausy5043/mod-dietpi/tree/main/machines/rbian). I added them only today. Idon't think they are of concern.

Adding to cmdline.txt: `loglevel=8` gives a bit more info.
```
Dec 05 15:18:28 DietPi systemd[1]: Found device /dev/disk/by-partuuid/0140a7f0-01.
Dec 05 15:18:29 DietPi systemd[1]: dietpi-fs_partition_resize.service: Succeeded.
Dec 05 15:18:29 DietPi systemd[1]: Finished DietPi-FS_partition_resize.
Dec 05 15:18:29 DietPi systemd[1]: Reached target Local File Systems (Pre).
Dec 05 15:18:29 DietPi systemd[1]: Reached target Swap.
Dec 05 15:18:29 DietPi systemd[1]: Mounting /tmp...
Dec 05 15:18:29 DietPi systemd[1]: Mounting /var/log...
Dec 05 15:18:29 DietPi systemd[1]: Starting File System Check on /dev/disk/by-partuuid/0140a7f0-01...
Dec 05 15:18:29 DietPi systemd[1]: Mounted /tmp.
Dec 05 15:18:29 DietPi systemd[1]: Mounted /var/log.
Dec 05 15:18:29 DietPi systemd[1]: Starting Flush Journal to Persistent Storage...
Dec 05 15:18:29 DietPi systemd-fsck[326]: Failed to stat /dev/disk/by-partuuid/0140a7f0-01: No such file or directory
Dec 05 15:18:29 DietPi systemd[1]: Starting Set console font and keymap...
Dec 05 15:18:29 DietPi systemd[1]: Condition check resulted in Store a System Token in an EFI Variable being skipped.
Dec 05 15:18:29 DietPi systemd[1]: Condition check resulted in Commit a transient machine-id on disk being skipped.
Dec 05 15:18:29 DietPi systemd[1]: systemd-fsck@dev-disk-by\x2dpartuuid-0140a7f0\x2d01.service: Main process exited, code=exited, status=1/FAILURE
Dec 05 15:18:29 DietPi systemd[1]: systemd-fsck@dev-disk-by\x2dpartuuid-0140a7f0\x2d01.service: Failed with result 'exit-code'.
Dec 05 15:18:29 DietPi systemd[1]: Stopped File System Check on /dev/disk/by-partuuid/0140a7f0-01.
Dec 05 15:18:29 DietPi systemd-journald[127]: Runtime Journal (/run/log/journal/74c21ee129a741d29bd70bdcd6851c3b) is 2.4M, max 19.1M, 16.7M free.
Dec 05 15:18:29 DietPi systemd[1]: Finished Flush Journal to Persistent Storage.
```

The `...failed to stat...` is remarkable.

> we have an ARMv8 image based on the official RPi OS 64-bit beta image available

Is this a different one from the ARMv8-Bullseye image from dietpi.com that I started with?

> Can you try to change the last script lines 

I changed the lines as you suggested. No change in behaviour.

MichaIng, 2021-12-05:
> Is this a different one from the ARMv8-Bullseye image from dietpi.com that I started with?

Ah sorry no, I though you meant the RPi OS 64-bit image. So then the question would be whether using that image without PREP but only copying your scripts and configs inside fails the same way or not.

> The `...failed to stat...` is remarkable.

Indeed, as that partition is previously detected:
```
Dec 05 08:13:06 DietPi kernel: mmcblk0: mmc0:aaaa SL08G 7.40 GiB
Dec 05 08:13:06 DietPi kernel:  mmcblk0: p1 p2
```
DietPi-FS_partition_resize rewrites the partition when expanding the root partition and filesystem. However, as long as you do not shrink it ([DietPi-Imager](https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-imager)), this is not required. Not sure how it could lead to this issue, but its the only think which touches the partition table and also the difference between first boot and subsequent boots. So when you use PREP, could you try to disable this service before rebooting:
```sh
systemctl disable dietpi-fs_partition_resize
```
Ah and since it logs to a dedicated log file, please also check that one (on the failed system), probably it contains some hint, error message or so:
```sh
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

Mausy5043, 2021-12-05:
This is interesting. 

Modified the script to end like this (as suggested):
```
systemctl disable dietpi-fs_partition_resize
sync; sync
sleep 60
reboot
```
This fixed the problem.

However, re-enabling the service afterwards on the re-installed system got me stuck with the intermediate state again on subsequent reboot, and rebooting again (and again) did not help. Where, previously, the intermediate state could be left by rebooting it now kept returning to the intermediate state.

After disabling the service again I could reboot into the normal system. Keeping this service disabled for now.

So, it seems the `dietpi-fs_partition_resize.service` is the main suspect.   
Something to think about in the week ahead. I'll try to get additional testresults later in the week. It is going to be a busy December so, response may be delayed.

MichaIng, 2021-12-05:
At least we found the culprit. But strange as it works without issues on our images where it actually does something. As in your case everything has maximum size already, the partition table is rewritten unchanged (or not touched at all, not sure how `sfdisk` handles it) and the filesystem is definitely not touched.

But the log file `/var/tmp/dietpi/logs/fs_partition_resize.log` may reveal whether something has gone wrong.

Mausy5043, 2021-12-06:
Not much to see there:
```
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/mmcblk0: 14.84 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x0140a7f0

Old situation:

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *      8192   270335   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      270336 31116287 30845952 14.7G 83 Linux

/dev/mmcblk0p2:
New situation:
Disklabel type: dos
Disk identifier: 0x0140a7f0

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *      8192   270335   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      270336 31116287 30845952 14.7G 83 Linux

The partition table has been altered.
resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 3855744 (4k) blocks long.  Nothing to do!
```

closed 2021-12-06 by Mausy5043

MichaIng, 2021-12-06:
No errors, no actual changes to neither partition table, nor filesystem. `partprobe /dev/mmcblk0` and `partx -u /dev/mmcblk0` are called to assure that the kernel gets informed about the partition size change (if any one), not sure whether those may hang for some reason (not blocking the script at least) or leaving the kernel info in a pending state. But strange then also that it only affects the boot partition while the root partition gets R/W remounted fine.

I'll keep this case in mind, so if anything similar happens, we need to further debug which command exactly causes it and why.

sistemicorp, 2022-01-04:
I believe this happened to me, but I didn't find this post before the problem "just went away".

    RPi 4B 2GB
    DietPi v7.9.3

I booted up and noticed that my docker service was not running, ssh'd into the RPi and saw the error,

    Failed to load DietPi-Globals

I did a `reboot` and the next boot was fine.  I didn't really do any other investigation.

I also had the service `dietpi-fs_partition_resize.service` (previously) enabled/unmasked before as I create SD cards from a shrinked (ISO) image (not the original DietPi image).  But several boot ups ago that service ran and expanded the filesystem, and now the service is `disabled` (but not masked).  I see from the resize script that after it runs, it disables itself.

So, to be clear, the boot up that failed to load globals the SD card filesystem was already expanded (several boots before), I do not believe `dietpi-fs_partition_resize.service` was running, as this is it's status,

    ● dietpi-fs_partition_resize.service - DietPi-FS_partition_resize
         Loaded: loaded (/etc/systemd/system/dietpi-fs_partition_resize.service; disabled; vendor preset: enabled)
         Active: inactive (dead)

If this happens again I will try and gather some info based on above and from other posts about this.

My only other thought is that i used https://github.com/Drewsif/PiShrink to make my shrinked ISO.  In that image I enable/unmask `dietpi-fs_partition_resize.service`.  Then after the image is written to a new SD card, it expands.  I realise that the `PiShrink` tool also leaves a script to expand the filesystem on first boot, but it doesn't work on the DietPi image.  Perhaps there is a PiShrink script running that causes the mount to fail/take a long time and causes the Globals to fail.  So I won't use that tool, and I will re-create a fresh DietPi image.  I did look for some evidence of the PiShrink script but I didn't see it.

MichaIng, 2022-01-04:
PiShrink uses `rc.local` to do the partition and filesystem expansion at the end of the boot process, at a stage where it is at least less secure to do so, but generally it does the same and should work. It also removes itself when done. Either way, I currently do not see a way how any of both could break the `/boot` mount, especially of course after at least another reboot.

Generally, if the mount fails again (hence DietPi-Globals fails to load), running directly `journalctl -u boot.mount` should give a hint.

We have btw as well a script to shrink and back an image: https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-imager
We use it for our fresh images, so it packs it into `.7z` with a hash file and the readme, but the basic methods to shrink filesystem, partition and image file are the same.

sistemicorp, 2022-01-04:
I think I found the way to reproduce the problem (for me anyways).
I can consistently cause the boot-up problem (dietpi-globals could not be found) happen by enabling `dietpi-fs_partition_resize.service` and the file system is **already** expanded.

The log at `/var/tmp/dietpi/logs/fs_partition_resize.log` is similiar to above, no errors and "Nothing to do!".

journalctl:

    Jan 04 22:18:18 p1125-man-asy systemd[1]: Reached target Basic System.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Started D-Bus System Message Bus.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Starting DietPi-RAMlog...
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Starting LSB: Lightweight SSH server...
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Started Start entropy gathering daemon (rngd).
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Starting Check for Raspberry Pi EEPROM updates...
    Jan 04 22:18:18 p1125-man-asy systemd[1]: dietpi-ramlog.service: Main process exited, code=exited, status=127/n/a
    Jan 04 22:18:18 p1125-man-asy systemd[1]: dietpi-ramlog.service: Failed with result 'exit-code'.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Failed to start DietPi-RAMlog.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Starting DietPi-PreBoot...
    Jan 04 22:18:18 p1125-man-asy systemd[310]: dietpi-preboot.service: Failed to locate executable /boot/dietpi/preboot: No such file or directory
    Jan 04 22:18:18 p1125-man-asy systemd[310]: dietpi-preboot.service: Failed at step EXEC spawning /boot/dietpi/preboot: No such file or directory
    Jan 04 22:18:18 p1125-man-asy systemd[1]: dietpi-preboot.service: Main process exited, code=exited, status=203/EXEC
    Jan 04 22:18:18 p1125-man-asy systemd[1]: dietpi-preboot.service: Failed with result 'exit-code'.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Failed to start DietPi-PreBoot.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Dependency failed for DietPi-PostBoot.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: dietpi-postboot.service: Job dietpi-postboot.service/start failed with result 'dependency'.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Reached target Network (Pre).
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Starting Network Manager...
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Starting ifup for eth0...

and,

    root@p1125-man-asy:~# journalctl -u dietpi-fs_partition_resize.service
    -- Journal begins at Tue 2022-01-04 22:18:17 GMT, ends at Tue 2022-01-04 22:25:27 GMT. --
    Jan 04 22:18:17 p1125-man-asy systemd[1]: Starting DietPi-FS_partition_resize...
    Jan 04 22:18:18 p1125-man-asy systemd[1]: dietpi-fs_partition_resize.service: Succeeded.
    Jan 04 22:18:18 p1125-man-asy systemd[1]: Finished DietPi-FS_partition_resize.

what else should I provide?

PS: So this happens to me because I enable `dietpi-fs_partition_resize.service` on my master SD card when I want to make an ISO of it, because I want the filesystem to expand on first boot (after I have shrunk that IMG file).  So the next time this master SD card boots, the error occurs, cause on the master the filesystem is already expanded.

MichaIng, 2022-01-04:
Probably we need to add a second sleep time after the partition was altered. I recognised in DietPi-Imager that this is sometimes required, else the follow up commands to detect filesystem or partition table give an empty output:
- https://github.com/MichaIng/DietPi/commit/fb4212c
- https://github.com/MichaIng/DietPi/commit/644ac44

Probably it is similar here, that the changes to the partition table, respectively the `partprobe`/`partx` calls leave the drive for some microseconds in a pending state where partitions cannot be detected/mounted.

MichaIng, 2022-01-04:
Could you retry after applying this change to `/var/lib/dietpi/services/fs_partition_resize.sh`: https://github.com/MichaIng/DietPi/commit/e8f4ec5

It doesn't fit to OP case, as the `ext4` partition was detected successfully, but probably it helps in your case.

MichaIng, 2022-01-05:
Changelog: https://github.com/MichaIng/DietPi/commit/12d301a