MichaIng/DietPi #5008: C2 Bullseye Image Clean install " insufficient free space" Error

bbbxl68, 2021-11-29:
When running clean install on Odroid C2, with 8Gb emmc Card. Kickstart is running but then failing on Software install. Seem some Error with the Filesystem....
 

DietPi-Software
─────────────────────────────────────────────────────
 Step: Applying initial first run setup steps

[ SUB1 ] DietPi-Set_software > passwords
[  OK  ] passwords  | Completed
[  OK  ] DietPi-Software | mount -o remount /var/log
[ SUB1 ] DietPi-Set_software > ntpd-mode (2)
[  OK  ] DietPi-Set_software | systemctl unmask systemd-timesyncd
[  OK  ] DietPi-Set_software | systemctl enable systemd-timesyncd
[  OK  ] DietPi-Set_software | systemctl stop systemd-timesyncd
[  OK  ] DietPi-Set_software | Setting in /boot/dietpi.txt adjusted: CONFIG_NTP_MODE=2
[  OK  ] ntpd-mode 2 | Completed
[  OK  ] DietPi-Software | Applied initial first run setup steps
[FAILED] DietPi-Software | Free space check: path=/ | available=142 MiB | required=500 MiB
[FAILED] DietPi-Software | Install aborted due to insufficient free space

MichaIng, 2021-11-29:
Many thanks for your report.

You are not `zorro` from the forum, are you? However, known issue and I'm currently investigating the cause. I did already repack the image with fresh partition table, but somehow the C2 denies to do an online resizing.

Just to verify that you see the same issue as the user from the forum:
```sh
cat /var/tmp/dietpi/logs/fs_partition_resize.log
resize2fs "$G_ROOTFS_DEV"
```

bbbxl68, 2021-11-29:
Hello MichaIng,

no i´m not zorro just an ordinary guy :)

root@DietPi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/mmcblk0: 7.28 GiB, 7818182656 bytes, 15269888 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xd3630000

Old situation:

Device         Boot  Start     End Sectors   Size Id Type
/dev/mmcblk0p1        2048  264191  262144   128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      264192 1721999 1457808 711.8M 83 Linux

/dev/mmcblk0p2:
New situation:
Disklabel type: dos
Disk identifier: 0xd3630000

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   264191   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      264192 15269887 15005696  7.2G 83 Linux

The partition table has been altered.
resize2fs 1.46.2 (28-Feb-2021)
resize2fs: No space left on device While checking for on-line resizing support
Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/mmcblk0: 7.28 GiB, 7818182656 bytes, 15269888 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xd3630000

Old situation:

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   264191   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      264192 15269887 15005696  7.2G 83 Linux

/dev/mmcblk0p2:
New situation:
Disklabel type: dos
Disk identifier: 0xd3630000

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   264191   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      264192 15269887 15005696  7.2G 83 Linux

The partition table has been altered.
resize2fs 1.46.2 (28-Feb-2021)
resize2fs: No space left on device While checking for on-line resizing support
Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
root@DietPi:~# resize2fs "$G_ROOTFS_DEV"
resize2fs 1.46.2 (28-Feb-2021)
Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
resize2fs: No space left on device While checking for on-line resizing support
root@DietPi:~#

MichaIng, 2021-11-29:
Okay, this is the exact same error. To be sure, you downloaded the image just now, with November 29 timestamp?

I'll create a last fresh image from Meveric's Buster pre-image. Not sure whether the old kernel can have something to do with it. Can you check also:
```sh
dmesg -l emerg,alert,crit,err
```

It that image doesn't work either, I'll to a larger migration to the Armbian kernel. Earlier or later this step would have been required anyway.

bbbxl68, 2021-11-29:
Yeah Image was downloaded just 2 hours ago.

root@DietPi:~# dmesg -l emerg,alert,crit,err
[    0.153581] tv_vout: major number 254 for disp
[    0.157002] genirq: Setting trigger mode 8 for irq 241 failed (gic_set_type+0x0/0x148)
[    0.157024] genirq: Setting trigger mode 8 for irq 242 failed (gic_set_type+0x0/0x148)
[    0.157104] genirq: Setting trigger mode 8 for irq 241 failed (gic_set_type+0x0/0x148)
[    0.157125] genirq: Setting trigger mode 8 for irq 242 failed (gic_set_type+0x0/0x148)
[    0.448816] kvm [1]: kvm_arch_timer: can't find DT node
[    0.448875] kvm [1]: error initializing Hyp mode: -19
[    0.473866] mtdoops: mtd device (mtddev=name/number) must be supplied
[    1.113481] ppmgr_driver_probe called
[    1.114329] amlvideo_init called
[    1.114329] amlvideo_create_instance called
[    1.114332] v4l2_dev.name=:amlvideo-000
[    1.116537] cectx c810023c.aocec: aml_cec_probe(): can't find hdmirx
[    4.594446] proc: unrecognized mount option "hidepid=invisible" or missing value
[   11.444308] proc: unrecognized mount option "hidepid=invisible" or missing value
[  149.135316] JBD2: resize2fs wants too many credits (1063 > 512)

MichaIng, 2021-11-29:
> JBD2: resize2fs wants too many credits (1063 > 512)

Interesting, that is related to the filesystem cache. It seems to be indeed related to old kernel versions: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=857080
A commit which changed something around this (replaced the error message) was done with Linux 4.3: https://github.com/torvalds/linux/commit/6d3ec14

It may be related to the recreating of the filesystem journal which we do recently, at least in combination with the older kernel. The filesystem itself actually has sufficient free space, but probably the journal fails to be expanded accordingly. What is the size of your root partition, and can you test whether resizing in two steps works?
```sh
fdisk -l /dev/mmcblk0
resize2fs /dev/mmcblk0p2 524288
resize2fs /dev/mmcblk0p2
```

bbbxl68, 2021-11-29:
root@DietPi:~# fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 7.28 GiB, 7818182656 bytes, 15269888 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xd3630000

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   264191   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      264192 15269887 15005696  7.2G 83 Linux
root@DietPi:~# resize2fs /dev/mmcblk0p2 524288 # This is 2 GiB
resize2fs 1.46.2 (28-Feb-2021)
Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
resize2fs: No space left on device While checking for on-line resizing support
root@DietPi:~# resize2fs /dev/mmcblk0p2
resize2fs 1.46.2 (28-Feb-2021)
Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
resize2fs: No space left on device While checking for on-line resizing support
root@DietPi:~#

MichaIng, 2021-11-29:
Okay, your SD card isn't very large and even a small resize doesn't work. I think it has to do with the journal reset then, which we do to remove recent filesystem change logs, allowing to further reduce the image size as well. I'm creating an Armbian-based image now, not wasting time with this legacy kernel.

bbbxl68, 2021-11-29:
It´s a ODROID-C2 eMMC Modul with only 8Gb, yes.
I check it out with an normal SD card and give you feedback.

MichaIng, 2021-11-29:
No need to test with an SD card, it pretty sure fails the same way when resizing to 2 GiB failed already. Let me build the Armbian-based image for you to test.

bbbxl68, 2021-11-29:
Aye thank you

MichaIng, 2021-11-29:
Okay I updated the image, please try again.

bbbxl68, 2021-11-29:
It works now for me. Very nice.

bbbxl68, 2021-11-29:
Issue installing Wordpress:

```

 Step: Installing Wordpress: website blog and publishing platform

[  OK  ] DietPi-Software | Checking URL: https://wordpress.org/latest.tar.gz
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://wordpress.org/latest.tar.gz -o latest.tar.gz
[  OK  ] DietPi-Software | tar xf latest.tar.gz --one-top-level=/var/www
[  OK  ] DietPi-Software | rm latest.tar.gz
[  OK  ] DietPi-Software | chown -R www-data:www-data /var/www/wordpress
[ INFO ] DietPi-Create_MySQL_DB | Creating MariaDB database wordpress for user wordpress
ERROR 1064 (42000) at line 1: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '9<P}Umwn(UnPDXmV{j+{FJ.i0GEGYEN';flush privileges' at line 1
[FAILED] Creating MariaDB database wordpress for user wordpress | Exited with error

```

Joulinar, 2021-11-29:
I created a new issue for Wordpress as this might be unrelated and to avoid mixing topics.

MichaIng, 2021-11-30:
Okay great, it worked well in the other case as well. That it's Linux 5.10 instead of 3.16 is another bonus. Probably Kodi/video playback is not as good supported, as not the Mali driver blobs are used, but open source Mesa Lima drivers, but it can be also the other way around that the latte perform better in the meantime. If you face any other issues, please report back, as it is possible that I missed some bits in our code for proper Armbian/upstream kernel handling vs Hardkernel/legacy one.

closed 2021-11-30 by MichaIng

bbbxl68, 2021-11-30:
I installed my OdroidC2 from scratch today with  your new Image without any trouble. Thank you for that.
And i could install Wordpress without trouble. 

Thank you so much. Love my DiePi-C2 again.