MichaIng/DietPi #5009: DietPi-Software | Issue installing Wordpress:

Joulinar, 2021-11-29:
Issue installing Wordpress:

```

 Step: Installing Wordpress: website blog and publishing platform

[  OK  ] DietPi-Software | Checking URL: https://wordpress.org/latest.tar.gz
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://wordpress.org/latest.tar.gz -o latest.tar.gz
[  OK  ] DietPi-Software | tar xf latest.tar.gz --one-top-level=/var/www
[  OK  ] DietPi-Software | rm latest.tar.gz
[  OK  ] DietPi-Software | chown -R www-data:www-data /var/www/wordpress
[ INFO ] DietPi-Create_MySQL_DB | Creating MariaDB database wordpress for user wordpress
ERROR 1064 (42000) at line 1: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near 'xxxxxxxxxxxxxxxxxxxxxxx';flush privileges' at line 1
[FAILED] Creating MariaDB database wordpress for user wordpress | Exited with error

```

_Originally posted by @bbbxl68 in https://github.com/MichaIng/DietPi/issues/5008#issuecomment-982061495_

Joulinar, 2021-11-29:
hmm I tested the installation on a RPi4B 64bit without issues 🤔 

```
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Wordpress: website blog and publishing platform

[  OK  ] DietPi-Software | Checking URL: https://wordpress.org/latest.tar.gz
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://wordpress.org/latest.tar.gz -o latest.tar.gz
[  OK  ] DietPi-Software | tar xf latest.tar.gz --one-top-level=/var/www
[  OK  ] DietPi-Software | rm latest.tar.gz
[  OK  ] DietPi-Software | chown -R www-data:www-data /var/www/wordpress
[ INFO ] DietPi-Create_MySQL_DB | Creating MariaDB database wordpress for user wordpress
[  OK  ] Creating MariaDB database wordpress for user wordpress | Completed
```

MichaIng, 2021-11-29:
I guess the global software password contains some invalid character, like an apostrophe/quoting character?

bbbxl68, 2021-11-30:
Yes you´re right. 
I try it with a "simpler" password without these characters. I try it tomorrow...
Weird that it runs on my Raspi3B with the same password (yes i´m lazy) without these Error.
Thank you so far Michaing i appreciate your work very much.

MichaIng, 2021-11-30:
Relevant is the password at the time when you install Wordpress, as this is used for the database. Then it shouldn't matter where you install it, also the syntax and MariaDB command quotation hasn't changed since the function exists, the password is quoted with regular single quotes `'password'`, so at least those cannot be used then. I can try to change that to the accent this: ` if this is less used 🤔.

MichaIng, 2021-12-08:
Nope, passwords can only be quoted with regular single quotes `'` while database names can only be quotes with backticks `. Only usernames can be quoted with the one one or the other. I do not understand the logic behind this syntax, but it is as it is 😄. I'll mark this issue as closed.

closed 2021-12-08 by MichaIng