MichaIng/DietPi #5010: Buster images are not available for download.

AzagraMac, 2021-11-30:
I have to restore the NanoPi NEO3, the current Bullseye image that is available on the official website is not suitable for the software I have to deploy, at least at present.

There is no way to download the latest stable version, like Raspberry has, which has old images available... 
If it is not possible, I will have to look for alternatives to DietPi. 

Thanks

MichaIng, 2021-11-30:
We do not offer legacy images, I'm sorry. If strongly suggest to get any issue with the software you want to run and current stable OS versions addressed, instead of freshly installing 2.5 years old distro versions, which will earlier or later cause you much more trouble.

Joulinar, 2021-11-30:
We have removed Buster images by purpose. Our aim is more to debug things on current stable Debian version than providing old images.

AzagraMac, 2021-11-30:
As of today it is unfeasible to certify the software for Bullseye, it is not very stable... I don't think it will be certified until at least after the summer of next year. 

Thank you, you will have to look for other solutions. 
Being up to date is not always the best thing to do...

AzagraMac, 2021-11-30:
> We have removed Buster images by purpose. Our aim is more to debug things on current stable Debian version than providing old images.

Leaving old, stable images available is not a bad option either, it is done in other systems. 

Right now, after several installations, Buster performs better as standard than Bullseye, but if I try to compile the software, it is not possible, because nowadays there are libraries and their version, which are in buster. We will see in a few months. 

My journey here with DietPi is over for now.... Thank you.

MichaIng, 2021-11-30:
As long as Armbian offers the Buster image, you can manually convert it, but as said, I strongly recommend against it: https://dietpi.com/docs/hardware/#make-your-own-distribution

Generally, if any software did not manage to be compatible with an OS versions that has fixed/frozen libraries and any major package versions since one year now, then probably this software is not maintained sufficiently active to be recommendable generally. Debian release schedules are fixed, it is not a surprise that Bullseye was released this summer and Bookworm will be in summer 2023, so every developer has plenty of time to get software compatible BEFORE the stable release of major distro versions.

We are not going to ship legacy EOL images only to run behind software developers or make them feel even more comfortable not maintaining their software.

If you tell us which exact issue you face on Bullseye with your particular software, then we may able to help you getting it running, btw. Often only minor changes are required, no guarantee but worth giving a shot before running outdated distros.

AzagraMac, 2021-11-30:
I understand you perfectly, in my company the problem occurs with the latest current version of MacOS, Monterey, just in September they finished approving and certifying the corporate VPN client software for BigSur, it does not work in Monterey, and it will not work until at least summer of next year, in fact, as of today, they have not even started the tests with Monterey due to the low level of use in the company. We will see after the Christmas holidays. 

The NanoPi, I use it for a Dev environment, without having to resort to VMs, the software is from Nutanix, and we have it certified up to Buster... and I do not think that Bullseye testing will begin until mid-January 2022, that testing and certification will be completed if nothing happens, after summer... 1 year to go. 

Apart from my needs... what I have noticed with Bullseye is an increase in temperature and ram consumption, the NanoPi is more drowned with Bullseye than with Buster.

MichaIng, 2021-11-30:
I understand that there are such cases and hence needs to setup new Buster systems when developing or needing compatibility with other running Buster systems. We however have not the capacity to keep up backwards compatibility for so long: When we keep offering Buster images now, we need to guarantee support for at least another two years, so that users do not install a fresh image which is then only supported for a short period of time. Debian itself has whole teams to keep up security maintenance and LTS for their own software repositories, but we have a large list of software options pulled from upstream where support for old Debian libraries/runtime systems/PHP and such is dropped much earlier. E.g. next Nextcloud release will drop PHP7.3 support, which is shipped with Buster. So to keep up support we need to hardcode a fixed old Nextcloud version into `dietpi-software`, keep/add backwards compatible CLI and config calls, etc. At some point we need to assure that at least all new DietPi systems won't imply this need anymore so that we are able to stop keeping and maintaining the additional code two years from now, without disappointing a too large number of (new) DietPi users. And Debian in general is one of the most conservative distros with it's 6 months future freeze, so that all important libraries and dependencies one needs to be compatible with has been released at least 6 months ago when the new Debian is released. This _should_ be a sufficient time corridor to start testing and guarantee compatibility for developers. However, we also see this not being always the case.

> Apart from my needs... what I have noticed with Bullseye is an increase in temperature and ram consumption, the NanoPi is more drowned with Bullseye than with Buster.

At least the core system components do not use significantly more RAM or CPU and the Armbian kernel version (currently 5.10) is the same independently of the underlying Debian version. In fact, on otherwise identically setup minimal testing VMs is even see a little decrease in memory usage, though it is in the error tolerance range of a few MiB. Probably `htop` reveals which process is using more RAM and/or CPU constantly?

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng