MichaIng/DietPi #5012: LXDE not displaying via RDP (but it worked for months)

jalsco, 2021-12-02:
DietPi v7.8.2  
Device model : ROCK Pi 4 (aarch64)
------
My thanks to all who create Diet-Pi !

I hoping for some assistance to solve a problem. I have already exhausted my own knowledge/skills. It is probably not a bug in DietPi, but that is possible.  I run my DietPi  headless. 

For many months I just connect by SSH and WinSCP. Those all still working.  When I want to connect to a LXDE desktop session (for many months without any problem. I just did that by Windows Remote Desktop Connection.  However - that has stopped working - with no changes on my side...  After RDP into the DietPi worked all day, I tried to connect and this problem started. 

The RDP itself is STILL working, including the login to sesman . But it  seems like, after successful login, the DeitPi has no LXDE desktop to display.  I tried to use `dietpi-software` to reinstall LXDE  however it says I already at the latest level and that it is already installed ....  

If anyone can help me get my desktop back, I would greatly appreciate it.  Even any suggestions ? 

Here's what I get : 

[image]

Joulinar, 2021-12-02:
What login option do you select on login screen? Did you tried to reboot your system?

jalsco, 2021-12-02:
Yes have rebooted. 
Login options when I use RDP :
[image]
which shows this : 
[image]

Then after some time - shows the "connection log" message I put into opening post . 

Also - This started a few days ago when I was on Buster.  So I then updated all system software, including upgrading to Bulls-eye .... but the exact same symptoms I have now on Bulls-eye as before. 

All other software that I use is working on Bulls-Eye : including Lighttpsd, MariaDB, Caddy Server, Matomo ..  Only this I have not yet fixed , but as I wrote this problem started before the upgrade to Bulls-eye

jalsco, 2021-12-02:
For diagnostic - I plugged in a HDMI cable and rebooted and my LXDE desktop displays nicely on that screen.  Then try again to connect again by RDP ... Still does not connect.   I also tried to restart LXDE using `lxpanelctl restart`  which gives this: `Cant connect to display: (null)`

jalsco, 2021-12-02:
It seems like this:  [https://askubuntu.com/questions/1373296/xrdp-connection-failed-login-failed-for-display-0-rejection](url)

But I do not see any solution in that post ?

Joulinar, 2021-12-02:
ok can you do following pls

```
reboot
# try to connect using RDP and post following afterwards
journalctl -u xrdp*
systemctl status xrdp*
ss -tulpn | grep LISTEN
```

jalsco, 2021-12-02:
[image]

[image]

[image]

Joulinar, 2021-12-02:
pls if possible, could you copy `journalctl -u xrdp*` directly from SSH terminal. Just copy/past it into the GitHub issue. Way easier to read 😉

jalsco, 2021-12-02:
-- Journal begins at Thu 2021-12-02 07:19:51 EST, ends at Thu 2021-12-02 07:55:35 EST. --
Dec 02 07:19:53 batopi systemd[1]: Starting xrdp session manager...
Dec 02 07:19:54 batopi xrdp-sesman[569]: (569)(281472932010288)[DEBUG] libscp initialized
Dec 02 07:19:54 batopi xrdp-sesman[569]: (569)(281472932010288)[DEBUG] Testing if xrdp-sesman can listen on 127.0.0.1 port 3350.
Dec 02 07:19:54 batopi xrdp-sesman[569]: (569)(281472932010288)[DEBUG] Closed socket 5 (AF_INET6 ::ffff:127.0.0.1 port 3350)
Dec 02 07:19:54 batopi xrdp-sesman[572]: (572)(281472932010288)[INFO ] starting xrdp-sesman with pid 572
Dec 02 07:19:54 batopi xrdp-sesman[572]: (572)(281472932010288)[INFO ] listening to port 3350 on 127.0.0.1
Dec 02 07:19:54 batopi systemd[1]: Started xrdp session manager.
Dec 02 07:19:54 batopi systemd[1]: Starting xrdp daemon...
Dec 02 07:19:54 batopi xrdp[581]: (581)(281473469145104)[INFO ] address [0.0.0.0] port [3389] mode 1
Dec 02 07:19:54 batopi xrdp[581]: (581)(281473469145104)[INFO ] listening to port 3389 on 0.0.0.0
Dec 02 07:19:54 batopi xrdp[581]: (581)(281473469145104)[INFO ] xrdp_listen_pp done
Dec 02 07:19:54 batopi systemd[1]: xrdp.service: Can't open PID file /run/xrdp/xrdp.pid (yet?) after start: Operation not permitted
Dec 02 07:19:54 batopi xrdp[581]: (581)(281473469145104)[DEBUG] Closed socket 7 (AF_INET6 :: port 3389)
Dec 02 07:19:55 batopi systemd[1]: Started xrdp daemon.
Dec 02 07:19:56 batopi xrdp[589]: (589)(281473469145104)[INFO ] starting xrdp with pid 589
Dec 02 07:19:56 batopi xrdp[589]: (589)(281473469145104)[INFO ] address [0.0.0.0] port [3389] mode 1
Dec 02 07:19:56 batopi xrdp[589]: (589)(281473469145104)[INFO ] listening to port 3389 on 0.0.0.0
Dec 02 07:19:56 batopi xrdp[589]: (589)(281473469145104)[INFO ] xrdp_listen_pp done
Dec 02 07:22:46 batopi xrdp[589]: (589)(281473469145104)[INFO ] Socket 12: AF_INET6 connection received from ::ffff:192.168.9.102 port 52082
Dec 02 07:22:46 batopi xrdp[589]: (589)(281473469145104)[DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.33.11 port 3389)
Dec 02 07:22:46 batopi xrdp[1789]: (1789)(281473469145104)[DEBUG] Closed socket 11 (AF_INET6 :: port 3389)
Dec 02 07:22:46 batopi xrdp[1789]: (1789)(281473469145104)[INFO ] Using default X.509 certificate: /etc/xrdp/cert.pem
Dec 02 07:22:46 batopi xrdp[1789]: (1789)(281473469145104)[INFO ] Using default X.509 key file: /etc/xrdp/key.pem
Dec 02 07:22:46 batopi xrdp[1789]: (1789)(281473469145104)[ERROR] Cannot read private key file /etc/xrdp/key.pem: Permission denied
Dec 02 07:22:46 batopi xrdp[1789]: (1789)(281473469145104)[DEBUG] TLSv1.3 enabled
Dec 02 07:22:46 batopi xrdp[1789]: (1789)(281473469145104)[DEBUG] TLSv1.2 enabled
Dec 02 07:22:46 batopi xrdp[1789]: (1789)(281473469145104)[DEBUG] Security layer: requested 11, selected 0
Dec 02 07:22:46 batopi xrdp[1789]: (1789)(281473469145104)[DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.33.11 port 3389)
Dec 02 07:22:46 batopi xrdp[589]: (589)(281473469145104)[INFO ] Socket 12: AF_INET6 connection received from ::ffff:192.168.9.102 port 52083
Dec 02 07:22:46 batopi xrdp[589]: (589)(281473469145104)[DEBUG] Closed socket 12 (AF_INET6 ::ffff:192.168.33.11 port 3389)
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 11 (AF_INET6 :: port 3389)
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] Using default X.509 certificate: /etc/xrdp/cert.pem
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] Using default X.509 key file: /etc/xrdp/key.pem
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[ERROR] Cannot read private key file /etc/xrdp/key.pem: Permission denied
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] TLSv1.3 enabled
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] TLSv1.2 enabled
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] Security layer: requested 0, selected 0
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] connected client computer name: DELL-XPS15
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] adding channel item name rdpdr chan_id 1004 flags 0x80800000
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] adding channel item name rdpsnd chan_id 1005 flags 0xc0000000
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] adding channel item name cliprdr chan_id 1006 flags 0xc0a00000
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] adding channel item name drdynvc chan_id 1007 flags 0xc0800000
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] Non-TLS connection established from ::ffff:192.168.9.102 port 52083: encrypted with standard RDP security
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] xrdp_000006fe_wm_login_mode_event_00000001
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] Loading keymap file /etc/xrdp/km-00000409.ini
Dec 02 07:22:46 batopi xrdp[1790]: (1790)(281473469145104)[WARN ] local keymap file for 0x00000409 found and doesn't match built in keymap, using local keymap file
Dec 02 07:22:53 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] xrdp_wm_log_msg: connecting to sesman ip 127.0.0.1 port 3350
Dec 02 07:22:53 batopi xrdp-sesman[572]: (572)(281472932010288)[INFO ] A connection received from ::ffff:127.0.0.1 port 51296
Dec 02 07:22:53 batopi xrdp[1790]: (1790)(281473469145104)[INFO ] xrdp_wm_log_msg: sesman connect ok
[... cut ...]
Dec 02 07:24:03 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
Dec 02 07:24:06 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
Dec 02 07:24:10 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
Dec 02 07:24:13 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
Dec 02 07:24:17 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
Dec 02 07:24:20 batopi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)

Joulinar, 2021-12-02:
@MichaIng 
can you have a look why session is terminating

jalsco, 2021-12-02:
[image]

Why does xrdp NOT have permission to read the keys ? Could that be the problem ?

Joulinar, 2021-12-02:
again, pls copy messages from SSH terminal if possible. 😉 

No the message is unrelated. I have same on my test system but RDP is working without issues.

Joulinar, 2021-12-02:
question: are you booting directly into a desktop all the time? even if no screen is connected? What is you current autostart option?

jalsco, 2021-12-03:
@Joulinar @MichaIng  Thanks for continuing to help me on this ! 

1. Unless there is some specific reason - e.g. diagnostic trouble-shooting or setting-up - **it is always booted with no screen/keyboard attached.**
2. It goes many **months  at a time with no reboot** (thanks to Dietpi's rock solid reliability 😍). Main reason for reboots are (unexpected) power outages.   Me trying to fix this problem over last few days has caused 4 or 5 reboots, which are the only reboots since Augusts that I know about.... there may have been some others from unknown power-outages (eg while I was sleeping). 
3.  **Auto-start option** - I will look it up... It is months since I looked at that.... It says `│ Current AutoStart Option: 5 ` That is a surprise ?  What is ` 5  : DietPi-CloudShell`  ?  Even now I just googled it, and I cannot find any clear meaninglful description of "Cloudshell" . Do you know of a link for that please  ?

[image]

Joulinar, 2021-12-03:
Option 5 are some statistics that are displayed on console. Usually you should have seen it if a monitor would be connected. Anyway can you set it to option 0 and reboot your system. I guess there is no need for an automated login if no screen is connected usually.

jalsco, 2021-12-03:
Ok I did that. Changed the `AutoStart` to  `option 0`  .. and then `reboot`  and it still has exactly the same  symptoms 😢

jalsco, 2021-12-03:
```
07:22:53 pi xrdp[1790]: (1790)(281473469145104)[DEBUG] xrdp_wm_log_msg: connecting to sesman ip 127.0.0.1 port 3350
07:22:53 pi xrdp-sesman[572]: (572)(281472932010288)[INFO ] A connection received from ::ffff:127.0.0.1 port 51296
07:22:53 pi xrdp[1790]: (1790)(281473469145104)[INFO ] xrdp_wm_log_msg: sesman connect ok
07:22:53 pi xrdp[1790]: (1790)(281473469145104)[DEBUG] xrdp_wm_log_msg: sending login info to session manager, please wait...
07:22:53 pi xrdp[1790]: (1790)(281473469145104)[DEBUG] return value from xrdp_mm_connect 0
07:22:53 pi xrdp-sesman[572]: (572)(281472932010288)[INFO ] ++ created session (access granted): username root, ip ::ffff:192.168.9.102:52083 - socket: 12
07:22:53 pi xrdp-sesman[572]: (572)(281472932010288)[INFO ] starting Xorg session...
07:22:53 pi xrdp[1790]: (1790)(281473469145104)[INFO ] xrdp_wm_log_msg: login successful for display 10
07:22:53 pi xrdp-sesman[1803]: (1803)(281472932010288)[INFO ] calling auth_start_session from pid 1803
07:22:53 pi xrdp-sesman[572]: (572)(281472932010288)[DEBUG] Closed socket 8 (AF_INET6 ::ffff:127.0.0.1 port 3350)
07:22:53 pi xrdp-sesman[1803]: pam_unix(xrdp-sesman:session): session opened for user root(uid=0) by (uid=0)
07:22:53 pi xrdp[1790]: (1790)(281473469145104)[DEBUG] xrdp_wm_log_msg: started connecting
07:22:53 pi xrdp-sesman[1803]: (1803)(281472932010288)[DEBUG] Closed socket 7 (AF_INET6 ::ffff:127.0.0.1 port 3350)
07:22:53 pi xrdp-sesman[1803]: (1803)(281472932010288)[DEBUG] Closed socket 8 (AF_INET6 ::ffff:127.0.0.1 port 3350)
07:22:56 pi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
07:23:00 pi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
07:23:03 pi xrdp-sesman[572]: (572)(281472932010288)[INFO ] ++ terminated session:  username root, display :10.0, session_pid 1803, ip ::ffff:192.168.9.102:52083 - socket: 12
07:23:03 pi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
07:23:07 pi xrdp[1790]: (1790)(281473469145104)[DEBUG] Closed socket 17 (AF_UNIX)
```
⭐@Joulinar ⭐@MichaIng Thanks again for all your help 🙏🏻

These seem the relevant lines from the log. The sequence is: 
1.`starting Xorg session...`
2.`xrdp_wm_log_msg: login successful for display 10`
3.`calling auth_start_session from pid 1803`
4.`session opened for user root(uid=0) by (uid=0)`
5.`xrdp_wm_log_msg: started connecting`
  .... and then...
6.`++ terminated session:  username root, display :10.0, session_pid 1803, ip ::ffff:192.168.9.102:52083 - socket: 12`

MichaIng, 2021-12-04:
Okay, so it's LXDE + XRDP and started on Buster already. So in attempt to replicate better using a Buster system as existing configuration are preserved during the Bullseye upgrade. I have only a headless aarch64 SBC here, but a virtual desktop should work the same. I'm expecting an aarch64 SBC with screen (to closer match the ROCK Pi 4) later today, if customs processing is finishing.

Do you remember when you installed LXDE/XRDP initially, when it still worked? And there was really no change, no APT upgrade or something when it suddenly stopped working?

jalsco, 2021-12-04:
(newpel) [Sat Dec 04] root@batopi: ~/virtualenvs/newpel$ cat ~/.xorgxrdp.10.log
[   255.248]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[   255.248] Build Operating System: linux Debian
[   255.248] Current Operating System: Linux batopi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64
[   255.248] Kernel command line: root=UUID=dfa3ab95-cede-4159-9841-4e31bafe853e rootwait rootfstype=ext4  consoleblank=0 loglevel=4 ubootpart=b670ed71-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u
[   255.249] Build Date: 13 April 2021  04:07:31PM
[   255.249] xorg-server 2:1.20.11-1 (https://www.debian.org/support)
[   255.249] Current version of pixman: 0.40.0
[   255.249]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   255.249] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   255.249] (++) Log file: ".xorgxrdp.10.log", Time: Thu Dec  2 23:42:14 2021
[   255.260] (++) Using config file: "/etc/X11/xrdp/xorg.conf"
[   255.260] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   255.260] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   255.268] (==) ServerLayout "X11 Server"
[   255.268] (**) |-->Screen "Screen (xrdpdev)" (0)
[   255.268] (**) |   |-->Monitor "Monitor"
[   255.269] (**) |   |-->Device "Video Card (xrdpdev)"
[   255.269] (**) |-->Input Device "xrdpMouse"
[   255.269] (**) |-->Input Device "xrdpKeyboard"
[   255.269] (**) Option "DontVTSwitch" "on"
[   255.269] (**) Option "AutoAddDevices" "off"
[   255.269] (**) Not automatically adding devices
[   255.269] (==) Automatically enabling devices
[   255.269] (==) Automatically adding GPU devices
[   255.269] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   255.294] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   255.294]    Entry deleted from font path.
[   255.294] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   255.294]    Entry deleted from font path.
[   255.294] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   255.294]    Entry deleted from font path.
[   255.294] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   255.295]    Entry deleted from font path.
[   255.295] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   255.295]    Entry deleted from font path.
[   255.295] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   255.295]    Entry deleted from font path.
[   255.295] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[   255.295] (==) ModulePath set to "/usr/lib/xorg/modules"
[   255.295] (**) Extension "DPMS" is disabled
[   255.295] (II) Loader magic: 0xaaaadcde0e28
[   255.295] (II) Module ABI versions:
[   255.295]    X.Org ANSI C Emulation: 0.4
[   255.295]    X.Org Video Driver: 24.1
[   255.295]    X.Org XInput driver : 24.1
[   255.295]    X.Org Server Extension : 10.0
[   255.297] (--) using VT number 2

[   255.297] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[   255.299] (II) xfree86: Adding drm device (/dev/dri/card1)
[   255.300] (II) xfree86: Adding drm device (/dev/dri/card0)
[   255.301] (II) no primary bus or device found
[   255.301]    falling back to /sys/devices/platform/display-subsystem/drm/card1
[   255.301] (II) "glx" will be loaded. This was enabled by default and also specified in the config file.
[   255.301] (II) LoadModule: "dbe"
[   255.301] (II) Module "dbe" already built-in
[   255.301] (II) LoadModule: "ddc"
[   255.301] (II) Module "ddc" already built-in
[   255.301] (II) LoadModule: "extmod"
[   255.301] (II) Module "extmod" already built-in
[   255.301] (II) LoadModule: "glx"
[   255.304] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[... 23 lines cut ...]
[   255.562]    ABI class: X.Org ANSI C Emulation, version 0.4
[   255.562] (II) LoadModule: "xrdpdev"
[   255.563] (II) Loading /usr/lib/xorg/modules/drivers/xrdpdev_drv.so
[   255.566] (EE) Failed to load /usr/lib/xorg/modules/drivers/xrdpdev_drv.so: /usr/lib/xorg/modules/drivers/xrdpdev_drv.so: undefined symbol: rdpRRModeDestroy
[   255.566] (EE) Failed to load module "xrdpdev" (loader failed, 0)
[   255.566] (II) LoadModule: "xrdpmouse"
[   255.567] (II) Loading /usr/lib/xorg/modules/input/xrdpmouse_drv.so
[   255.568] (EE) Failed to load /usr/lib/xorg/modules/input/xrdpmouse_drv.so: /usr/lib/xorg/modules/input/xrdpmouse_drv.so: undefined symbol: rdpUnregisterInputCallback
[   255.568] (EE) Failed to load module "xrdpmouse" (loader failed, 0)
[   255.568] (II) LoadModule: "xrdpkeyb"
[   255.569] (II) Loading /usr/lib/xorg/modules/input/xrdpkeyb_drv.so
[   255.570] (EE) Failed to load /usr/lib/xorg/modules/input/xrdpkeyb_drv.so: /usr/lib/xorg/modules/input/xrdpkeyb_drv.so: undefined symbol: rdpUnregisterInputCallback
[   255.570] (EE) Failed to load module "xrdpkeyb" (loader failed, 0)
[   255.570] (EE) No drivers available.
[   255.570] (EE)
Fatal server error:
[   255.570] (EE) no screens found(EE)
[   255.570] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[   255.570] (EE) Please also check the log file at ".xorgxrdp.10.log" for additional information.
[   255.570] (EE)
[   255.571] (EE) Server terminated with error (1). Closing log file.
(newpel) [Sat Dec 04] root@batopi: ~/virtualenvs/newpel$

jalsco, 2021-12-04:
_Do you remember when you installed LXDE/XRDP initially, when it still worked?_ 

It was installed with Buster approx July/August (2021), installed by direct attached screen, and after fully setup the screen was removed.  I tried out both VNC and RDP in the first month or two  (both worked), but easier to comply with the network here, so I stuck with  RDP,  and VNC was removed.  

I access the DietPi many hours every day. But mostly is by SSH and WinSCP. I log into   LXDE  (always by RDP) maybe 3 or 4 times each day because there's a couple things I prefer checking on the Desktop GUI (incl - Firefox).  . But each session into RDP usually only about 5-10minutes.   

RDP/LXDE worked fully until a few days ago - I had logged into it a few times already that day. There was some problem here with power on that day. It looked like the DietPi had not rebooted after power back (i.e no lights, SSH and ping not responsive )  so live power cable was removed , (wait), the re-insert power.  A few minutes later all the lights flashing, ping and SSH working, but RDP showing current symptoms. ...   I tried my best .. but couldn't get it working.   Later that night (same day) I installed Bulls-eye - hoping - that the upgrade would re-install LXDE and RDP and wipe out the problem. It did not..  So therefore raised this Github issue.   

All he other things on the DietPI contain important data.  Caddy, MAriaDB, etc... and some websites being served . The LXDE desktop can be wiped out (reset to defaults ) and that would be no problem. That is not true of the other systems. I say that, because maybe that it the easier option to fix it...   But I do  not know. 

We know now - that LXDE desktop is there (proven by the attaching HDMI screen). The log also shows that the "log in" and "port access" is working. Seems like jsut that last step - of sesman handing over to an RDP session.  
  
APT upgrades done regularly (every few days )... I been building some python/javascript websites by Pelican in last few weeks. Trying out many of the plugins and other Python packages, lots of installing / uninstallng etc... BUT  none of that requires LXDE or any GUI desktop.    Mostly RDP was only  used for me to log in and grab a few screenshots from Firefox running on the DietPi... 

Only really need 1 ( or maybe 2 ) concurrent LXDE sessions... I do not doubt how well it handles the concurrency.

Joulinar, 2021-12-04:
Hmm your system is having issues to load libraries. Can you share content of one of the directories

```
ls -la /usr/lib/xorg/modules/input/
```

MichaIng, 2021-12-04:
> There was some problem here with power on that day.

Probably that corrupted some required file. Please try to reboot with a force fsck and then reinstall the XRDP packages:
```sh
> /forcefsck
reboot
# after reboot
cat /run/initramfs/fsck.log
apt install --reinstall xrdp xorgxrdp
```

jalsco, 2021-12-05:
```
[Sun Dec 05] root@batopi: /$ pwd
/
[Sun Dec 05] root@batopi: /$ /forcefsck
-bash: /forcefsck: No such file or directory
[Sun Dec 05] root@batopi: /$ cd root
[Sun Dec 05] root@batopi: ~$ /forcefsck
-bash: /forcefsck: No such file or directory
[Sun Dec 05] root@batopi: ~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            920M     0  920M   0% /dev
tmpfs           199M  5.8M  193M   3% /run
/dev/mmcblk1p1   59G  7.5G   51G  13% /
tmpfs           992M     0  992M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.0G   80K  1.0G   1% /tmp
tmpfs            50M     0   50M   0% /var/log
[Sun Dec 05] root@batopi: ~$
```

`/forcefsck`  not running 😢😢  I am logged in as root . 🤷‍♂️

Joulinar, 2021-12-05:
You did not copied the exact command. This should create a file on root level

```
> /forcefsck
```

jalsco, 2021-12-05:
ah ok - ` > ` I have woken up a bit stupid this morning.. Sorry 🙏🏻

jalsco, 2021-12-05:
```
[Sun Dec 05] root@batopi: ~$ apt install --reinstall xrdp xorgxrdp
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 2 reinstalled, 0 to remove and 0 not upgraded.
Need to get 485 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 xrdp arm64 0.9.12-1.1 [428 kB]
Get:2 https://deb.debian.org/debian bullseye/main arm64 xorgxrdp arm64 1:0.2.12-1 [56.4 kB]
Fetched 485 kB in 3s (153 kB/s)
(Reading database ... 49972 files and directories currently installed.)
Preparing to unpack .../xrdp_0.9.12-1.1_arm64.deb ...
Unpacking xrdp (0.9.12-1.1) over (0.9.12-1.1) ...
Preparing to unpack .../xorgxrdp_1%3a0.2.12-1_arm64.deb ...
Unpacking xorgxrdp (1:0.2.12-1) over (1:0.2.12-1) ...
Setting up xrdp (0.9.12-1.1) ...
Setting up xorgxrdp (1:0.2.12-1) ...
Processing triggers for libc-bin (2.31-13+deb11u2) ...
```

```
[Sun Dec 05] root@batopi: ~$ cat /run/initramfs/fsck.log
Log of fsck -C -a -V -t ext4 /dev/mmcblk1p1
Sun Dec  5 07:06:44 2021

fsck from util-linux 2.33.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk1p1] fsck.ext4 -a -C0 /dev/mmcblk1p1
/dev/mmcblk1p1: clean, 237564/3800736 files, 2196653/15626240 blocks

Sun Dec  5 07:06:44 2021
----------------
[Sun Dec 05] root@batopi: ~$
```

But it did not fix the problem - still same.   I will get a new `xorgxrdp.10.log`

jalsco, 2021-12-05:
**xorgxrdp.10.log**

```

[Sun Dec 05] root@batopi: ~$ cat ~/.xorgxrdp.10.log
[   172.333]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[   172.333] Build Operating System: linux Debian
[   172.333] Current Operating System: Linux batopi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64
[   172.333] Kernel command line: root=UUID=dfa3ab95-cede-4159-9841-4e31bafe853e rootwait rootfstype=ext4  consoleblank=0 loglevel=4 ubootpart=b670ed71-01                                                                                                                                                                                                                                    usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u
[   172.333] Build Date: 13 April 2021  04:07:31PM
[   172.333] xorg-server 2:1.20.11-1 (https://www.debian.org/support)
[   172.333] Current version of pixman: 0.40.0
[   172.333]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   172.333] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   172.334] (++) Log file: ".xorgxrdp.10.log", Time: Sun Dec  5 02:18:58 2021
[   172.344] (++) Using config file: "/etc/X11/xrdp/xorg.conf"
[   172.344] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   172.344] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   172.350] (==) ServerLayout "X11 Server"
[   172.350] (**) |-->Screen "Screen (xrdpdev)" (0)
[   172.350] (**) |   |-->Monitor "Monitor"
[   172.351] (**) |   |-->Device "Video Card (xrdpdev)"
[   172.351] (**) |-->Input Device "xrdpMouse"
[   172.351] (**) |-->Input Device "xrdpKeyboard"
[   172.351] (**) Option "DontVTSwitch" "on"
[   172.351] (**) Option "AutoAddDevices" "off"
[   172.351] (**) Not automatically adding devices
[   172.351] (==) Automatically enabling devices
[   172.351] (==) Automatically adding GPU devices
[   172.351] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   172.378] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   172.378]    Entry deleted from font path.
[   172.378] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   172.378]    Entry deleted from font path.
[   172.378] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   172.378]    Entry deleted from font path.
[   172.378] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   172.378]    Entry deleted from font path.
[   172.378] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   172.378]    Entry deleted from font path.
[   172.378] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   172.378]    Entry deleted from font path.
[   172.378] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[   172.378] (==) ModulePath set to "/usr/lib/xorg/modules"
[   172.378] (**) Extension "DPMS" is disabled
[   172.378] (II) Loader magic: 0xaaaad9160e28
[   172.378] (II) Module ABI versions:
[   172.378]    X.Org ANSI C Emulation: 0.4
[   172.378]    X.Org Video Driver: 24.1
[   172.378]    X.Org XInput driver : 24.1
[   172.378]    X.Org Server Extension : 10.0
[   172.379] (--) using VT number 2

[   172.379] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[   172.381] (II) xfree86: Adding drm device (/dev/dri/card1)
[   172.381] (II) xfree86: Adding drm device (/dev/dri/card0)
[   172.381] (II) no primary bus or device found
[   172.381]    falling back to /sys/devices/platform/display-subsystem/drm/card1
[   172.381] (II) "glx" will be loaded. This was enabled by default and also specified in the config file.
[   172.381] (II) LoadModule: "dbe"
[   172.382] (II) Module "dbe" already built-in
[   172.382] (II) LoadModule: "ddc"
[   172.382] (II) Module "ddc" already built-in
[... 28 lines cut ...]
[   172.621] (II) LoadModule: "xrdpdev"
[   172.623] (II) Loading /usr/lib/xorg/modules/drivers/xrdpdev_drv.so
[   172.626] (EE) Failed to load /usr/lib/xorg/modules/drivers/xrdpdev_drv.so: /usr/lib/xorg/modules/drivers/xrdpdev_drv.so: undefined symbol: rdpRRModeDe                                                                                                                                                                                                                                   stroy
[   172.626] (EE) Failed to load module "xrdpdev" (loader failed, 0)
[   172.626] (II) LoadModule: "xrdpmouse"
[   172.627] (II) Loading /usr/lib/xorg/modules/input/xrdpmouse_drv.so
[   172.629] (EE) Failed to load /usr/lib/xorg/modules/input/xrdpmouse_drv.so: /usr/lib/xorg/modules/input/xrdpmouse_drv.so: undefined symbol: rdpUnregist                                                                                                                                                                                                                                   erInputCallback
[   172.629] (EE) Failed to load module "xrdpmouse" (loader failed, 0)
[   172.629] (II) LoadModule: "xrdpkeyb"
[   172.629] (II) Loading /usr/lib/xorg/modules/input/xrdpkeyb_drv.so
[... cut ...]
[   172.631] (EE) Failed to load module "xrdpkeyb" (loader failed, 0)
[   172.631] (EE) No drivers available.
[   172.631] (EE)
Fatal server error:
[   172.631] (EE) no screens found(EE)
[   172.631] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[   172.631] (EE) Please also check the log file at ".xorgxrdp.10.log" for additional information.
[   172.631] (EE)
[   172.631] (EE) Server terminated with error (1). Closing log file.
[Sun Dec 05] root@batopi: ~$
```

MichaIng, 2021-12-05:
> Failed to load /usr/lib/xorg/modules/libxorgxrdp.so: /usr/lib/xorg/modules/libxorgxrdp.so: undefined symbol: glamor_xv_init

glamor is the 2D hardware acceleration method for the X11 KMS/modesetting DDX, which is used by default. The DRM devices for KMS are there:
> [   172.381] (II) xfree86: Adding drm device (/dev/dri/card1)
[   172.381] (II) xfree86: Adding drm device (/dev/dri/card0)

Starting the LXDE desktop from local console `startx` works fine, does it? I wonder whether glamor needs to be disabled explicitly for XRDP, but the question why this was not required before and I never saw it being required in any other case. However, we can play around with it a bit. First of all create a defined X11 config, with KMS and glamor:
```sh
rm /etc/X11/xorg.conf # just in case there is a legacy config
cat << '_EOF_' > /etc/X11/xorg.conf.d/20-modesetting.conf
Section "Device"
    Identifier  "Rockchip Graphics"
    Driver      "modesetting"
    Option      "AccelMethod"    "glamor"
    Option      "DRI"            "2"
EndSection
_EOF_
```
Check back the logs. Since the above AFAIK matches the default, it shouldn't change something. Then let's try to disable glamor:
```sh
sed -i 's/"glamor"/"none"/' /etc/X11/xorg.conf.d/20-modesetting.conf
```
And try again.

jalsco, 2021-12-05:
*Starting the LXDE desktop from local console startx works fine, does it?*
I never explicitly started  LXDE using `startx`, because when I say I use `ssh` I mean I use `putty` over an ssh connection.  What we know is that when my DietPi boots up with screen and keyboard attached, it boots into LXDE desktop which runs without any problem.  I am assuming that is the same as  *Starting the LXDE desktop from local console *.

jalsco, 2021-12-05:
```
[Sun Dec 05] root@batopi: ~/virtualenvs/newpel$ rm /etc/X11/xorg.conf # just in case there is a                  legacy config
rm: cannot remove '/etc/X11/xorg.conf': No such file or directory
[Sun Dec 05] root@batopi: ~/virtualenvs/newpel$ cat << '_EOF_' > /etc/X11/xorg.conf.d/20-modesetting.conf
Section "Device"
    Identifier  "Rockchip Graphics"
    Driver      "modesetting"
    Option      "AccelMethod"    "glamor"
    Option      "DRI"            "2"
EndSection
_EOF_
[Sun Dec 05] root@batopi: ~/virtualenvs/newpel$

```

jalsco, 2021-12-05:
```
[Sun Dec 05] root@batopi: ~/virtualenvs/newpel$ cat ~/.xorgxrdp.10.log
[ 30034.878]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[ 30034.879] Build Operating System: linux Debian
[ 30034.879] Current Operating System: Linux batopi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64
[ 30034.879] Kernel command line: root=UUID=dfa3ab95-cede-4159-9841-4e31bafe853e rootwait rootfstype=ext4  consoleblank=0 loglevel=4 ubootpart=b670ed71-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u
[ 30034.879] Build Date: 13 April 2021  04:07:31PM
[ 30034.879] xorg-server 2:1.20.11-1 (https://www.debian.org/support)
[ 30034.879] Current version of pixman: 0.40.0
[ 30034.879]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[ 30034.879] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[ 30034.879] (++) Log file: ".xorgxrdp.10.log", Time: Sun Dec  5 10:36:41 2021
[ 30034.880] (++) Using config file: "/etc/X11/xrdp/xorg.conf"
[ 30034.880] (==) Using config directory: "/etc/X11/xorg.conf.d"
[ 30034.880] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[ 30034.880] (==) ServerLayout "X11 Server"
[ 30034.880] (**) |-->Screen "Screen (xrdpdev)" (0)
[ 30034.880] (**) |   |-->Monitor "Monitor"
[ 30034.881] (**) |   |-->Device "Video Card (xrdpdev)"
[ 30034.881] (**) |   |-->GPUDevice "Video Card (xrdpdev)"
[ 30034.881] (**) |-->Input Device "xrdpMouse"
[ 30034.881] (**) |-->Input Device "xrdpKeyboard"
[ 30034.881] (**) Option "DontVTSwitch" "on"
[ 30034.881] (**) Option "AutoAddDevices" "off"
[ 30034.881] (**) Not automatically adding devices
[ 30034.881] (==) Automatically enabling devices
[ 30034.881] (==) Automatically adding GPU devices
[ 30034.882] (==) Max clients allowed: 256, resource mask: 0x1fffff
[ 30034.882] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[ 30034.882]    Entry deleted from font path.
[ 30034.882] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[ 30034.882]    Entry deleted from font path.
[ 30034.882] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[ 30034.882]    Entry deleted from font path.
[ 30034.882] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[ 30034.882]    Entry deleted from font path.
[ 30034.882] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[ 30034.882]    Entry deleted from font path.
[ 30034.882] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[ 30034.882]    Entry deleted from font path.
[ 30034.882] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[ 30034.882] (==) ModulePath set to "/usr/lib/xorg/modules"
[ 30034.882] (**) Extension "DPMS" is disabled
[ 30034.882] (II) Loader magic: 0xaaaab4c20e28
[ 30034.882] (II) Module ABI versions:
[ 30034.882]    X.Org ANSI C Emulation: 0.4
[ 30034.882]    X.Org Video Driver: 24.1
[ 30034.882]    X.Org XInput driver : 24.1
[ 30034.882]    X.Org Server Extension : 10.0
[ 30034.884] (--) using VT number 2

[ 30034.884] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[ 30034.886] (II) xfree86: Adding drm device (/dev/dri/card1)
[ 30034.887] (II) xfree86: Adding drm device (/dev/dri/card0)
[ 30034.887] (II) no primary bus or device found
[ 30034.887]    falling back to /sys/devices/platform/display-subsystem/drm/card1
[ 30034.887] (II) "glx" will be loaded. This was enabled by default and also specified in the config file.
[ 30034.887] (II) LoadModule: "dbe"
[ 30034.888] (II) Module "dbe" already built-in
[ 30034.888] (II) LoadModule: "ddc"
[ 30034.888] (II) Module "ddc" already built-in
[ 30034.888] (II) LoadModule: "extmod"
[ 30034.888] (II) Module "extmod" already built-in
[... 26 lines cut ...]
[ 30034.907] (II) LoadModule: "xrdpdev"
[ 30034.907] (II) Loading /usr/lib/xorg/modules/drivers/xrdpdev_drv.so
[ 30034.908] (EE) Failed to load /usr/lib/xorg/modules/drivers/xrdpdev_drv.so: /usr/lib/xorg/modules/drivers/xrdpdev_drv.so: undefined symbol: rdpRRModeDestroy
[ 30034.908] (EE) Failed to load module "xrdpdev" (loader failed, 0)
[ 30034.908] (II) LoadModule: "xrdpmouse"
[ 30034.908] (II) Loading /usr/lib/xorg/modules/input/xrdpmouse_drv.so
[ 30034.909] (EE) Failed to load /usr/lib/xorg/modules/input/xrdpmouse_drv.so: /usr/lib/xorg/modules/input/xrdpmouse_drv.so: undefined symbol: rdpUnregisterInputCallback
[ 30034.909] (EE) Failed to load module "xrdpmouse" (loader failed, 0)
[ 30034.909] (II) LoadModule: "xrdpkeyb"
[ 30034.909] (II) Loading /usr/lib/xorg/modules/input/xrdpkeyb_drv.so
[ 30034.909] (EE) Failed to load /usr/lib/xorg/modules/input/xrdpkeyb_drv.so: /usr/lib/xorg/modules/input/xrdpkeyb_drv.so: undefined symbol: rdpUnregisterInputCallback
[ 30034.909] (EE) Failed to load module "xrdpkeyb" (loader failed, 0)
[ 30034.909] (EE) No drivers available.
[ 30034.909] (EE)
Fatal server error:
[ 30034.909] (EE) no screens found(EE)
[ 30034.909] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[ 30034.909] (EE) Please also check the log file at ".xorgxrdp.10.log" for additional information.
[ 30034.910] (EE)
[ 30034.910] (EE) Server terminated with error (1). Closing log file.
[Sun Dec 05] root@batopi: ~/virtualenvs/newpel$
```

MichaIng, 2021-12-05:
Is the error message still the same? I.e. in `~/.xorgxrdp.10.log` this one:
> Failed to load /usr/lib/xorg/modules/libxorgxrdp.so: /usr/lib/xorg/modules/libxorgxrdp.so: undefined symbol: glamor_xv_init

jalsco, 2021-12-05:
yes - seems so -  from log I just posted - I think u replied before I posted it 
```
[ 30034.893] (II) LoadModule: "xorgxrdp"
[ 30034.893] (II) Loading /usr/lib/xorg/modules/libxorgxrdp.so
[ 30034.906] (EE) Failed to load /usr/lib/xorg/modules/libxorgxrdp.so: /usr/lib/xorg/modules/libxorgxrdp.so: undefined symbol: glamor_xv_init
[ 30034.906] (EE) Failed to load module "xorgxrdp" (loader failed, 0)
```

MichaIng, 2021-12-05:
I compared the Debian Bullseye XRDP X11 config with the upstream one here: https://raw.githubusercontent.com/neutrinolabs/xorgxrdp/devel/xrdpdev/xorg.conf

Interesting difference (aside of a fix for keyboard/mouse inputs, which we do apply already), is that Debian's config contains `Load "glamoregl"`, while the upstream config does not. It pulls EGL libraries as dependency, but not the OpenGLES ones. It should be installed by `dietpi-software`, but please assure it is:
```sh
apt install libgles2
```
And if that still does not work, let's see whether using the upstream configuration without the `glamoregl` module load works:
```sh
curl -sSfL 'https://raw.githubusercontent.com/neutrinolabs/xorgxrdp/devel/xrdpdev/xorg.conf' -o /etc/X11/xrdp/xorg.conf
```
Since it explicitly sets `Section "Device"`, it makes sense that our attempt with the `20-modesetting.conf` has no effect. Remove that again:
```sh
rm /etc/X11/xorg.conf.d/20-modesetting.conf
```

jalsco, 2021-12-06:
```
[Mon Dec 06] root@batopi: ~$ apt install libgles2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libgles2 is already the newest version (1.3.2-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[Mon Dec 06] root@batopi: ~$ ^C
[Mon Dec 06] root@batopi: ~$ curl -sSfL 'https://raw.githubusercontent.com/neutrinolabs/xorgxrdp/devel/xrdpdev/xorg.conf' -o /etc/X11/xrdp/xorg.conf
[Mon Dec 06] root@batopi: ~$ rm /etc/X11/xorg.conf.d/20-modesetting.conf
rm: remove regular file '/etc/X11/xorg.conf.d/20-modesetting.conf'? y
```

**Should there be some step to restart Xorg after those changes ?** 

Here's the log: 

```
[Mon Dec 06] root@batopi: ~$ cat ~/.xorgxrdp.10.log
[ 79398.503]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[ 79398.503] Build Operating System: linux Debian
[ 79398.503] Current Operating System: Linux batopi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64
[ 79398.503] Kernel command line: root=UUID=dfa3ab95-cede-4159-9841-4e31bafe853e rootwait rootfstype=ext4  consoleblank=0 loglevel=4 ubootpar                                                                                                                                                                                                                                                t=b670ed71-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u
[ 79398.503] Build Date: 13 April 2021  04:07:31PM
[ 79398.503] xorg-server 2:1.20.11-1 (https://www.debian.org/support)
[ 79398.503] Current version of pixman: 0.40.0
[ 79398.503]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[ 79398.503] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[ 79398.503] (++) Log file: ".xorgxrdp.10.log", Time: Mon Dec  6 00:19:24 2021
[ 79398.504] (++) Using config file: "/etc/X11/xrdp/xorg.conf"
[ 79398.504] (==) Using config directory: "/etc/X11/xorg.conf.d"
[ 79398.504] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[ 79398.504] (**) Option "defaultserverlayout" "X11 Server"
[ 79398.504] (**) ServerLayout "X11 Server"
[ 79398.504] (**) |-->Screen "Screen (xrdpdev)" (0)
[ 79398.504] (**) |   |-->Monitor "Monitor"
[ 79398.504] (**) |   |-->Device "Video Card (xrdpdev)"
[ 79398.505] (**) |-->Input Device "xrdpMouse"
[ 79398.505] (**) |-->Input Device "xrdpKeyboard"
[ 79398.505] (**) Option "DontVTSwitch" "on"
[ 79398.505] (**) Option "AutoAddDevices" "off"
[ 79398.505] (**) Not automatically adding devices
[ 79398.505] (==) Automatically enabling devices
[ 79398.505] (==) Automatically adding GPU devices
[ 79398.505] (==) Max clients allowed: 256, resource mask: 0x1fffff
[ 79398.505] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[ 79398.505]    Entry deleted from font path.
[ 79398.505] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[ 79398.505]    Entry deleted from font path.
[ 79398.505] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[ 79398.505]    Entry deleted from font path.
[ 79398.505] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[ 79398.505]    Entry deleted from font path.
[ 79398.505] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[ 79398.505]    Entry deleted from font path.
[ 79398.505] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[ 79398.505]    Entry deleted from font path.
[ 79398.505] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[ 79398.505] (==) ModulePath set to "/usr/lib/xorg/modules"
[ 79398.505] (**) Extension "DPMS" is disabled
[ 79398.505] (II) Loader magic: 0xaaaac3bc0e28
[ 79398.505] (II) Module ABI versions:
[ 79398.505]    X.Org ANSI C Emulation: 0.4
[... 43 lines cut ...]
[ 79398.519] (II) LoadModule: "xrdpdev"
[ 79398.519] (II) Loading /usr/lib/xorg/modules/drivers/xrdpdev_drv.so
[ 79398.519] (EE) Failed to load /usr/lib/xorg/modules/drivers/xrdpdev_drv.so: /usr/lib/xorg/modules/drivers/xrdpdev_drv.so: undefined symbol                                                                                                                                                                                                                                                : rdpRRModeDestroy
[ 79398.519] (EE) Failed to load module "xrdpdev" (loader failed, 0)
[ 79398.519] (II) LoadModule: "xrdpmouse"
[ 79398.520] (II) Loading /usr/lib/xorg/modules/input/xrdpmouse_drv.so
[ 79398.520] (EE) Failed to load /usr/lib/xorg/modules/input/xrdpmouse_drv.so: /usr/lib/xorg/modules/input/xrdpmouse_drv.so: undefined symbol                                                                                                                                                                                                                                                : rdpUnregisterInputCallback
[ 79398.520] (EE) Failed to load module "xrdpmouse" (loader failed, 0)
[ 79398.520] (II) LoadModule: "xrdpkeyb"
[ 79398.520] (II) Loading /usr/lib/xorg/modules/input/xrdpkeyb_drv.so
[ 79398.520] (EE) Failed to load /usr/lib/xorg/modules/input/xrdpkeyb_drv.so: /usr/lib/xorg/modules/input/xrdpkeyb_drv.so: undefined symbol:                                                                                                  [... cut ...]
[ 79398.520] (EE) Failed to load module "xrdpkeyb" (loader failed, 0)
[ 79398.520] (EE) No drivers available.
[ 79398.520] (EE)
Fatal server error:
[ 79398.520] (EE) no screens found(EE)
[ 79398.520] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[ 79398.520] (EE) Please also check the log file at ".xorgxrdp.10.log" for additional information.
[ 79398.520] (EE)
[ 79398.520] (EE) Server terminated with error (1). Closing log file.
[Mon Dec 06] root@batopi: ~$
```

MichaIng, 2021-12-06:
Ah yes, restarting XRDP is probably required:
```sh
systemctl restart xrdp
```
Can you also check whether there are any other X11 configurations:
```
for i in /etc/X11/xorg.conf /etc/X11/xorg.conf.d/* /usr/share/X11/xorg.conf.d/*
[[ -f $i ]] || continue
echo "${i}:"
cat "$i"
done
```

jalsco, 2021-12-07:
`/etc/X11/xorg.conf` <- does not exist

`/etc/X11/xrdp/xorg.conf` attached
[xorg.conf.txt](https://github.com/MichaIng/DietPi/files/7666601/xorg.conf.txt)

```
[Tue Dec 07] root@batopi: /$ ls /etc/X11/xorg.conf.d/* -al
-rw-r--r-- 1 root root 121 Jul 15 07:25 /etc/X11/xorg.conf.d/98-dietpi-disable_dpms.conf
```
```
[Tue Dec 07] root@batopi: /$ ls /usr/share/X11/xorg.conf.d/*.* -al
-rw-r--r-- 1 root root 1350 Apr 13  2021 /usr/share/X11/xorg.conf.d/10-quirks.conf
-rw-r--r-- 1 root root 1429 Aug 17  2020 /usr/share/X11/xorg.conf.d/40-libinput.conf
```

MichaIng, 2021-12-07:
Can you try to invalidate the config which aims to disabling DPMS/screen blanking:
```sh
mv /etc/X11/xorg.conf.d/98-dietpi-disable_dpms.conf{,.off}
```
Everything else is package defaults, `/etc/X11/xrdp/xorg.conf` the upstream default which we just downloaded. I'm puzzled and I think we need the help of the XRDP developer to find out where `undefined symbol: glamor_xv_init` could come from.

MichaIng, 2021-12-07:
We got a great reply from the XRDP developer. First of all, your last log: Was this after you did `systemctl restart xrdp`?
Can you please run as well the following commands and paste their output here:
```sh
G_AG_CHECK_INSTALL_PREREQ binutils-common
nm -D /usr/lib/xorg/modules/libglamoregl.so | grep glamor_xv_init
```
And since generally it is preferred, let's re-add the glamor module to be loaded:
```sh
G_CONFIG_INJECT 'Load "glamoregl"' '    Load "glamoregl"' /etc/X11/xrdp/xorg.conf 'Load "vbe"'
systemctl restart xrdp
```
Strangely I don't see it being loaded anywhere in your logs above. I do suspect, since glamor support was added ~2 years ago, after Debian Buster, that the old `/etc/X11/xrdp/xorg.conf` did not contain loading this module yet, and it didn't get updated since we manually applied the patch for broken keyboard/mouse input. But the newer XRDP expects/requires the module to be loaded. This doesn't explain why it didn't work on Buster, but I guess this was a different reason.

I just verified that on Buster `Load "glamoregl"` is not done, so the above two commands should indeed fix it, at least it would be a very well matching explanation.

jalsco, 2021-12-07:
Yes, I think the log was after I did the reload.
That is awesome that the XRDP developer reply so fast !

jalsco, 2021-12-07:
OMG ! 
❤⭐❤⭐❤⭐❤⭐❤⭐❤⭐❤
😊👍🏻😊👍🏻😊👍🏻😊👍🏻😊👍🏻😊👍🏻😊👍🏻

jalsco, 2021-12-07:
[image]

MichaIng, 2021-12-07:
Okay, I compared Bullseye and Buster config files and there is another difference. To do this cleaner, you may do the following:
```sh
mv /etc/X11/xrdp/xorg.conf.dpkg-dist /etc/X11/xrdp/xorg.conf
GGI_PRESERVE=1 G_CONFIG_INJECT 'Option[[:blank:]]+"DefaultServerLayout"' '    Option "DefaultServerLayout" "X11 Server"' /etc/X11/xrdp/xorg.conf 'Section[[:blank:]]+"ServerFlags"'
systemctl restart xrdp
```
So the full Bullseye config is used and the fix for keyboard/mouse input re-applied cleanly.
__________
Ah great, so we found the culprit! I'll add the above two commands to our Bullseye upgrade guide so others won't run into the same issue again.

jalsco, 2021-12-07:
Change made : 
```
(newpel) [Tue Dec 07] root@batopi: ~/virtualenvs/newpel$ mv /etc/X11/xrdp/xorg.conf.dpkg-dist /etc/X11/xrdp/xorg.conf
GGI_PRESERVE=1 G_CONFIG_INJECT 'Option[[:blank:]]+"DefaultServerLayout"' '    Option "DefaultServerLayout" "X11 Server"' /etc/X11/xrdp/xorg.conf 'Section[[:blank:]]+"ServerFlags"'
systemctl restart xrdp
mv: overwrite '/etc/X11/xrdp/xorg.conf'? y
[  OK  ] G_CONFIG_INJECT | Added setting     Option "DefaultServerLayout" "X11 Server" to /etc/X11/xrdp/xorg.conf after line Section "ServerFlags"
```

and confirmed - still working ✔🙏🏻

MichaIng, 2021-12-07:
Great, I added this step to the blog post as a post upgrade software migrations step: https://dietpi.com/blog/?p=811#2.2.4-software-migrations

jalsco, 2021-12-07:
wow ! I had almost given up.... Thanks again !

I think this issue can be closed -  unless any more comments need to be added for future reference.

MichaIng, 2021-12-07:
The hint about the module from the XRDP developer did point me into the right direction 🙏. Jep I'll mark this as closed/solved then.

closed 2021-12-07 by MichaIng

jalsco, 2021-12-07:
@MichaIng  are u still there  ?  I lost something  in the last step ...  the screenshot I sent shows a txt file on the desktop (minimised) name starts with "NETWORK..." ... after the very last step.. that is gone.  I assume it is in backup or? , maybe even somewhere closer ? I just cannot find it now ?

MichaIng, 2021-12-07:
What I see is a task, not a text file, probably an editor with a text file opened? Did you save it before restarting the XRDP server? If not, then I'm afraid it's lost, if you did save it, then it should be still there.

jalsco, 2021-12-07:
hmmm...  it was there when i took screen shot. becasue it minised it for the screen shot.  Yeah- it is an app - xpad - which was open with a txt file. Ok - I will find it in the backups. thanks again

MichaIng, 2021-12-07:
It is not located/stored at the desktop at least. The question it where you did store this file to. If you did store it anywhere and not just kept it in the open text editor, then it should be where you saved it the last time.

jalsco, 2021-12-07:
ok - all good - thanks !

jalsco, 2021-12-07:
found !  😁