MichaIng/DietPi #5014: Raspberry zero serial console login problem

ad7718, 2021-12-02:
#### Required Information
- DietPi version | DietPi_RPi-ARMv6-Bullseye (IDK exact version, because I can't login)
- Distro version | -
- Kernel version | -
- SBC model | Raspberry Pi Zero
- Power supply used | 5V 2A Xiaomi charger
- SDcard used | 2GB Kingston

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?
Yes.

#### Steps to reproduce
1. Connect to UART0 using USB-serial adapter
2. Connect with Putty or Screen (default settings, 115200)

#### Expected behaviour
User can login using serial console and use menus, terminal must be working without internet connection

#### Actual behaviour
When I try to enter login/password, it seems that terminal sends some extra characters or they are broken on SBC side. I've somehow logged in using cuteCom and after that I've proceeded with putty, menus are almost unusable, I need to click on arrows several times to move cursor and menu behaviour is unpredictable. Also, as my SBC have no WIFI I want to use it without internet connection(maybe all OK with that, I need working login and menus to check that). I've tested Raspbian on the same SBC and had no problems with serial console.

MichaIng, 2021-12-02:
> I need to click on arrows several times to move cursor and menu behaviour is unpredictable.

On the serial console you have a cursor?!?

Otherwise not sure, the exact same serial console and login getty is enabled on DietPi than on Raspberry Pi OS: `/dev/serial0`, hence the primary serial console, which is UART0 on models without onboard WiFi/BT adapter and UART1 on those with WiFi/BT adapter (since Bluetooth then by default uses the `/dev/ttyAMA0` UART0 device). Do you see boot/kernel messages nicely in PuTTY not being malformed?

> Also, as my SBC have no WIFI, I want to use it without internet connection(maybe all OK with that, I need working login and menus to check that).

That is currently not gracefully handled. The scripts called on first login expect a network connection so that an initial update and in case initial software installs can be done. Also I wonder how you want to use that RPi Zero if there is nothing installed on it and not even `apt` can be used? Or do you manually (offline) copy packages/software onto that device? Or do you mean that only after the initial setup? In case then this will be not an issue. At best disable the MOTD in `dietpi-banner`, DietPi update checks in `/boot/dietpi.txt` and change the network time sync mode to `custom` in `dietpi-config` > Advanced Option, so daily cron jobs do not try to download the current MOTD and perform a network time sync. Keep in mind to re-enable it if you want to do any updates, as the system clock will definitely by out of sync and hence HTTPS connections may fail due to certificates detected as out of valid range.

ad7718, 2021-12-02:
> On the serial console you have a cursor?!?

I mean menu item selector in this case.

> Do you see boot/kernel messages nicely in PuTTY not being malformed?

No, I can't see them on DietPi, only on Raspbian.

> Also I wonder how you want to use that RPi Zero if there is nothing installed on it and not even apt can be used? Or do you manually (offline) copy packages/software onto that device?

In case of this board, I'm manually installing WI-Fi driver first, then I'm using apt and other things. But copying programs without the network and running them without internet is also needed option, for example for data loggers and other things, internet is not always required. It will be good if there'll be an option to get bare console on first boot(no problems with this on Raspbian).

> At best disable the MOTD in dietpi-banner

It's unusable now, because of broken serial.

I've made some screenshots:
[image]
[image]
[image]
[image]

MichaIng, 2021-12-02:
It looks like you have two login consoles enabled somehow. Did you change anything about that in `config.txt` or `cmdline.txt`?
The image ships with boot messages and a login getty active on `/dev/serial0` only. If it is enabled on `/dev/ttyAMA0` as well, since the first is only a symlink to the second required due to the different primary serial console devices on different RPi models, you have doubled boot output and a doubled login console on the same actual device, hence doubled input as well. If you used our current ARMv6 image and you did not manually change anything on it, then probably for some reason `/dev/ttyAMA0` was auto-enabled, not sure. A solution may then be to mask the two related services. During first run setup it is then anyway converted from `/dev/serial0` to `/dev/ttyAMA0` or `/dev/ttyS0` depending on the RPi model.

About network, we have an offline mode request pending: #2915
The idea is to have a `dietpi.txt` flag, so that all actions which require network access are skipped but the essential setup steps still done. This could then be selected from these network check error prompts as well.

So you have a WiFi dongle and it does not work with the shipped WiFi firmware OOTB? If so, generally you can use a custom script that is executed before even initial network setup is done: https://github.com/MichaIng/DietPi/blob/733d81e/dietpi.txt#L65-L68
You can place the required firmware and a custom script to install it, into the boot/FAT partition right after flashing the image, then adjust `dietpi.txt` to have WiFi enabled.

ad7718, 2021-12-02:
> It looks like you have two login consoles enabled somehow. Did you change anything about that in config.txt or cmdline.txt?
No, it's fresh install.

>About network, we have an offline mode request pending: #2915
That would be good.

>So you have a WiFi dongle and it does not work with the shipped WiFi firmware OOTB?
Tried with Ralink MT7601U USB dongle, it didn't worked on DietPI, but worked on Raspbian.

MichaIng, 2021-12-02:
> Tried with Ralink MT7601U USB dongle, it didn't worked on DietPI, but worked on Raspbian.

That isn't possible as Raspberry Pi OS and DietPi ship exactly (I mean exactly) the same firmware and kernel packages. Did you test it on the recent Raspberry Pi OS **Bullseye**? And how did you try to configure it? Prior to first boot via `dietpi.txt` + `dietpi-wifi.txt` and/or as well via `dietpi-config`? And how/where does it exactly fail?
```sh
journalctl -u ifup@wlan0
dmesg -l emerg,alert,crit,err
```

Mysterious about the serial console. On Raspberry Pi Zero W there is unexpected serial getty started, but it `/dev/serial0` points to `/dev/ttyS0` there, so in theory those could behave differently. To solve it for now, you can add a script `Automation_Custom_PreScript.sh` to the FAT partition (where `dietpi.txt` is located) as mentioned, to explicitly disable and set a mask for the conflicting serial getty:
```sh
#!/bin/dash
systemctl disable --now serial-getty@ttyAMA0
systemctl mask serial-getty@ttyAMA0
```
I'll retest this on RPi 2 again, which doesn't have WiFi/BT either, hence should behave the same regarding UART/serial devices and consoles.

ad7718, 2021-12-02:
> Did you test it on the recent Raspberry Pi OS Bullseye?

Yes.

>And how did you try to configure it? Prior to first boot via dietpi.txt + dietpi-wifi.txt and/or as well via dietpi-config? And how/where does it exactly fail?

On Raspbian I've created /boot/wpa_supplicant.conf file and entering login/pass/country code there, it connects to wifi on the first boot. On DietPi I've edited dietpi-wifi.txt and entered aWIFI_SSID[0] and aWIFI_KEY[0] parameters, also entered correct country code in dietpi.txt(AUTO_SETUP_NET_WIFI_COUNTRY_CODE). I can only check if it works by scanning local network, and RPi does not appear there, maybe I'll find that causes that if serial will be working.

> To solve it for now, you can add a script Automation_Custom_PreScript.sh to the FAT partition (where dietpi.txt is located) as mentioned, to explicitly disable and set a mask for the conflicting serial getty:

Tried this, doesn't work, same behaviour.

MichaIng, 2021-12-02:
> Tried this, doesn't work, same behaviour.

Note that this only works (this script executed) on first boot. Can you access the ext4 filesystem? If so, the script could be placed into `/var/lib/dietpi/postboot.d` from where it is executed at the end of every boot session (hence should be removed once succeeded).

Joulinar, 2021-12-02:
WiFi would need to be activated before first boot by setting

```
AUTO_SETUP_NET_WIFI_ENABLED=1
```

ad7718, 2021-12-02:
> Note that this only works (this script executed) on first boot. Can you access the ext4 filesystem? If so, the script could be placed into `/var/lib/dietpi/postboot.d` from where it is executed at the end of every boot session (hence should be removed once succeeded).

That worked, but first login attempt always fails, reproduced 2 times. Menu is working now.

ad7718, 2021-12-02:
@Joulinar 
> WiFi would need to be activated before first boot by setting
> 
> ```
> AUTO_SETUP_NET_WIFI_ENABLED=1
> ```

Oh, that was not obvious. But still no connection.

MichaIng, 2021-12-02:
Similarly, this is only applied on very first boot. If the serial console works now, you can use `dietpi-config` to enable and configure WiFi. You should see the SSID that you added to `dietpi-wifi.txt` listed already.

ad7718, 2021-12-02:
WiFi works now.

ad7718, 2021-12-03:
Now dietpi-software runs every time the system starts, I'm ending the wizard with "continue with DietPi as a pure minimal image?" and getting this:
[image]

MichaIng, 2021-12-03:
Please hit `Cancel` and paste the output of:
```sh
cat /var/tmp/dietpi/logs/dietpi-firstrun-setup.log
```
DietPi-Software must have failed at some point as the install state wasn't incremented as it should have been. Strange only that you should have seen another error prompt about this error then as the script cannot exit with an error code without either an interactive error handler prompt of when you manually interrupt it (CTRL+C).

ad7718, 2021-12-03:
```
[FAILED] DietPi-Software | Free space check: path=/ | available=392 MiB | required=500 MiB
[FAILED] DietPi-Software | Install aborted due to insufficient free space
```

Interesting, why it wants 500MiB to finish, I don't want to install anything.

MichaIng, 2021-12-03:
You have an SD card with less than 4 GiB? This is the minimum which works without manual changes. Or was the filesystem not expended successfully?

The check however can actually be removed. It was mainly required as `dietpi-software` did an APT upgrade where all upgraded packages are by default first downloaded to disk, then extracted and then installed over the existing ones. So in combination with possibly newly created APT cache and list files this can easily take several 100 MiBs when the last APT upgrade was a while back. And of course the chosen software installs take space. But `dietpi-software` does not generally upgrade all APT packages anymore, and furthermore it is thoroughly error-handled in the meantime, so that any error due to "really" insufficient free space leads to an error prompt or exit. So no need to have this 500 MiB failsafe free space check.

To remove the space check:
- `sed -i '/G_CHECK_FREESPACE/d' /boot/dietpi/dietpi-software`

MichaIng, 2021-12-03:
Space check removed for next release, it indeed doesn't make much sense anymore: https://github.com/MichaIng/DietPi/commit/d2c3aff897442b382c14f645b5428fb884285037
Little but significant typo in commit text: `The script is not thoroughly error handled` => `The script is now thoroughly error handled` 😉

ad7718, 2021-12-03:
> You have an SD card with less than 4 GiB? This is the minimum which works without manual changes. Or was the filesystem not expended successfully?

Yes, I have 2 GiB microsd :) (See first post)
Good to know that you fixed that. I hope serial problem will also be fixed some day. Thank you)

MichaIng, 2021-12-04:
We'll think about falling back to WiFi automatically if no Ethernet interface is available. Currently it is only done the other way round. Through that fallback makes often no sense as it requires WiFi credentials, of course. This could be checked additionally. But finally I aim to merge `dietpi-wifi.txt` back into `dietpi.txt`. The aim was to allow configuring multiple WiFi SSIDs directly, but I think this is nearly never used or required for the very first boot and that one needs to modify two config files and the large probably confusing (?) `dietpi-wifi.txt` causes more issues than benefits. So the idea is to have all WiFi settings in one place, but for a single SSID only. Multiple SSIDs can be configured afterwards, e.g. via `dietpi-config`. Opinions?

I did now try to replicate the serial console issue, first on Raspberry Pi Zero W, which has a dedicated WiFi/BT adapter, hence the primary `/dev/serial0` pints to `/dev/ttyS0` mini/software UART, and, could replicate the issue even there: Both serial getty's are running, even that the image ships with only one enabled, and indeed there is only one enabled:
```
# ls -al /etc/systemd/system/getty.target.wants/
total 8
drwxr-xr-x  2 root root 4096 Nov 21 15:15 .
drwxr-xr-x 13 root root 4096 Nov 21 15:16 ..
lrwxrwxrwx  1 root root   34 Nov 21 14:54 getty@tty1.service -> /lib/systemd/system/getty@.service
lrwxrwxrwx  1 root root   41 Nov 21 15:15 serial-getty@serial0.service -> /lib/systemd/system/serial-getty@.service
```
So `serial-getty@ttyS0.service` got started by something else. It's status shows `enabled-runtime;`, instead of `enabled;`.

I found an explanation now:
- We already know that systemd automatically enables the login getty on serial consoles used for kernel/boot messages, at least when it is the primary console to print to. This is the reason we do not only disable, but mask related services when those are disabled via `dietpi-config`.
- We do not use `ttyS0` or `ttyAMA0` in `/boot/cmdline.txt`, but `serial0` to be compatible with RPi models with and without onboard WiFi/BT on first boot, following Raspberry Pi OS as well.
- The problem is now that `serial0` and `serial1` are created via udev rules, depending on the detected RPi model (see `/etc/udev/rules.d/99-com.rules`), hence cannot be used by the kernel at boot, so it falls back to the first available serial console:
    ```
    [    2.623275] uart-pl011 20201000.serial: cts_event_workaround enabled
    [    2.623410] 20201000.serial: ttyAMA0 at MMIO 0x20201000 (irq = 81, base_baud = 0) is a PL011 rev2
    [    2.625935] bcm2835-aux-uart 20215040.serial: there is not valid maps for state default
    [    2.626707] printk: console [ttyS0] disabled
    [    2.626871] 20215040.serial: ttyS0 at MMIO 0x20215040 (irq = 53, base_baud = 50000000) is a 16550
    [    2.627133] printk: console [ttyS0] enabled
    ```
- Since we ship images with the related actual serial getty's only disabled, but not masked, the fallback actual getty service gets enabled automatically.
- This is something we should report to RPi devs as well. The `console=serial0,115200` kernel command-line argument is invalid as this device does not exist at this boot stage. DietPi converts it automatically to the actual serial console device as part of the firstrun setup, depending on the RPi model, but this doesn't help on first boot, of course.

However, the solution is pretty simply: We need to mask the two actual serial getty services in PREP. The correct one will be unmasked anyway as part of the firstrun setup serial0 => ttyS0/ttyAMA0 migration.

MichaIng, 2021-12-04:
Solved with: https://github.com/MichaIng/DietPi/commit/a1f64c8

ad7718, 2021-12-04:
That's great!

MichaIng, 2021-12-04:
All RPi images have been repacked to contain the symlinks, Cloudflare cache cleared, so they are ready to be downloaded.

The WiFi issue has been sorted (as of `dietpi.txt` + `dietpi-wifi.txt` which both need to be configured), right?

The free space check is still there until next DietPi release in December 11th, but can be manually removed via `sed -i '/G_CHECK_FREESPACE/d' /boot/dietpi/dietpi-software` or removing that line (near the end of the file) manually after flashing/before booting the image. Note that we officially support only SD cards of 4 GiB size or larger: https://dietpi.com/docs/install/#prerequisites
I wasn't aware that smaller ones are still around, I have some very old ones here, but all minimum 4 GiB 😄. With 2 GiB basically it works, but on RPi Zero with 512 MiB RAM, a swap file of size 1.5 GiB is tried to be created to have a reasonable minimum of 2 GiB total memory available, but this does not fit and hence fails on a 2 GiB card. However, it doesn't break anything else, you can manually create a smaller swap file when required or leave it disabled. There are some software installs which require more RAM during installation or compiling, just keep that in mind.

ad7718, 2021-12-04:
> The WiFi issue has been sorted (as of dietpi.txt + dietpi-wifi.txt which both need to be configured), right?
In default Raspbian I need to only create /boot/wpa_supplicant.conf with my country code and SSID/password and after that wifi works. In DietPi I need to add(or change, I don't remember) AUTO_SETUP_NET_WIFI_ENABLED=1 in dietpi.txt, change AUTO_SETUP_NET_WIFI_COUNTRY_CODE in dietpi.txt, enter SSID/password in dietpi-wifi.txt. So I need to change two different non-consequent sections in dietpi.txt and one in dietpi-wifi.txt which is not so obvious. I think it's better to have AUTO_SETUP_NET_WIFI_ENABLED enabled by default and maybe move country code in dietpi-wifi.txt for easier setup.

I don't remember how much swap was allocated, but I remember that after install I had only ~800 of disk space used and had no problem with that.

Joulinar, 2021-12-04:
It depends on usage of the system. We have a couple of software title inside the repository that require way more ram/swap than your entire system has :wink:

ad7718, 2021-12-04:
I understand, this is just my case. And of course there is software which will not fit in 1.5Gb swap + 512Mb RAM.

MichaIng, 2021-12-04:
We'll not have WiFi unconditionally enabled, but as mentioned above I agree that it shouldn't be required to edit two files, and we may think about using WiFi as fallback at least when SSID + key are configured and no Ethernet interface is available and plugged (no carrier signal, when trying to bring it up).

closed 2021-12-11 by MichaIng