MichaIng/DietPi #5019: Open Beta v7.9 | Please help testing and hardening the upcoming release

MichaIng, 2021-12-04:
RC version | v7.9.0
---------- | -------
v7.9.0 PR | #5017
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2021-12-11**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av7.9
____
### Beta v7.9.0
_(2021-12-04)_

#### Changes
- DietPi-Dashboard | On fresh installs, password protection is now enabled by default, using the global software password. You can apply or change this manually by following the instructions in our documentation: https://dietpi.com/docs/software/system_stats/#dietpi-dashboard
- DietPi-Dashboard | The default TCP network port has been changed from 8088 to 5252 to resolve a port conflict with InfluxDB. This does not affect existing installs, but you will be asked whether to apply this change during DietPi update. Many thanks to @blablazzz for reporting this issue: https://github.com/MichaIng/DietPi/issues/4966
- DietPi-Backup | A backup archive with a selectable amount of backups to keep can be created now. Backups are rotated automatically and if the maximum amount has been reached, the oldest backup is used as basis for the incremental new backup sync, to reduce writes and increase speed. Many thanks to `@johnvick` and many others for requesting this feature: https://dietpi.com/phpbb/viewtopic.php?t=3593
- DietPi-Backup | Backups can now be stored outside of /mnt into any directory or mount point as long as the filesystem supports symlinks and UNIX permissions.
- DietPi-Software | The file server choice menu has been removed. Most file servers can run concurrently, hence it is not required to remove e.g. a Samba server to install an FTP server, so that a dedicated menu lost its reason. File servers can be selected the regular way via "Browse/Search Software" menus or CLI. The related dietpi.txt setting has been removed for new images as well, but it will still be respected when present. For an automated install with new images, use the "AUTO_SETUP_INSTALL_SOFTWARE_ID" setting instead.
- DietPi-Software | SABnzbd: For fresh installs, file logging has been disabled in favour of journal logging. All service and process logs can hence now be reviewed via: journalctl -u sabnzbd
- DietPi-Software | Apache: Fresh installs and reinstalls will be configured with PHP-FPM instead of mod_php. As a requirement, the event MPM is used instead of prefork. This reduces memory usage and increases access performance significantly on concurrent requests, as the Apache parent process does not need to fork a new child process for every single process. We further optimised it by spawning a single static child process only while handling concurrent requests by a sufficient amount of process threads. This allows Apache to share memory efficiently and makes it quite lightweight. There are no downsides known when using a single process only, compared to multiple processes with less threads each. Related StackExchange questions: https://superuser.com/questions/1611015/apache2-mpm-event-more-threads-vs-more-processes
- DietPi-Software | Apache: Our default/base configuration is now added as separate file, so that the main apache2.conf is not touched anymore. Furthermore the default vhost is now pre-created before the package install, so that it can be skipped on a reinstall to not overwrite user customisations. These in combination allow for a safe and clean reinstall without breaking any changes done by the admin, with the little exception that the webroot is set to /var/www which is required for all our software options which make use of an external webserver.
- DietPi-Software | Apache: Our new default config provides maximum privacy settings and security headers. It is trivial to override these with own configurations, on vhost or directory level.
- DietPi-Software | Apache: Logging is now done to journal (journalctl -u apache2) by default and the ServerName directive added with the local IP, to mute related startup warnings. This may imply access and CORS failures when applications check for the server name as allowed hostname but a different external IPs/hostnames was used for access. In such case generally applications provide a way to define a list of permitted hostnames. Without a server name set, usually webserver simply apply the HTTP_HOST header, which bypasses every related check. Apache, according to the logged warning, seems to use 127.0.1.1 then.
- DietPi-Software | Kodi: On Raspberry Pi Bullseye systems, the new official Raspberry Pi repository build for Kodi 19.3 is now installed. You can apply the upgrade manually by reinstalling Kodi: dietpi-software reinstall 31
- DietPi-Software | Kodi: The addon repository is now installed by default with all Kodi installs, which previously was the case only on RPi and Odroids. When currently missing, it can be manually installed: apt install kodi-repository-kodi
- DietPi-Software | Gitea: The service runs now as dedicated user "gitea" with its home directory "/mnt/dietpi_userdata/gitea" to allow easy transfer and usage of SSH keys for remote access. This applies to newly installed or reinstalled Gitea instances. Many thanks to @LilTrublMakr for reporting the related limitation with the previously used "dietpi" user: https://github.com/MichaIng/DietPi/issues/4620

#### Fixes
- Raspberry Pi | Resolved an issue in our images where on first boot two serial login consoles on the generic symlinked and actual serial devices could have been started. This doubled inputs and in turn broke successful username and password login via serial console on first boot. Many thanks to @ad7718 for reporting this issue: https://github.com/MichaIng/DietPi/issues/5014 
- DietPi-Backup | Resolved an issue where backup and restore failed if a non-default backup location is used, as a wrong log file path was used. This is a v7.8 regression. Many thanks to `@Malinka` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?p=39909#p39909
- DietPi-Software | Resolved a v7.8 regression where ReadyMedia, Deluge, Sonarr and Jellyfin installs failed with an error on "usermod", since the services were not stopped first. This has been loved via live patches for v7.8 as well.
- DietPi-Software | Transmission: Resolved a v7.8 regression where on fresh installs the intended configuration was not deployed. Many thanks to `@kannz` and `@alessandro.psrt` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=9567, https://dietpi.com/phpbb/viewtopic.php?t=9683
- DietPi-Software | SABnzbd: Resolved an issue where installs failed on ARMv6 and ARMv7 Stretch systems. Many thanks to @bensp for reporting this issue: https://github.com/MichaIng/DietPi/issues/4997

Gurky-Kronos, 2021-12-06:
Having issues installing Software due to Sonarr and Mono having no releases for Bullseye. Keep in mind I was trying to reinstall Deluge due to the webui not loading after the update. 

#### Details:
- Date           | Mon Dec  6 10:47:45 EST 2021
- DietPi version | v7.9.0 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Debian mini.iso
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPi 5.10.0-9-amd64 #1 SMP Debian 5.10.70-1 (2021-09-30) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Update to Bullseye windo sonarr and mono installed
2. Try installing anything. 

#### Expected behaviour:

- It should install the software you chose. 

#### Actual behaviour:

- Getting error prompts because Sonarr and Mono
#### Extra details:

- I know this is to be expected due to the article but should there be a way to check for Buster instead of Bullseye?

#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://downloads.plex.tv/repo/deb public InRelease
Hit:4 https://apt.ombi.app/develop jessie InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Ign:7 https://download.mono-project.com/repo/debian bullseye InRelease
Err:8 https://download.mono-project.com/repo/debian bullseye Release
  404  Not Found [IP: 72.21.81.200 443]
Ign:9 https://apt.sonarr.tv/debian bullseye InRelease
Err:10 https://apt.sonarr.tv/debian bullseye Release
  404  Not Found [IP: 37.187.255.43 443]
Reading package lists...
E: The repository 'https://download.mono-project.com/repo/debian bullseye Release' does not have a Release file.
E: The repository 'https://apt.sonarr.tv/debian bullseye Release' does not have a Release file.

MichaIng, 2021-12-06:
As described in our Bullseye upgrade blog post, Mono and Sonarr repositories do not provide dedicated Bullseye packages yet, but the Buster packages work just fine:
```sh
sed -i 's/bullseye/buster/' /etc/apt/sources.list.d/*mono*.list /etc/apt/sources.list.d/*sonarr*.list
apt update
```

MichaIng, 2021-12-06:
### Beta v7.9.1 has been merged: #5028

It fixes and re-enables Deluge on Raspberry Pi ARMv6 userland (Raspbian based) systems and contains some minor code and whiptail menu test and visual polishing. Ah further preparations for F2FS images have been done. At least for Raspberry Pi we will provide some for testing soon, stay tuned 🙂.

MichaIng, 2021-12-10:
### Beta v7.9.2 has been merged: #5037

Some old bugs have been solved:
- On RPi 1 and Zero and incompatible UnRAR was installed, which has been corrected now.
- With Apache webserver, the rTorrent RPC proxy was not configured correctly. Newly installed and reinstalled rTorrent instances will have this solved now.
- The Odroid 3.5" LCD panel was not configured correctly as it requires SPI to be actively disabled, at least on our old Odroid C2 images with Linux v3.16.85.
- On RPi hardware accelerated Chromium video playback as been enabled for additional patented media formats, including H.264. This can be done manually: `apt install chromium-codecs-ffmpeg-extra` (RPi only)

MichaIng, 2021-12-11:
### Beta v7.9.3 has been merged: #5052

Preparing the final release for the next hours.

MichaIng, 2021-12-11:
### DietPi v7.9 has been released: #5053

Many thanks to all testers ❤️!

closed 2021-12-11 by MichaIng