MichaIng/DietPi #5020: `sudo reboot` hangs Radxa Zero

shytikov, 2021-12-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | ```G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'```
- Distro version | `bullseye`
- Kernel version | `Linux vizsla 5.10.69-7-amlogic-gfd159ba07d5c #gfd159ba07d5c SMP PREEMPT Thu Nov 4 14:38:15 CST 2021 aarch64 GNU/Linux`
- SBC model | `Radxa Zero (aarch64)`
- Power supply used | regular 5V 2A
- SDcard used | run from eMMC

#### Additional Information (if applicable)
- Software title | bare installation
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | `f8336972-44bc-4767-894f-9db656937162`

#### Steps to reproduce

1. run `sudo reboot`

#### Expected behaviour

- SBC should reboot and connect to WiFi and start responding on SSH calls

#### Actual behaviour

- SBC closes SSH connection, disappears from WiFi clients lists, never goes up again

#### Extra details

- Physically unpluging and plugging SBC again, resets it and it goes up again

shytikov, 2021-12-05:
I do have headless install, so I cannot see any additional clues, but if you can help me how I can push diagnostic logs to you, I would gladly help!

Joulinar, 2021-12-05:
I guess is same issue as this https://github.com/MichaIng/DietPi/issues/4969

MichaIng, 2021-12-05:
Yes, I'll mark it as closed as duplicate. This needs to be resolved bootloader wise. I'm still waiting for our samples 🤔, when having those, I may be able to do more testing/debugging with this.

There are instructions about how to compile and install a fixed U-Boot manually: https://forum.radxa.com/t/instruction-of-how-to-fix-2-seconds-delay-at-boot-and-not-working-reboot/7898
I hope Radxa will ship U-Boot packages soon, otherwise I'll try to get it working with these instructions once the samples are here and update our images accordingly.

closed 2021-12-05 by MichaIng