MichaIng/DietPi #5029: E: The repository 'https://download.docker.com/linux/debian 11 Release' does not have a Release file.

poggingfish, 2021-12-05:
#### Details:
- Date           | Sun Dec  5 22:00:36 GMT 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Install a package from dietpi-software
#### Expected behaviour:

- It should get a release file from https://download.docker.com/linux/debian.
#### Actual behaviour:

- It is failing when trying to get a release file from https://download.docker.com/linux/debian.
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Ign:3 https://download.docker.com/linux/debian 11 InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Err:6 https://download.docker.com/linux/debian 11 Release
  404  Not Found [IP: 13.226.15.124 443]
Hit:7 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
E: The repository 'https://download.docker.com/linux/debian 11 Release' does not have a Release file.
```

Joulinar, 2021-12-05:
Hi,

I guess you docker source file is incorrect. Did you created it yourself or did you used `dietpi-software` to install docker? Can you share following

```
cat /etc/apt/sources.list.d/docker.list
```

poggingfish, 2021-12-05:
No such file or directory

poggingfish, 2021-12-05:
And im fairly sure i used dietpi-software.

Joulinar, 2021-12-05:
can you check where docker is defined as source

```
grep -r docker /etc/apt/
```

poggingfish, 2021-12-05:
```
/etc/apt/sources.list:deb [arch=arm64] https://download.docker.com/linux/debian 11 stable
/etc/apt/sources.list:# deb-src [arch=arm64] https://download.docker.com/linux/debian 11 stable
grep: /etc/apt/trusted.gpg: binary file matches

```

Joulinar, 2021-12-05:
ok this was not done by `dietpi-software` as we create an own source list file.

https://github.com/MichaIng/DietPi/blob/561f7a7724e51623742a016f12c03bef66c235f3/dietpi/dietpi-software#L11554-L11556

pls remove the 2 docker lines from /etc/apt/sources.list and create the an own one for docker

```
eval echo 'deb https://download.docker.com/linux/debian/ bullseye stable' > /etc/apt/sources.list.d/docker.list
apt update
```

poggingfish, 2021-12-05:
That seemed to fix it. Thanks alot!

closed 2021-12-05 by poggingfish