MichaIng/DietPi #5030: Wifi hotspot not providing internet on pi 4

poggingfish, 2021-12-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model | RPI 4 4GB
- SSD used: Generic 32GB SSD

#### Additional Information (if applicable)
- Software title | wifi hotspot + tor hotspot
- Was the software title installed freshly or updated/migrated? It was just installed.
- Can this issue be replicated on a fresh installation of DietPi? Possibly.

#### Steps to reproduce

1. Download wifi hotspot
2. Connect

#### Expected behaviour

- I should get internet connection via tor

#### Actual behaviour

- I am getting no internet connection at all.

MichaIng, 2021-12-06:
Many thanks for your report.

The iptables rules are setup for IPv4 only. For the WiFi Hotspot this is easy to extend to IPv6 but for the Tor Hotspot I guess we'd need to enable an IPv6 listener as well. To verify that this is the issue, can you try to connect to an exclusive IPv4 host, like `github.com`?

Please also show the active iptables rules:
```sh
iptables -n -L
iptables -t nat -n -L
```
_You can mask any identifying IPs, when present, but there should be none._

poggingfish, 2021-12-06:
fixed itself.

closed 2021-12-06 by poggingfish

Joulinar, 2021-12-06:
Maybe you like to share how you fixed it 😉

poggingfish, 2021-12-06:
I dont know how i fixed it. i only know it fixed itself haha

MichaIng, 2021-12-06:
😄 probably an intermediate issue with the service state or so. There are a lot of services involved, the DHCP server, hostapd, Tor and then interface and sysctl forwarding settings. However, good that it works now.