MichaIng/DietPi #5034: Docker Compose not working

CommandLineN00b, 2021-12-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | 5V 3.5A CanaKit
- SDcard used | SanDisk ultra 128GB

#### Additional Information (if applicable)
- Software title | Docker Compose
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes, re-flashed and tried again and same issue
- Bug report ID | a291b7c7-a12e-439a-8bd9-c7dea82b9775

#### Steps to reproduce

1. sudo curl -L "https://github.com/docker/compose/releases/download/1.29.2/docker-compose-$(uname -s)-$(uname -m)" -o /usr/local/bin/docker-compose
2. sudo chmod +x /usr/local/bin/docker-compose
3. docker-compose --version

#### Expected behaviour

- I should get: docker-compose version 1.29.2, build 1110ad01

#### Actual behaviour

- ... I am getting this returned: /usr/local/bin/docker-compose: line 1: Not: command not found

#### Extra details
- my name is not a joke... I am completely new to this and have almost no idea what I am doing... go easy on me.

Joulinar, 2021-12-06:
pls use `dietpi-software` to install Docker Compose.

MichaIng, 2021-12-06:
Many thanks for your report.

Note that this is not the method we use in `dietpi-software` for installing Docker Compose, as we install the Python module via `pip`. I had a look into this file, it has an ELF header. It is Docker Compose v2 while the Python module is v1. Looks like the new version is not Python, but a written in Go.

Go binaries should be self-contained, so not sure what is missing. The "not found" error can indeed happen when the binary is for a foreign architecture. Can you verify that you are indeed using an ARMv8 userland image and did not manually switch to the aarch64 kernel?
```sh
dpkg --print-architecture
```
And try to download the correct binary explicitly:
```sh
curl -fL 'https://github.com/docker/compose/releases/download/v2.2.2/docker-compose-linux-aarch64' -o /usr/local/bin/docker-compose
chmod +x /usr/local/bin/docker-compose
```
Assure that there are not error messages, e.g. the `/usr/local/bin` directory may not exist.

Joulinar, 2021-12-06:
@MichaIng 
not sure how big the change on v2 is but Docker guys offer a tool for backward compatibility with docker-compose v1.x https://github.com/docker/compose-switch

MichaIng, 2021-12-06:
The migration script does... moreless nothing. It downloads the Go binary like above and sets up `update-alternatives` so that one can switch between v1 and v2. Through this means to use and understand `update-alternatives` 😄. There is no config migration or so done, so it seems to be backwards compatible.

When we do the migration, we shouldn't force it but inform users that a reinstall will apply it. I'd then simply uninstall the Python module and install the standalone binary 🙂.

Joulinar, 2021-12-06:
or we just apply v2 on new installations only and don't force a migration?

MichaIng, 2021-12-06:
I don't want to force it on a DietPi upgrade, but I also don't want to add/keep conditional Python 3 dependency and code in `dietpi-software`. So if everyone got interactively informed/warned during DietPi update, I think it should be fine to do the migration automatically on a  reinstall. Since Docker Compose v1 can be updated via `pip`, there is no reason why someone would do a reinstall as long as the instance is not somehow broken. So I think in such case it's fine to migrate.

CommandLineN00b, 2021-12-06:
simply using the dietpi-software feature worked like a charm. sorry for the dumb question, but really appreciate the immediate responses, and for going easy on me :)

MichaIng, 2021-12-06:
Great. But thanks back for opening this issue, since I wasn't aware of Docker Compose v2 at all 😄. This should be more lightweight and cause less long-term issues with e.g. distro upgrades, as it does not depend on any external runtime, so I'm happy to implement it with DietPi v8.0 (since v7.9 is in beta already).

CommandLineN00b, 2021-12-07:
Glad my extreme ignorance led to a useful discovery :) i'll keep asking stupid questions!

closed 2022-02-11 by MichaIng