MichaIng/DietPi #5035: started fail2ban says "masked"

jalsco, 2021-12-07:
tag as "question" please. 

Trying to start (or restart) fail2ban - says "masked" - so now what ? 
```
[Tue Dec 07] root@batopi: /$ systemctl restart fail2ban
Failed to restart fail2ban.service: Unit fail2ban.service is masked.
[Tue Dec 07] root@batopi: /$
```
Seems like it is not running as `fail2ban-client` does nothing

Joulinar, 2021-12-07:
You can try following

```
systemctl unmask fail2ban
systemctl start fail2ban
```

jalsco, 2021-12-07:
```
[Tue Dec 07] root@batopi: /$ systemctl unmask fail2ban
Removed /etc/systemd/system/fail2ban.service.
[Tue Dec 07] root@batopi: /$ systemctl start fail2ban
Failed to start fail2ban.service: Unit fail2ban.service not found.
```

jalsco, 2021-12-07:
I had always assumed that fail2ban was running. But it seems like it was never running. It is installed... I do not know how to start it ??

MichaIng, 2021-12-07:
Very strange indeed, not only that it was masked but also that then no actual service is present. Probably it got autoremoved during the Buster => Bullseye upgrade you did, e.g. due to the upgrade from Python 3.7 to Python 3.9 🤔. Please reinstall it:
```sh
dietpi-software reinstall 73
```
This might be something we need to add to the reinstall list on our blog post.

jalsco, 2021-12-07:
whoa - that seems like it is re-installing a lot of stuff ! like ... everything 😱

MichaIng, 2021-12-07:
It should only install (when missing) the two packages `python3-systemd` and `fail2ban`. Was the distro upgrade from Buster to Bullseye probably incomplete? Did you follow the guide [here](https://dietpi.com/blog/?p=811)?

jalsco, 2021-12-07:
[image]

jalsco, 2021-12-07:
I checked that: mariadb and lighttpd and php7.4 - all still working 😉✔
and also:  caddyserver 😁✔

and **YES**  now the `fail2ban-client` command is working 

Thank you so much !

closed 2021-12-07 by jalsco