MichaIng/DietPi #5036: NordVPN is not starting at all

Martin-Fox, 2021-12-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`buster 1
`
- Kernel version | `uname -a`
`Linux DietPi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 3 Model B (armv7l)
`

#### Additional Information (if applicable)
- Software title | NordVPN
- Bug report ID | `3f4c727c-83d6-4150-9daf-ec5f1834accc`

```
root@DietPi:~# systemctl status dietpi-vpn.service
● dietpi-vpn.service - VPN Client (DietPi)
   Loaded: loaded (/etc/systemd/system/dietpi-vpn.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-12-06 16:42:39 GMT; 14s ago
  Process: 8416 ExecStart=/usr/sbin/openvpn --suppress-timestamps --nobind --config /etc/openvpn/client.ovpn (code=exited, stat
us=203/EXEC)
 Main PID: 8416 (code=exited, status=203/EXEC)

Dec 06 16:42:39 DietPi systemd[1]: Starting VPN Client (DietPi)...
Dec 06 16:42:39 DietPi systemd[8416]: dietpi-vpn.service: Failed to execute command: No such file or directory

Dec 06 16:42:39 DietPi systemd[8416]: dietpi-vpn.service: Failed at step EXEC spawning /usr/sbin/openvpn: No su
ch file or directory
Dec 06 16:42:39 DietPi systemd[1]: dietpi-vpn.service: Main process exited, code=exited, status=203/EXEC
Dec 06 16:42:39 DietPi systemd[1]: dietpi-vpn.service: Failed with result 'exit-code'.
Dec 06 16:42:39 DietPi systemd[1]: Failed to start VPN Client (DietPi).
root@DietPi:~#

```
#### Steps to reproduce
1. Launch dietpi-launcher
2. Select dierpi-vpn
3. Select NordVPN
4. Select Apply

#### Expected behaviour
Connection should be established and VPN should be active

#### Actual behaviour

```
root@DietPi:~# dietpi-launcher 
[ INFO ] DietPi-VPN | Checking for required APT packages: openvpn
[  OK  ] DietPi-VPN | All required APT packages are already installed.
[  OK  ] DietPi-VPN | curl -sSfLO https://downloads.nordcdn.com/configs/archives/servers/ovpn.zip
[  OK  ] DietPi-VPN | mkdir -p /etc/openvpn/nordvpn
[  OK  ] DietPi-VPN | unzip -o ovpn.zip -d /etc/openvpn/nordvpn
[  OK  ] DietPi-VPN | rm ovpn.zip
[ INFO ] DietPi-VPN | Populating VPN server list, please wait...
[  OK  ] DietPi-VPN | systemctl enable dietpi-vpn
[  OK  ] DietPi-VPN | chmod +x /var/lib/dietpi/dietpi-vpn/static_up.sh /var/lib/dietpi/dietpi-vpn/static_down.sh
[  OK  ] DietPi-VPN | umask 0077
[ INFO ] DietPi-VPN | Generating OVPN file, please wait...
[  OK  ] DietPi-VPN | cp -f /etc/openvpn/nordvpn/ovpn_udp/pl160.nordvpn.com.udp.ovpn /etc/openvpn/client.ovpn
[  OK  ] DietPi-VPN | umask 0022
[  OK  ] DietPi-VPN | chmod 0600 /var/lib/dietpi/dietpi-vpn/settings_ovpn.conf /var/lib/dietpi/dietpi-vpn/settings_dietpi.conf /etc/openvpn/client.ovpn
[  OK  ] DietPi-VPN | chown root:root /var/lib/dietpi/dietpi-vpn/settings_ovpn.conf /var/lib/dietpi/dietpi-vpn/settings_dietpi.conf /etc/openvpn/client.ovpn
[  OK  ] DietPi-VPN | Setting in /etc/openvpn/client.ovpn adjusted: auth-user-pass /var/lib/dietpi/dietpi-vpn/settings_ovpn.conf
[  OK  ] DietPi-VPN | Added setting script-security 2 to /etc/openvpn/client.ovpn after line auth SHA512
[  OK  ] DietPi-VPN | Added setting up /var/lib/dietpi/dietpi-vpn/static_up.sh to /etc/openvpn/client.ovpn after line script-security 2
[  OK  ] DietPi-VPN | Added setting down /var/lib/dietpi/dietpi-vpn/static_down.sh to /etc/openvpn/client.ovpn after line up /var/lib/dietpi/dietpi-vpn/static_up.sh
[  OK  ] DietPi-VPN | sed -i /^[[:blank:]]*route-up[[:blank:]]/d /etc/openvpn/client.ovpn
[  OK  ] DietPi-VPN | sed -i /^[[:blank:]]*route-pre-down[[:blank:]]/d /etc/openvpn/client.ovpn
[FAILED] DietPi-VPN | systemctl restart dietpi-vpn
[FAILED] DietPi-VPN | systemctl restart dietpi-vpn

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Mon Dec  6 16:41:19 GMT 2021
- DietPi version | v7.7.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `systemctl restart dietpi-vpn`
- Exit code      | 1
- Software title | DietPi-VPN
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
```
Job for dietpi-vpn.service failed because the control process exited with error code.
See "systemctl status dietpi-vpn.service" and "journalctl -xe" for details.
```
---------------------------------------------------------------------
[FAILED] DietPi-VPN | Unable to continue, DietPi-VPN will now terminate.

- Press any key to return to DietPi-Launcher...

```

Joulinar, 2021-12-07:
The system checked the availablity of the `openvpn` package and it seems to be installed. However the executable is missing still. Can you try to reinstall OpenVPN

```
apt install --reinstall openvpn
```

Martin-Fox, 2021-12-07:
It works!!

closed 2021-12-07 by Martin-Fox