MichaIng/DietPi #5038: Tor hotspot no internet (however vanilla wifi hotspot works)

vielka442, 2021-12-07:
Using the latest diet pi image on a Raspberry pi 3, with a usb wifi dongle attached, I installed the wifi hotspot and everything worked great. I tested by using another raspberry pi with diet pi and used the internet test and everything was good.

When I then installed the Tor hotspot and repeated the test, while the wifi was connected, no internet connection was detected and it failed the test.

The hotspot on the pi 3 was connected to my router via ethernet. The other pi 4 was connected only via the wifi created by the wifi hotspot or the tor hotspot. It only ever works when I install only the wifi hotspot, so I know the hardware is working ok.

the other thing is I use another pi connected to the router, which runs pi-hole. The router is at 192.168.1.1 and the pi-hole is at 192.168.1.10 and the router as this ip set as the DNS server. Should I be using any of these numbers in the Tor configuration? again there was no issue with this setup when it was simply the wifi hotspot.

What can I do to try and debug?
Thank you for your time and for developing this amazing collection of software.

MichaIng, 2021-12-07:
Many thanks for your report.

So the internet connection from the Tor Hotspot server itself was broken? Did you try to connect to a pure IPv4 host?
```sh
curl -IL https://github.com/
```
From what I see we configure only IPv4 routing rules, this needs to be expanded to cover both, IPv4 and IPv6.

Can you also check the Tor logs:
```sh
journalctl -u tor
```

vielka442, 2021-12-08:
Thank you for the response @MichaIng

Using Diet Pi internet connectivity test on the connected pi, I tired the default connection to:
https://dietpi.com then tired https://google.com and https://apple.com

I never considered IPv4 vs 6 hosts. I did however also try turning off IPV6 on the connected pi, and it is off on my router and pi-hole, but might not be on the hotspot.

I will try the commands you suggested tomorrow and report back.
Thank you.

vielka443, 2021-12-12:
Hi @MichaIng just a follow up on this thread as I have ran the commands you requested on a fresh install.

I was able to get a successful connection on the connected Pi running DietPi via an IPv4 host as you suspected. But I was unable to get a successful IPV6 connection, resulting in an inconsistent experience.

I also turned on every option for IPV6 on my router + on the attached Pi-Hole, as well as testing with those options off.

Below are the results of the commands run on the pi attached to the router, running the tor hotspot. 

Thanks for investigating.

`**root@DietPi:~# journalctl -u tor**`

-- Journal begins at Sat 2021-12-10 17:46:00 GMT, ends at Sat 2021-12-10 19:48:00 GMT. --
Dec 10 17:46:40 DietPi systemd[1]: Starting Anonymizing overlay network for TCP (multi-instance-master)...
Dec 10 17:46:40 DietPi systemd[1]: Finished Anonymizing overlay network for TCP (multi-instance-master).
root@DietPi:~# 

`**root@DietPi:~# curl -IL https://github.com/**`

HTTP/2 200 
server: GitHub.com
date: Sat, 11 Dec 2021 20:00:00 GMT
content-type: text/html; charset=utf-8
vary: X-PJAX, X-PJAX-Container, Accept-Language, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
content-language: en-US
etag: W/"66f7ebb47da1643d49f1b389f0c07e4c"
cache-control: max-age=0, private, must-revalidate
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: origin-when-cross-origin, strict-origin-when-cross-origin
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com github.githubassets.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com customer-stories-feed.github.com spotlights-feed.github.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/ github.githubassets.com; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
set-cookie: _gh_sess=xxx; Path=/; HttpOnly; Secure; SameSite=Lax
set-cookie: _octo=xxx; Path=/; Domain=github.com; Expires=Sun, 11 Dec 2022 20:16:00 GMT; Secure; SameSite=Lax
set-cookie: logged_in=no; Path=/; Domain=github.com; Expires=Sun, 11 Dec 2022 20:10:00 GMT; HttpOnly; Secure; SameSite=Lax
accept-ranges: bytes
x-github-request-id: xxx

MichaIng, 2021-12-12:
Okay, many thanks for the feedback. I'll have a closer look the next days to get IPv6 support enabled.

vielka443, 2021-12-12:
No problem, many thanks to you.

MichaIng, 2022-02-04:
First step, all `iptables` rules are now applied for IPv6 as well: https://github.com/MichaIng/DietPi/commit/4fb93c6
Testing now whether for Tor there is an additional step required.

MichaIng, 2022-02-05:
Okay final implementation requires adding an IPv6 range to the DHCP server and a static IPv6 IP for the system. I'll postpone this to next release since this will include a network rework with WiFi Hotspot becoming an option in the network configuration script instead of a software option. There the IPv6 range will be added automatically when generally enabled.

What I however tested is that DHCP clients currently do not get any IPv6 address (aside of the LLA they apply to themselves) and hence all network requests are done via IPv4 automatically and nothing is bypassing Tor. One can of course force IPv4 via `curl -6` or `ping -6` but this simply times out without a non-LLA IPv6 address.