MichaIng/DietPi #5040: Network problems (problems with 9.9.9.9?)

lupa18, 2021-12-07:
#### Details:
- Date           | Tue Dec  7 21:11:17 GMT 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCK64 (aarch64) (ID=43)
- Kernel version | `Linux rock64 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT 4e944f81-8f74-4ca3-b2ee-8670d12fca49.7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:22/bugreport/`
- Exit code      | 28
- Software title | DietPi-BugReport
#### Steps to reproduce:

1.Not sure
#### Expected behaviour:

- Having all network services working
#### Actual behaviour:

- Network problems are causing lot of other problems: with homeassitant, mpd, etc..
#### Extra details:

- Ping to 9.9.9.9 not working but to 1.1.1.1 yes
#### Additional logs:
```
* Resolving timed out after 8000 milliseconds
* Closing connection 0
curl: (28) Resolving timed out after 8000 milliseconds
* Resolving timed out after 8000 milliseconds
* Closing connection 1
curl: (28) Resolving timed out after 8000 milliseconds
```

MichaIng, 2021-12-07:
Many thanks for your report.

You can also increase the network test timeout, or check whether Quad9 DNS is reachable at all and how long it takes:
```sh
time ping -c 1 9.9.9.9
time curl -I https://ssh.dietpi.com/
```

How are you connected, via Ethernet or WiFi? Shouldn't make much difference, but did you try to switch from DHCP to static IP?

lupa18, 2021-12-07:
Hi @MichaIng and thanks!!

```
$ time ping -c 1 9.9.9.9
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.

--- 9.9.9.9 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms

real	0m10,003s
user	0m0,001s
sys	0m0,002s
```

and:

```
$ time curl -I https://ssh.dietpi.com/
HTTP/2 301 
x-content-type-options: nosniff
x-frame-options: SAMEORIGIN
x-xss-protection: 1; mode=block
x-robots-tag: all
x-download-options: noopen
x-permitted-cross-domain-policies: none
referrer-policy: no-referrer-when-downgrade
expect-ct: enforce, max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
content-security-policy: upgrade-insecure-requests; frame-ancestors 'self'; form-action 'self'; base-uri https://dietpi.com/matomo/index.php https://dietpi.com/grafana/; default-src 'none'; object-src 'none'; style-src 'self' 'unsafe-inline' https://fonts.googleapis.com; font-src 'self' https://fonts.gstatic.com data:; img-src * data:; script-src 'self' 'unsafe-inline' 'unsafe-eval' https://unpkg.com https://google.com/recaptcha/ https://www.google.com/recaptcha/ https://www.gstatic.com/recaptcha/; frame-src 'self' https://www.youtube-nocookie.com https://google.com/recaptcha/ https://www.google.com/recaptcha/; manifest-src 'self'; connect-src 'self' https://api.github.com
permissions-policy: accelerometer=(), autoplay=(), camera=(), document-domain=(), encrypted-media=(), fullscreen=(self "https://www.youtube-nocookie.com"), geolocation=(), gyroscope=(), magnetometer=(), microphone=(), midi=(), payment=(), usb=(), screen-wake-lock=()
cache-control: public, max-age=86399
strict-transport-security: max-age=31536000; includeSubDomains; preload
location: https://dietpi.com/
content-type: text/html; charset=iso-8859-1
date: Tue, 07 Dec 2021 21:27:25 GMT
server: Apache

real	0m0,759s
user	0m0,029s
sys	0m0,000s
```

lupa18, 2021-12-07:
I connect via Wifi. 
I could try to switch to static IP... should I use dietpi-config or do it manual?

Joulinar, 2021-12-07:
If a simple ping to Quad9 failed, it might be blocked somewhere. Maybe the ISP? Better to use a different DNS / IP check.

MichaIng, 2021-12-07:
10 seconds for a simple ping to a unicast IP is indeed very long. Connection to our server is fast, but this was the HTTP(S) server, not the SFTP server, so we may try to increase the timeout for that one.

You can apply a static IP manually as well, if you know how to do it in `/etc/network/interfaces`, otherwise use `dietpi-config`, where you can copy the currently applied IP/netmask/gateway to turn it into static IP info.

lupa18, 2021-12-07:
I was with static IP already. 

The problem was solved changing to cloudflare

The weird is because of this, mpd and homeassistant wasn't working well. HA couldn't configure several subsistems, and all home control become a mess.

MichaIng, 2021-12-07:
Okay great. Probably HA and MPD had issues when resolving hostnames themselves. The surely use the system resolver as well.

closed 2021-12-07 by MichaIng