MichaIng/DietPi #5041: Chromium Camera Support buggy

belveder79, 2021-12-08:
### Creating a bug report/issue

#### Required Information
- DietPi version:
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
```
- Distro version: bullseye 0
- Kernel version: Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model: RPi 3A+
- Power supply used: different ones
- SDcard used: SanDisk extreme 16GB

#### Additional Information (if applicable)
- Chromium
- fresh image 7.8.2 and dietpi-software/dietpi-config setup
- should be reproduced

#### Steps to reproduce

- Start Chromium in Kiosk mode
- Go to a website using a camera
- Grant permissions

#### Expected behaviour

- See camera working

#### Actual behaviour

- Camera works only for the first time
- Does not work any more at any time until you reset permissions and grant again

#### Extra details

- tried it with RPi cam and any USB cam - same behaviour
- not sure if dietpi uses a custom Chromium port or not, however, it works in Firefox (but that one does not have a Kiosk mode)

MichaIng, 2021-12-08:
Many thanks for your report.

> not sure if dietpi uses a custom Chromium port or not, however, it works in Firefox (but that one does not have a Kiosk mode)

It's the build from the Raspberry Pi repository that is also pre-installed on Raspberry Pi OS.

Please check `/root/.chromium-browser.init` respectively `/etc/chromium.d/custom_flags`, there is a flag `--temp-profile` inside which at least sounds suspicious. Probably removing it helps.

belveder79, 2021-12-09:
unfortunately that did not solve it, although I admit it looks suspicious (like some other flags)

will continue investigating, but can anyone reproduce this e.g. with [this webpage](https://webcamtests.com/viewer)?

MichaIng, 2021-12-09:
I will try to replicate, also compare with Chromium based browsers on Windows to rule out that it's a Chromium upstream issue.

MichaIng, 2022-04-22:
I haven't tested on RPi but on VM: On https://www.greencheck.gv.at/ I do see my camera image after hitting "Allow" once, reloading the page does not hide it. After 60 seconds without focus, the browser tab goes into energy savings mode, then I need to hit "continue" to see the camera image again. Also after reboot, the image is shown immediately, hence the information to allow camera access for this website was stored persistently.

With "until you reset permissions and grant again" you mean that you need to "Allow" camera access for this website again? So far unable to replicate.

MichaIng, 2022-05-27:
I'll mark this as closed, feel free to reopen when the issue persists.

closed 2022-05-27 by MichaIng