MichaIng/DietPi #5046: beta branch issue

KalixtShawxo, 2021-12-10:
So I followed the instructions from here:
https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

`root@Rock64plus:/# G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /boot/dietpi.txt
[  OK  ] G_CONFIG_INJECT | Setting in /boot/dietpi.txt adjusted: DEV_GITBRANCH=beta
root@Rock64plus:/# `

I did the update to an existing install on my Rock64+ 4gb SBC version of DietPi (clean install from earlier in the day)
and I tried to do the "dietpi-update"  i get the following:
` DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/beta/.update/version
curl: (6) Could not resolve host: raw.githubusercontent.com
[FAILED] DietPi-Update | Unable to get latest version.
[ INFO ] DietPi-Update | Please check the target Git repository and your network connection, then rerun dietpi-update.
[ INFO ] DietPi-Update | If this error persists, please report at: https://github.com/MichaIng/DietPi/issues
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://ftp.us.debian.org/debian bullseye InRelease
Hit:2 http://ftp.us.debian.org/debian bullseye-updates InRelease
Hit:3 http://ftp.us.debian.org/debian bullseye-backports InRelease
Hit:4 http://armbian.tnahosting.net/apt bullseye InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | No APT upgrades were found, not creating file: /run/dietpi/.apt_updates
root@Rock64plus:/#`

so there appears to be a problem with "raw.githubusercontent.com"? 
I tried it 3 times no issues with internet connectivity. 

any suggestions?  or os the Beta branch not applicable to my installation?

thanks in advance.

Joulinar, 2021-12-10:
This seems to be a local issue to resolve the DNS name `raw.githubusercontent.com`. Do you use any custom DNS server? Or AdBlock?

KalixtShawxo, 2021-12-10:
well, I do have a pi-hole installed on my network.  I wouldn't think that would stop it.... but as I kept trying I actually got it to go through after about the 10th try of going back and from beta to master and retrying dietpi-update . 

inside dietpi-update it kept getting errors but retrys would then go. eventually it completed
` OK  ] DietPi-Patch | Patched to DietPi v7.9
[ INFO ] DietPi-Update | APT autopurge, please wait...
[  OK  ] DietPi-Update | APT autopurge
[  OK  ] DietPi-Update | Incremental patching to v7.9.1 completed
[ INFO ] DietPi-Update | Checking for new available live patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Completed

[ INFO ] DietPi-Update | Current version : v7.9.1
[ INFO ] DietPi-Update | Latest version  : v7.9.1
[  OK  ] DietPi-Survey | Sending survey data
[ SUB1 ] DietPi-Services > restart 
[  OK  ] DietPi-Services | restart : cron
root@Rock64plus:/# `

so I guess it's a non-issue. or at least a minor issue at this point.   
will try out the beta and hopefully be of some help to the cause. 

thanks.

Joulinar, 2021-12-10:
The issue is not with DietPi beta. It's an issue with your DNS resolution. Check within Pihole logs how DNS resolution was done for `raw.githubusercontent.com`.

KalixtShawxo, 2021-12-10:
sorry, poor wording on my part. did not mean it was an issue with DietPi after all. a minor issue on my part. 
looking into the issue, Pi-hole is in the green. 
`<html>
<body>

21-12-10 02:32:32 | A | raw.githubusercontent.com | 192.168.2.22 | OK (cache) | IP (0.5ms) | Blacklist
-- | -- | -- | -- | -- | -- | --
2021-12-10 02:32:32 | AAAA | raw.githubusercontent.com | 192.168.2.22 | OK (cache) | NODATA (0.4ms)

</body>
</html>`
of course that is now....so I have no idea what the issue was an hour ago, the logs don't show but the last 100 entries.

KalixtShawxo, 2021-12-10:
I'll close this out, since it's not a Diet-Pi issue.

closed 2021-12-10 by KalixtShawxo

MichaIng, 2021-12-10:
By default you should have 48 hours of Pi-hole query logs when installed via `dietpi-software` and AFAIK a year when using their installer directly. These logs are the right place to look at when facing DNS issues again.

Another general recommendation: It is not required and not recommended to use Pi-hole for DNS by the Pi-hole host system itself. Usually you do not run a web browser on this host server system where blocking ads would be required, and it implies the risk that the server is sawing off the branch on which it is sitting when doing updates or other maintenance tasks or when Pi-hole fails for other reasons. So unless you have a specific reason, I suggest to use an upstream DNS directly 🙂.

KalixtShawxo, 2021-12-10:
Ok. I really appreciate the info and advice. 
I do not have the Pi-hole on the same unit I was putting the beta branch on. 
The Pi-hole is on a RasPi Zero separate from the Rock64 I was putting the dietpi beta branch on.