MichaIng/DietPi #5047: package repository issue with beta branch, bad or corrupted entries?

KalixtShawxo, 2021-12-10:
#### Details:
- Date           | Fri Dec 10 05:37:37 EST 2021
- DietPi version | v7.9.1 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCK64 (aarch64) (ID=43)
- Kernel version | `Linux Rock64plus 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Cleaner
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- update packages
#### Actual behaviour:

- somehow the package repositories got screwed up trying to install Chrome
#### Extra details:

- I opened the sources.list file to remove the bad entries but they are not there, and I can't find where the bad package files entries are to remove them. 
#### Additional logs:
```
Get:1 http://dl.google.com/linux/chrome/deb stable InRelease [1811 B]
Hit:2 http://mirror.12z.eu/pub/linux/armbian/apt bullseye InRelease
Err:1 http://dl.google.com/linux/chrome/deb stable InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78BD65473CB3BD13
Hit:3 https://deb.debian.org/debian bullseye InRelease
Hit:4 https://deb.debian.org/debian bullseye-updates InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists...
W: GPG error: http://dl.google.com/linux/chrome/deb stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78BD65473CB3BD13
E: The repository 'http://dl.google.com/linux/chrome/deb stable InRelease' is not signed.
```
#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

complete Sources.list file:
                                                                               
deb https://deb.debian.org/debian/ bullseye main contrib non-free
deb https://deb.debian.org/debian/ bullseye-updates main contrib non-free
deb https://deb.debian.org/debian-security/ bullseye-security main contrib non-free
deb https://deb.debian.org/debian/ bullseye-backports main contrib non-free
- ...

KalixtShawxo, 2021-12-10:
never mind. I found the problem I think.

closed 2021-12-10 by KalixtShawxo

Joulinar, 2021-12-10:
This is an issue outside DietPi as you added a 3rd party repository for `dl.google.com` where the public key was incorrect or missing.

MichaIng, 2021-12-10:
Here an info about the repository: https://www.google.com/linuxrepositories/
Especially verify and in case repeat the step to install the key.

Generally as unbranded alternative Chromium is available via `dietpi-software` and the regular Debian APT repository.

KalixtShawxo, 2021-12-10:
Thanks. I was trying to install chrome cause the chromium install would not let me login to my Google account and sync, and it was causing the SBC to crash. 
Now I just have Firefox on it and won't have my sync with Google chrome

Joulinar, 2021-12-10:
Yeah Google don't like the original Chromium browser and see's it as insecure. But this is a global issues on quite a lot of Linux based systems if I'm not mistaken. And I never have seen a real solution.

MichaIng, 2021-12-10:
> Yeah Google don't like the original Chromium browser and see's it as insecure.

They see their own source code is insecure?!? I think it makes sense and is intended, also from user perspective, that the unbundled Chromium browser does not synchronise with the commercial Google servers. Me at least wouldn't want this and I wouldn't like even seeing this feature in Chromium. If one wants to sync chromium Chrome, then it makes sense to consequently use Chrome 🙂. Obviously via Google repo it is trivial to install it as alternative to Chromium, so I don't see any issue here?

Joulinar, 2021-12-10:
I was always at the point that the open source chromium package is not the same as the global Google company

MichaIng, 2021-12-10:
Yes that is true, through Google's Chrome uses the Chromium source code and packs it's commercial features and branding on top, like the ability to sync data with Google servers. At least this is how I understand and logically expect it 😅.