MichaIng/DietPi #5048: DietPi-Software | Nextcloud Setup buddy? No ncc command, no user.

roman1528, 2021-12-10:
Did you edited the Nexcloud installation-script?

the ncc command is not available, there is no user (root) in DB, so you can't login.
at my first install i could choose either dietpi should activate SSL or not.

#### Required Information
- DietPi version | 7.8.2
- Distro version | bullseye
- Kernel version | 15.16.0-rc3-amd64 __ same with stable 15.10.0-9-amd64
- SBC model | Native PC (x86_64) EFI
- Power supply used | 350W SFX
- SDcard used | NVMe 250GB

#### Additional Information (if applicable)
- Software title | Nextcloud
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? three trys... same faults

#### Steps to reproduce

1. choose Nextcloud in dietpi-software with Apache2, PHP 7.4, mySQL (MariaDB) and install

#### Expected behaviour

- e.g. ncc maintenance:mode --on ... activate maintenance mode.
- login normally (root and root-pw) to http(s)://IP/nextcloud

#### Actual behaviour

- bash: command not found
- can't login (no user in oc_accounts)

Joulinar, 2021-12-10:
Did you tried to reboot after installation

MichaIng, 2021-12-10:
Logout/login or `. /etc/bashrc.d/dietpi.bash` is sufficient. `ncc` is a shell function, so it needs to be loaded once Nextcloud is installed, which cannot be done from within a subshell or script.

roman1528, 2021-12-10:
> Did you tried to reboot after installation

Yes I tried. every time. actually doing a new setup cause of driver mistakes... trying it ones again.

> Logout/login or `. /etc/bashrc.d/dietpi.bash` is sufficient. `ncc` is a shell function, so it needs to be loaded once Nextcloud is installed, which cannot be done from within a subshell or script.
> 
> > can't login (no user in oc_accounts)
> 
> That indeed is strange. Any failure should have lead to an error prompt during `dietpi-software` run. Do you have the output/logs from the `dietpi-software` run? Where do you see this `no user in oc_accounts` error exactly? I cannot imagine this being an output of the web interface, so do you see this in the Nextcloud log file? And you did try to login with `admin` account and default software password?

there is no user in mysql-DB. table should be oc_accounts or else.

same as above... trying it one mor time and give you mor information^^

thank so far.

MichaIng, 2021-12-10:
Please uninstall Nextcloud first, else the download of a fresh Nextcloud instance and the database creation is skipped:
```sh
dietpi-software uninstall 114
dietpi-software install 114
```
Indeed there should be an `oc_accounts` table with an `admin` user. If this is not present, there should be some error shown during the installation. You can as well check the existence and content of the database table:
```sh
mysql -e 'select uid from nextcloud.oc_accounts'
```

roman1528, 2021-12-10:
I would cry... my system recogniced my issue writing and does the install of nextcloud very well.
all is working... web-login, ncc-command

login with admin was a great hint^^ thanks.

only not normal message during install:
debconf: delaying package configuration, since apt-utils is not installed

`root@DietPi:~# mysql -e 'select uid from nextcloud.oc_accounts'
+-------+
| uid   |
+-------+
| admin |
+-------+`

[image]

MichaIng, 2021-12-10:
Okay great.

> debconf: delaying package configuration, since apt-utils is not installed

This is normal since, as stated, the `apt-utils` package is not installed. For this which are interested in the full background:
- `debconf` is a `dpkg` feature for interactive and non-interactive package configuration, using `whiptail` based console dialogues with fallback to even simpler `read` prompts, but also applies defaults when installs are done non-interactively.
- As an example you can run `debconf-show tzdata` to see the selected time zones for the `tzdata` package, which uses  `debconf` for selecting and storing this info. You can re-call this configuration via `dpkg-reconfigure tzdata` to get an impression. As you may recognise, `dietpi-config` calls just that for its time zone option 😉.
- When doing a large number of package installs or upgrades, it may be great when all packages could be interactively configured right at the start, so that for the long lasting extractions, installs, maintainer scripts (which imply applying the `debconf` selections) one can go and make some sports or drink a coffee without the need to keep watching the APT install process to be ready for whenever a certain package asks for an interactive `debconf` selection.
- This is where the file `/etc/apt/apt.conf.d/70debconf` kicks in. It is invoked by APT to extract only the `debconf` templates of all packages which are about to be installed or upgraded, to allow pre-configuring those before the actual installs start, using another command `dpkg-preconfigure`. This is a perl script which calls `apt-extracttemplates` for the actual template extraction. And that one is shipped by the dedicated `apt-utils` package. Hence when `apt-utils` is not installed, `apt-extracttemplates` is not available, `dpkg-preconfigure` recognises this and exits with the notification you see.
- We think it is fine to not have this pre-configuration feature available by default. It doesn't harm in any way and the message is quite self-explaining, at least for those users where it does actually play a role. For inexperienced users the defaults are pretty fine and the dialogues may be more confusing than helpful. Also dialogues are usually not shown on package upgrades, hence it does not apply for `tzdata` and similar packages which are pre-installed already, but only for newly installed packages in very most cases. And all our scripts call APT with a special variable `DEBIAN_FRONTEND=noninteractive` set, which instructs `debconf` to skip all interactive dialogues and keep previous selections or use the defaults instead. Without this, an unattended first run setup and installs wouldn't be possible as it could hang on a `debconf` dialog.

MichaIng, 2021-12-11:
I'll mark this as closed then. Feel free to reopen when required.

closed 2021-12-11 by MichaIng