MichaIng/DietPi #5050: Real VNC install Dietpi

xmicky, 2021-12-10:
### Creating a bug report/issue

#### Details:
- Date           | Fri Dec 10 20:34:56 GMT 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux Pi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `cp -a /root/.vnc/config.d/vncserver-x11-virtual /root/.vnc/config.d/Xvnc`
- Exit code      | 1
- Software title | DietPi-Software

failed on two fresh dietpi installations

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2021-12-10:
Hi,

do you have the entire log/error message from the installation?

MichaIng, 2021-12-10:
I think it is related to #4992.
Seems like not only is the `/root/.vnc/config.d/Xvnc` password file not always created, but I guess here `/root/.vnc/config.d/vncserver-x11-virtual` wasn't either. You can solve/work around it by opening a subshell and calling:
```sh
vncpasswd -virtual <<< "dietpi
dietpi"
```
It should create both files, but somehow it sometimes doesn't for unknown reasons. That `cp -a` command is already an attempt to fix it, together with the fact that he above `vncpasswd -virtual` is already called two times, but somehow it doesn't work as expected and we need to find out why/in which case those files are not created in the first place, or probably removed again, not sure, even that the command's output clearly states that both are created.

MichaIng, 2021-12-11:
So checked back:
```
[  OK  ] DietPi-Software | APT install for: realvnc-vnc-server
[  OK  ] DietPi-Software | systemctl daemon-reload
[  OK  ] DietPi-Software | systemctl enable vncserver
[  OK  ] DietPi-Software | systemctl disable vncserver-virtuald vncserver-x11-serviced
[  OK  ] DietPi-Software | chmod +x /usr/local/bin/vncserver
ls: cannot access '/root/.vnc/config.d': No such file or directory
Setting "Password" VNC parameter for Virtual Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/Xvnc
Successfully set "Password" VNC parameter in
/root/.vnc/config.d/vncserver-x11-virtual
total 8
-rw-r--r-- 1 root root 26 Dec 11 17:12 Xvnc
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11-virtual
Setting "Password" VNC parameter for Virtual Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/Xvnc
Successfully set "Password" VNC parameter in
/root/.vnc/config.d/vncserver-x11-virtual
total 8
-rw-r--r-- 1 root root 26 Dec 11 17:12 Xvnc
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11-virtual
Setting "Password" VNC parameter for Service Mode server
Successfully set "Password" VNC parameter in /root/.vnc/config.d/vncserver-x11
total 12
-rw-r--r-- 1 root root 26 Dec 11 17:12 Xvnc
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11-virtual
total 12
-rw-r--r-- 1 root root 26 Dec 11 17:12 Xvnc
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11-virtual

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Finalising install

[  OK  ] DietPi-Software | systemctl daemon-reload
2021-12-11 17:12:11 [ INFO ] DietPi-RAMlog | Storing /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store...
2021-12-11 17:12:11 [  OK  ] DietPi-RAMlog | Stored /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store.
[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Install completed

[  OK  ] DietPi-Survey | Sending survey data
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : cron
[ INFO ] DietPi-Software | Starting installed services, not controlled by DietPi-Services
total 12
-rw-r--r-- 1 root root 26 Dec 11 17:12 Xvnc
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11-virtual
[  OK  ] DietPi-Software | systemctl start vncserver
total 12
-rw-r--r-- 1 root root 26 Dec 11 17:12 Xvnc
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11-virtual
2021-12-11 17:12:18 root@Building-ARMv6-Bullseye:~# l .vnc/config.d/
total 8.0K
-rw-r--r-- 1 root root  0 Dec 11 17:12 .Xvnc-v5-marker
-rw-r--r-- 1 root root  0 Dec 11 17:12 Xvnc
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11
-rw-r--r-- 1 root root 26 Dec 11 17:12 vncserver-x11-virtual
```
So at least here all files are correctly created, but the first start of RealVNC somehow clears the content of the `Xvnc` file, which breaks the start. Not sure whether this marker play a role, it is until then the only additional file created on service start and the issue can be replicated by only removing `/root/.vnc/config.d` and doing a reinstall.

MichaIng, 2021-12-11:
Indeed, pre-creating that file seems to solve it. Worked repeatedly on Bullseye, will test on on all distro versions, ARMv7 and ARMv8 to be sure. `Xvnc` is also altered:
```
# cat .vnc/config.d/Xvnc
_LastUpdateCheckSuccessTime=int64:01d7eeab6fc66de8
_LastUpdateCheckTime=int64:01d7eeab6fc66de8
Password=6c722429f303d65b
```
The first two lines were added on service start, `vncserver-x11-virtual` remains untouched. So it seems like if the marker is not present, `Xvnc` is cleared for some reason. Or it is cleared in general before the password is re-added together with the timestamps, but the last step is skipped/fails when the marker is not present.

Looks buggy to me: If `vncpasswd -virtual` is used then under no circumstances that intentionally entered password must be removed, at least not without giving some explicit notification that it is done and why...

MichaIng, 2021-12-11:
Hopefully finally fixed with: https://github.com/MichaIng/DietPi/commit/d0ace4f
Changelog: https://github.com/MichaIng/DietPi/commit/93dbb0f

xmicky, 2021-12-11:
Sorry, I struggled to find the log - I was looking in directory  var but I
was distracted.
I see all your emails and understand the problem is fixed.

So…
I updated to beta 7.9 and reinstalled real VNC server without any errors.
I couldn’t connect using vnc client (I have connected when using raspberypi
os)
Running dietpi-services I see from the report that vncserver : failed
Am I missing something?

I want to make anther comment regarding the dietpi user directories.
Noteably when using Homeassistant with HACS installed.
HACS is unable to install any integrations into homeassistant because of
the folder permissions.
I was also unable to edit the homeassistant configuration.yaml via the
samba server for the same reason.
I just set all the folder permissions for the user directories to rw.
But, others may not work it out!

I’m running this pi headless via my iPad , so SSH, vnc and samba are
essential.

BTW: so far I’m really enjoying the experience great software .

Thanks for all the hard work

Mick

MichaIng, 2021-12-11:
Can you please show the service logs:
```sh
journalctl -u vncserver
```

xmicky, 2021-12-11:
> Can you please show the service logs:
>
> journalctl -u vncserver

-- Journal begins at Sat 2021-12-11 21:05:51 GMT, ends at Sat 2021-12-11
22:29:32 GMT. --
Dec 11 21:06:04 Pi systemd[1]: Started VNC Server (DietPi).
Dec 11 21:06:04 Pi systemd[512]: pam_unix(login:session): session opened
for user root(uid=0) by (uid=0)
Dec 11 21:06:05 Pi vncserver[512]: [  OK  ] RealVNC detected
Dec 11 21:06:05 Pi vncserver[512]: [ INFO ] Starting virtual desktop at
display :1 in 1280x720x16
Dec 11 21:06:05 Pi vncserver[512]: Error: Setting password failed
[NoVncPasswd]
Dec 11 21:06:05 Pi vncserver[512]: VNC(R) Server 6.7.2 (r43081) ARMv8-A
(Sep 20 2020 23:57:07)
Dec 11 21:06:05 Pi vncserver[512]: Copyright (C) 2002-2020 RealVNC Ltd.
Dec 11 21:06:05 Pi vncserver[512]: RealVNC and VNC are trademarks of
RealVNC Ltd and are protected by trademark
Dec 11 21:06:05 Pi vncserver[512]: registrations and/or pending trademark
applications in the European Union,
Dec 11 21:06:05 Pi vncserver[512]: United States of America and other
jurisdictions.
Dec 11 21:06:05 Pi vncserver[512]: Protected by UK patent 2481870; US
patent 8760366; EU patent 2652951.
Dec 11 21:06:05 Pi vncserver[512]: See https://www.realvnc.com for
information on VNC.
Dec 11 21:06:05 Pi vncserver[512]: For third party acknowledgements see:
Dec 11 21:06:05 Pi vncserver[512]: https://www.realvnc.com/docs/6/foss.html
Dec 11 21:06:05 Pi vncserver[512]: OS: Debian GNU/Linux 11, Linux 5.10.63,
aarch64
Dec 11 21:06:05 Pi vncserver[512]: You must specify a password for VNC
authentication.
Dec 11 21:06:05 Pi systemd[1]: vncserver.service: Main process exited,
code=exited, status=1/FAILURE
Dec 11 21:06:05 Pi systemd[1]: vncserver.service: Failed with result
'exit-code'.
Dec 11 22:03:28 Pi systemd[1]: Started VNC Server (DietPi).
Dec 11 22:03:28 Pi systemd[8641]: pam_unix(login:session): session opened
for user root(uid=0) by (uid=0)
Dec 11 22:03:28 Pi vncserver[8641]: [  OK  ] RealVNC detected
Dec 11 22:03:28 Pi vncserver[8641]: [ INFO ] Starting virtual desktop at
display :1 in 1280x720x16
Dec 11 22:03:28 Pi vncserver[8780]: Setting "Password" VNC parameter for
Virtual Mode server
Dec 11 22:03:28 Pi vncserver[8780]: Error: Password not specified
Dec 11 22:03:28 Pi vncserver[8641]: Error: Setting password failed
[NoVncPasswd]
Dec 11 22:03:28 Pi vncserver[8641]: VNC(R) Server 6.7.2 (r43081) ARMv8-A
(Sep 20 2020 23:57:07)
Dec 11 22:03:28 Pi vncserver[8641]: Copyright (C) 2002-2020 RealVNC Ltd.
Dec 11 22:03:28 Pi vncserver[8641]: RealVNC and VNC are trademarks of
RealVNC Ltd and are protected by trademark
Dec 11 22:03:28 Pi vncserver[8641]: registrations and/or pending trademark
applications in the European Union,
Dec 11 22:03:28 Pi vncserver[8641]: United States of America and other
jurisdictions.
Dec 11 22:03:28 Pi vncserver[8641]: Protected by UK patent 2481870; US
patent 8760366; EU patent 2652951.
Dec 11 22:03:28 Pi vncserver[8641]: See https://www.realvnc.com for
information on VNC.
Dec 11 22:03:28 Pi vncserver[8641]: For third party acknowledgements see:
Dec 11 22:03:28 Pi vncserver[8641]: https://www.realvnc.com/docs/6/foss.html
Dec 11 22:03:28 Pi vncserver[8641]: OS: Debian GNU/Linux 11, Linux 5.10.63,
aarch64
Dec 11 22:03:28 Pi vncserver[8641]: You must specify a password for VNC
authentication.
Dec 11 22:03:28 Pi systemd[1]: vncserver.service: Main process exited,
code=exited, status=1/FAILURE
Dec 11 22:03:28 Pi systemd[1]: vncserver.service: Failed with result
'exit-code'.
***@***.***:/boot$

MichaIng, 2021-12-11:
Can you show the content of the config directory:
```sh
ls -Al /root/.vnc/config.d
```

xmicky, 2021-12-11:
***@***.***:/boot$ sudo ls -Al /root/.vnc/config.d
total 0
-rw-r--r-- 1 root root 0 Dec 11 22:03 .Xvnc-v5-marker
***@***.***:/boot$

MichaIng, 2021-12-11:
Do you still have the output of the RealVNC reinstall in the console backlog? There should be a part about three files bring created in this directory. Is there any error message among them? I did 6 install tests on different distro versions and architectures, and here it worked very reliably now. But in your case the exact symptom/error is also a different one it seems: Not only one file was missing, but none seems to be created at all, which is why the `cp -a` failed, a case we never observed before 🤔.

Also try to apply the password now (as root user):
```sh
vncpasswd -virtual <<< 'dietpi
dietpi'
vncpasswd -service <<< 'dietpi
dietpi'
ls -Al /root/.vnc/config.d
```
This should tell you that three files are created in that directory and those of course also really should be created. Password in this case is "dietpi", which you of course can change as you like.
___________
For Home Assistant HACS, please open a new issue about that, so we do not mix up two topics.

xmicky, 2021-12-11:
***@***.***:~$ sudo ls -Al /root/.vnc/config.d
total 12
-rw-r--r-- 1 root root  0 Dec 11 22:03 .Xvnc-v5-marker
-rw-r--r-- 1 root root 26 Dec 11 22:49 Xvnc
-rw-r--r-- 1 root root 26 Dec 11 22:51 vncserver-x11
-rw-r--r-- 1 root root 26 Dec 11 22:49 vncserver-x11-virtual
***@***.***:~$

Ok. I ran your suggested commands and rebooted the system.
Services now shows vnc service as started
However my real VNC client on my IPad reports it
“cannot connect because the port which the computer is listening cannot be
contacted”

MichaIng, 2021-12-11:
Was this `ls -Al /root/.vnc/config.d` before or after you ran the `vncpasswd` commands? However this works now. To assure that the service listens to the port:
```sh
ss -tlpn
```

xmicky, 2021-12-11:
It was before.
The result of running your command didn’t allow the vnc clent to connect. -
same as before.
I also note there now a FAIL for vpn service

***@***.***:~$ sudo dietpi-services
[  OK  ] DietPi-Services | restart : nmbd
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : redis-server
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : php7.4-fpm
[  OK  ] DietPi-Services | restart : lighttpd
[  OK  ] DietPi-Services | restart : qbittorrent
[  OK  ] DietPi-Services | restart : filebrowser
[  OK  ] DietPi-Services | restart : home-assistant
[  OK  ] DietPi-Services | restart : cron
[  OK  ] DietPi-Services | restart : dropbear
[  OK  ] DietPi-Services | restart : vncserver
[FAILED] DietPi-Services | restart : dietpi-vpn
[  OK  ] DietPi-Services | restart : dietpi-cloudshell
***@***.***:~$ sudo ss -tlpn
State        Recv-Q       Send-Q                                 Local
Address:Port             Peer Address:Port      Process

LISTEN       0            80
127.0.0.1:3306                  0.0.0.0:*
 users:(("mariadbd",pid=2136,fd=19))

LISTEN       0            511
127.0.0.1:6379                  0.0.0.0:*
 users:(("redis-server",pid=2059,fd=7))

LISTEN       0            50
0.0.0.0:139                   0.0.0.0:*
 users:(("smbd",pid=2044,fd=45))

LISTEN       0            5
0.0.0.0:5901                  0.0.0.0:*
 users:(("Xvnc-core",pid=2485,fd=7))

LISTEN       0            1024
0.0.0.0:80                    0.0.0.0:*
 users:(("lighttpd",pid=2238,fd=4))

LISTEN       0            128
0.0.0.0:6001                  0.0.0.0:*
 users:(("Xvnc-core",pid=2485,fd=1))

LISTEN       0            1000
0.0.0.0:22                    0.0.0.0:*
 users:(("dropbear",pid=2349,fd=4))

LISTEN       0            128
0.0.0.0:8123                  0.0.0.0:*
 users:(("hass",pid=2296,fd=17))

LISTEN       0            50
0.0.0.0:445                   0.0.0.0:*
 users:(("smbd",pid=2044,fd=44))

LISTEN       0            30
 192.168.2.92%wlan0:6881                  0.0.0.0:*
 users:(("qbittorrent-nox",pid=2253,fd=19))

LISTEN       0            30
 127.0.0.1%lo:6881                  0.0.0.0:*
 users:(("qbittorrent-nox",pid=2253,fd=15))

LISTEN       0            511
 [::1]:6379                     [::]:*
 users:(("redis-server",pid=2059,fd=8))

LISTEN       0            50
 [::]:139                      [::]:*
 users:(("smbd",pid=2044,fd=43))

LISTEN       0            5
[::]:5901                     [::]:*
 users:(("Xvnc-core",pid=2485,fd=6))

LISTEN       0            1024
 [::]:80                       [::]:*
 users:(("lighttpd",pid=2238,fd=5))

LISTEN       0            128
[::]:6001                     [::]:*
 users:(("Xvnc-core",pid=2485,fd=0))

LISTEN       0            4096
  *:8084                        *:*
 users:(("filebrowser",pid=2263,fd=7))

LISTEN       0            1000
 [::]:22                       [::]:*
 users:(("dropbear",pid=2349,fd=5))

LISTEN       0            128
[::]:8123                     [::]:*
 users:(("hass",pid=2296,fd=18))

LISTEN       0            50
  *:1340                        *:*
 users:(("qbittorrent-nox",pid=2253,fd=41))

LISTEN       0            50
 [::]:445                      [::]:*
 users:(("smbd",pid=2044,fd=42))

LISTEN       0            30
[fe80::dea6:32ff:fe2d:a39b]%wlan0:6881                     [::]:*
 users:(("qbittorrent-nox",pid=2253,fd=22))

LISTEN       0            30
 [::1]%lo:6881                     [::]:*
 users:(("qbittorrent-nox",pid=2253,fd=18))

***@***.***:~$

MichaIng, 2021-12-11:
```
LISTEN 0 5 0.0.0.0:5901 0.0.0.0:* users:(("Xvnc-core",pid=2485,fd=7))
LISTEN 0 5 [::]:5901 [::]:* users:(("Xvnc-core",pid=2485,fd=6))
```
All fine, RealVNC is running and listening on the intended ports. Please verify that you use the right hostname/IP to connect and that it generally can be connected to, e.g. the webserver via browser.

Also not that it runs a virtual desktop by default on VNC screen 1, so you usually need to append ` :1` to the hostname/IP in the VNC client.

xmicky, 2021-12-11:
Great! it works connected to screen 1
Thank you.

MichaIng, 2021-12-11:
Okay great. If you need to connect to a desktop session that is locally running on the system, instead of a virtual one, you can switch shared desktop mode: https://dietpi.com/docs/software/remote_desktop/#realvnc-server
RealVNC then listens on port `5900` (screen `:0`) and it can then be connected without appending the screen index (`:0` is then assumed automatically). I'm currently not sure which service the package by default starts with (on Raspberry Pi OS), probably even both modes in parallel, so this may be the reason why you didn't need to add `:1` there.

xmicky, 2021-12-11:
Thank you for all the help.
I’ll try shared the desktop mode.
Mick

MichaIng, 2021-12-12:
I'll mark this as closed now to get the v7.9 milestone cleared. However, keep me updated about whether the shared desktop mode works fine for you.

And of course feel free to open an issue about Home Assistant with HACS.

closed 2021-12-12 by MichaIng

xmicky, 2021-12-12:
Thank you. One more thing.
Now the vnc server is running it is consuming a lot of processor power.
My cpu was continuously at 25 percent when not connected via the client.
I just stopped the vnc server and the cpu % dropped to 1%

Thanks
Mick

MichaIng, 2021-12-12:
In virtual mode or shared mode?

The virtual (default) mode runs a full desktop session constantly, so you should see it in RAM usage, but as long as there is no desktop application running, it shouldn't load the CPU much 🤔.

xmicky, 2021-12-12:
In virtual mode with no vnc client connected.
25% of a quad core a72 CPU is a lot

Mick

MichaIng, 2021-12-12:
Is it the same when you start a desktop on a local screen? A VNC server shouldn't produce any more CPU load than the desktop/X session that is running within. You can also use `htop` to detect the exact process that is causing the CPU load.

xmicky, 2021-12-12:
Hi, an update FYI
CPU usage at 25% when the real vnc server does not have a client attached
implies there is a problem with the server software or the integration.
I proved the high CPU usage by stopping and starting the service.
I've now dumped real vnc server and installed Tiger vnc.
Now my CPU usage is back to 1% when the system idles with home assistant
running.
Regards
Mick

Joulinar, 2021-12-12:
I observed something similar when the RealVNC client connect is not closed correctly. There it seems some process stuck.

xmicky, 2021-12-12:
Both results were obtained when viewing the virtual desktop for both real
vnc and Tiger vnc.
I have to look again to see if there is any difference between the virtual
and user desktop.
I have to say that the real-vnc server desktop display was very sluggish
compared to the Tiger vnc. I'm using the same real vnc viewer for both.
I'm looking at htop with Tiger vnc client active and I'm seeing less than
2% CPU.
Mick

xmicky, 2021-12-12:
My results are from restart so, with no vnc session initiated.

Joulinar, 2021-12-12:
My RPi4B idle time is around 99.8% after a reboot having RealVNC installed.

```
top - 14:58:11 up 6 min,  1 user,  load average: 0.00, 0.02, 0.00
Tasks: 110 total,   1 running, 109 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.0 us,  0.2 sy,  0.0 ni, 99.8 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7862.9 total,   7528.1 free,     88.7 used,    246.1 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.   7656.4 avail Mem
```

Just having a RealVNC session open, still keeps my system at 99% idle time

```
top - 15:01:24 up 10 min,  1 user,  load average: 0.00, 0.02, 0.00
Tasks: 112 total,   1 running, 111 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.4 us,  0.3 sy,  0.0 ni, 99.3 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7862.9 total,   7508.2 free,    102.9 used,    251.8 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.   7641.7 avail Mem

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
   1264 root      20   0    6704   3576   2860 S   1.0   0.0   0:00.74 htop
    623 root      20   0   60236  31596  15556 S   0.7   0.4   0:01.41 Xvnc-core
   1247 root      20   0    8828   3232   2692 R   0.7   0.0   0:01.53 top
   1263 root      20   0  306276  31696  25332 S   0.3   0.4   0:00.55 x-terminal-emul

```

xmicky, 2021-12-12:
I uninstalled tiger vnc and reinstalled real VNC.
Dietpi services showed vncserver failed.
I applied the command line given before -
Vncpasswd -virtual <<<< ‘deipi …..
(Only this one)
And started the vnc service.
Result service running but high cpu use……

DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar
(support)
  0[|                                                0.7%]
Mem[|||||||||||||                              388M/3.72G]
  1[|||||||||||||||||||||||||||||||||||||           68.0%] Swp[
                                   0K/0K]
  2[|||||||||||||                                   22.4%] Tasks: 52; 2
running
  3[|||                                              3.3%] Load average:
0.89 0.55 0.24
Avg[|||||||||||||                                   23.6%] Uptime: 00:14:49
    PID△USER        RES  NI CPU%   TIME+  Command

      1 root       9792   0  0.0  0:02.17 /sbin/init

    131 root       7376   0  0.0  0:00.73 ├─ /lib/systemd/systemd-journald
    177 root       4876   0  0.0  0:00.28 ├─ /lib/systemd/systemd-udevd
    271 root       2600   0  0.0  0:00.02 ├─ /usr/bin/vmtouch -l
/boot/dietpi /boot/dietpi.txt
    288 messagebu  3684   0  0.0  0:00.19 ├─ /usr/bin/dbus-daemon --system
--address=systemd: --nofork --nopidfile --sy
    291 root        704   0  0.0  0:00.03 ├─ /usr/sbin/rngd -f
    296 root       6520   0  0.0  0:00.18 ├─ /lib/systemd/systemd-logind
    330 root        184   0  0.0  0:00.00 ├─ /usr/sbin/dropbear -p 22 -W
65536
   1507 root       2500   0  0.0  0:00.44 │  └─ /usr/sbin/dropbear -p 22 -W
65536
   1508 dietpi     5492   0  0.0  0:00.09 │     └─ -bash
   2179 dietpi     3776   0  1.3  0:01.49 │        └─ htop
    408 root       5424   0  0.0  0:00.12 ├─ /sbin/dhclient -4 -v -i -pf
/run/dhclient.eth0.pid -lf /var/lib/dhcp/dhcli
  0[|||||||||                                       15.8%]
Mem[|||||||||||||                              388M/3.72G]
  1[||||||||||||||||||||||||||||||||||||            64.5%] Swp[
                                   0K/0K]
  2[                                                 0.0%] Tasks: 52; 2
running
  3[||||||||||                                      15.2%] Load average:
0.90 0.56 0.25
Avg[|||||||||||||                                   23.9%] Uptime: 00:14:51
    PID△USER        RES  NI CPU%   TIME+  Command

      1 root       9792   0  0.0  0:02.17 /sbin/init

    131 root       7376   0  0.0  0:00.73 ├─ /lib/systemd/systemd-journald
    177 root       4876   0  0.0  0:00.28 ├─ /lib/systemd/systemd-udevd
    271 root       2600   0  0.0  0:00.02 ├─ /usr/bin/vmtouch -l
/boot/dietpi /boot/dietpi.txt
    288 messagebu  3684   0  0.0  0:00.19 ├─ /usr/bin/dbus-daemon --system
--address=systemd: --nofork --nopidfile --sy
    291 root        704   0  0.0  0:00.03 ├─ /usr/sbin/rngd -f
    296 root       6520   0  0.0  0:00.18 ├─ /lib/systemd/systemd-logind
    330 root        184   0  0.0  0:00.00 ├─ /usr/sbin/dropbear -p 22 -W
65536
   1507 root       2500   0  0.0  0:00.44 │  └─ /usr/sbin/dropbear -p 22 -W
65536
   1508 dietpi     5492   0  0.0  0:00.09 │     └─ -bash
   2179 dietpi     3776   0  0.7  0:01.50 │        └─ htop
    408 root       5424   0  0.0  0:00.12 ├─ /sbin/dhclient -4 -v -i -pf
/run/dhclient.eth0.pid -lf /var/lib/dhcp/dhcli
F1Help  F2Setup F3SearchF4FilterF5List  F6SortByF7Nice -F8Nice +F9Kill
 F10Quit

MichaIng, 2021-12-12:
First of all, please update to DietPi v7.9 to have this `vncpasswd` issue solved 🙂. If you have any chance for future console pastes, please do not post them by email but directory on github.com, so that you can wrap it info code fences. It is otherwise extremely hard to read. I see ~90% CPU usage but I do not see the RealVNC process in the list, or any other process with high CPU load. I think you need to scroll down a little to see them, `Xvnc-core` and a bunch of other desktop-related processes.

In case of RealVNC virtual desktop: When you logout via desktop menu instead of just closing the session client/window-wise, indeed the whole session together with the service stops, so it needs to be restarted then: `systemctl restart vncserver`
With TigerVNC logout for this reason simply has no effect, so this cannot happen by accident.

I didn't observe a hanging service when closing the connection somehow unclean client-wise, but to be true I haven't tested the RealVNC client for a while but keep using the tiny portable TightVNC viewer for testing 😄: https://www.tightvnc.com/download/1.3.10/tightvnc-1.3.10_x86_viewer.zip
However, no recommendation as it is stone age old, there are [newer non-portable TightVNC viewer versions available](https://www.tightvnc.com/download.php), also [TigerVNC](https://sourceforge.net/projects/tigervnc/files/stable/) or [UltraVNC](https://uvnc.com/downloads/ultravnc.html) as maintained modern alternatives 😉.

xmicky, 2021-12-12:
Ok will do.
I am on 17.9 beta so the vnc problem should be fixed in that?
Sorry to say my SSH terminal is not allowing me to scroll down the htop
display I sent all I could see. I could go and plug the hdmi in to the TV
but I will be unable to copy the result

an update:
 I changed diet.txt so that I should see the shared destop. I had to apply
the password fix as before.
The server is now running but so far I cannot connect - port not
contactable - I'm leaving the index off the end of the IP.
CPU showing a low percentage.

MichaIng, 2021-12-12:
The arrow down or page down buttons on the keyboard do not work? It could be also sorted by CPU usage, but I guess if scrolling down does not work, then tabbing through the fields to sort the process list will be similarly difficult 🤔.

> I am on 17.9 so the vnc problem should be fixed in that?

The RC version is important, it should be v7.9.3.

> I changed diet.txt so that I should see the shared destop.

Note that this requires a locally running desktop/X session. You can achieve this as well without any screen attached by using `dietpi-autostart 2`, which means an automatic login and desktop session on local console. Running the menu interactively `dietpi-autostart` and selecting `2  : Automatic login` also allows you to select a different user, so it is not automatically a `root` user desktop session.