MichaIng/DietPi #5054: DNS settings in dietpi.txt are ignored/not applied (AUTO_SETUP_NET_STATIC_DNS)

nils-trubkin, 2021-12-11:
#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'

- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
- SBC model | RPi 3 Model B+ (aarch64)
- Power supply used | USB port on router
- SDcard used | SanDisk Ultra 32GB

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated?
- Fresh
- Can this issue be replicated on a fresh installation of DietPi?
- Yes

#### Steps to reproduce

1. Fresh Install
2. Set static IP (AUTO_SETUP_NET_USESTATIC=1)
3. Set custom DNS (eg. AUTO_SETUP_NET_STATIC_DNS=1.1.1.1 9.9.9.9)

#### Expected behaviour
Diet-Pi Config showing custom DNS (eg. 1.1.1.1 9.9.9.9)

#### Actual behaviour

Diet-Pi Config showing default DNS (eg. 9.9.9.9 149.112.112.112)

Joulinar, 2021-12-11:
Hi, 

Can you share following pls

`cat /etc/resolv.conf`

MichaIng, 2021-12-11:
It was a purely visual bug: On first boot, as `resolvconf` is not pre-installed, the static DNS server is applied to `/etc/resolv.conf` directly and `/etc/network/interfaces` stays untouched in this regards. `dietpi-config` however keeps showing the value from `/etc/network/interfaces`, even when it is commented and ineffective. Fixed it with these two commints:
- https://github.com/MichaIng/DietPi/commit/84d6ae4
- https://github.com/MichaIng/DietPi/commit/b9943d3

`dietpi-config` now only shows the value of `/etc/network/interfaces` if it is actually uncommented/effective and falls back to the value in `/etc/resolv.conf` which is definitely effective but where we cannot know whether it was applied as static DNS, via DHCP or dynamically via `resolvconf`.

Additionally, to have it aligned, on first boot the entry in `/etc/network/interfaces` is adjusted even if `resolvconf` is not present, in an ineffective/commented way, so that it matches with the value in `/etc/resolv.conf` to avoid confusion.

One minor issue is that when you apply two nameserver IPs, only one is shown in `dietpi-config`. Ah simple to fix as well...

nils-trubkin, 2021-12-11:
> Hi,
> 
> Can you share following pls
> 
> `cat /etc/resolv.conf`

Hello, probably not needed at this moment but I'll post it here anyway:

nameserver 1.1.1.1
nameserver 9.9.9.9

MichaIng, 2021-12-11:
Okay great, this verifies that the correct DNS server is in use. `dietpi-config` will show it correctly with next version released in 2 hours or so.

closed 2021-12-11 by MichaIng