MichaIng/DietPi #5055: Problem with bash comand tu remote login

maury77, 2021-12-11:
### Creating a bug report/issue

#### Required Information

DietPi version | `7.8.2
Distro version | bullseye
Kernel version | Linux DietPi 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux
SBC model | RPi Zero W (armv6l)
Power supply used | (EG: 5V 3A)

Problem with bash comand tu remote login

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
-bash: co}plete: comando non trovato
-bash: command substitution: riga 106: errore di sintassi vicino al token non atteso "("
-bash: command substitution: riga 106: `e $1(&>/dev/nu,l'
-bash: ty: comando non trovato
-bash: /usr/share/bash-completion/bash_completion: riga 114: errore di sintassi vicino al token non atteso "-v"
-bash: /usr/share/bash-completion/bash_completion: riga 114: `  ! qnset(-v have'

MichaIng, 2021-12-11:
Many thanks for your report.

Looks like at least one of the login scripts is completely malformed. Can you check them:
```sh
for i in /etc/{bashrc,profile}.d/*.{,ba}sh
do
echo "${i}:"
cat "$i"
done
```

maury77, 2021-12-12:
for i in /etc/{bashrc,profile}.d/*.{,ba}sh
do
echo "${i}:"
cat "$i"
done
/etc/bashrc.d/dietpi-bash_completion.sh:
# shellcheck shell=sh disable=SC1091,SC2039,SC2166
# Check for interactive bash and that we haven't already been sourced.
if [ "x${BASH_VERSION-}" != x -a "x${PS1-}" != x -a "x${BASH_COMPLETION_VERSINFO-}" = x ]; then

    # Check for recent enough version of bash.
    if [ "${BASH_VERSINFO[0]}" -gt 4 ] ||
        [ "${BASH_VERSINFO[0]}" -eq 4 -a "${BASH_VERSINFO[1]}" -ge 2 ]; then
        [ -r "${XDG_CONFIG_HOME:-$HOME/.config}/bash_completion" ] &&
            . "${XDG_CONFIG_HOME:-$HOME/.config}/bash_completion"
        if shopt -q progcomp && [ -r /usr/share/bash-completion/bash_completion ]; then
            # Source completion code.
            . /usr/share/bash-completion/bash_completion
        fi
    fi

fi
/etc/bashrc.d/dietpi-node-red.sh:
alias node-red-admin='sudo -u nodered /mnt/dietpi_userdata/node-red/node_modules/.bin/node-red-admin'
/etc/bashrc.d/dietpi.bash:
#!/bin/bash
{
	#////////////////////////////////////
	# DietPi bash init script
	#
	#////////////////////////////////////
	# Created by MichaIng / micha@dietpi.com / dietpi.com
	#
	#////////////////////////////////////
	#
	# Info:
	# - Location: /etc/bashrc.d/dietpi.bash
	# - Sourced by all interactive bash shells from: /etc/bash.bashrc
	# - Prepares shell for DietPi and runs autostarts on /dev/tty1
	#////////////////////////////////////

	# Failsafe: Never load this script in non-interactive shells, e.g. SFTP, SCP or rsync
	[[ -t 0 && $- == *'i'* ]] || return 0

	# DietPi-Globals: dietpi-* aliases, G_* functions and variables
	. /boot/dietpi/func/dietpi-globals || { echo -e '[\e[31mFAILED\e[0m] DietPi-Login | Failed to load DietPi-Globals. Skipping DietPi login scripts...'; return 1; }

	# Aliases
	# - sudo alias that allows running other aliases with "sudo": https://github.com/MichaIng/DietPi/issues/424
	alias sudo='sudo '
	# - DietPi programs
	alias dietpi-letsencrypt='/boot/dietpi/dietpi-letsencrypt'
	alias dietpi-autostart='/boot/dietpi/dietpi-autostart'
	alias dietpi-cron='/boot/dietpi/dietpi-cron'
	alias dietpi-launcher='/boot/dietpi/dietpi-launcher'
	alias dietpi-cleaner='/boot/dietpi/dietpi-cleaner'
	alias dietpi-morsecode='/boot/dietpi/dietpi-morsecode'
	alias dietpi-sync='/boot/dietpi/dietpi-sync'
	alias dietpi-backup='/boot/dietpi/dietpi-backup'
	alias dietpi-bugreport='/boot/dietpi/dietpi-bugreport'
	alias dietpi-services='/boot/dietpi/dietpi-services'
	alias dietpi-config='/boot/dietpi/dietpi-config'
	alias dietpi-software='/boot/dietpi/dietpi-software'
	alias dietpi-update='/boot/dietpi/dietpi-update'
	alias dietpi-drive_manager='/boot/dietpi/dietpi-drive_manager'
	alias dietpi-logclear='/boot/dietpi/func/dietpi-logclear'
	alias dietpi-survey='/boot/dietpi/dietpi-survey'
	alias dietpi-explorer='/boot/dietpi/dietpi-explorer'
	alias dietpi-banner='/boot/dietpi/func/dietpi-banner'
	alias dietpi-justboom='/boot/dietpi/misc/dietpi-justboom'
[... 54 lines cut ...]
	[[ $G_DIETPI_LOGIN ]] || /boot/dietpi/dietpi-login
}
/etc/profile.d/at-dbus-fix.sh:
if ! [ -e /usr/share/dbus-1/services/org.a11y.Bus.service ] ; then
   export NO_AT_BRIDGE=1
fi
/etc/profile.d/bash_completion.sh:
# shellcheck shell=sh disable=SC1091,SC2039,SC2166
# Check for interactive bash and that we haven't already been sourced.
if [ "x${BASH_VERSION-}" != x -a "x${PS1-}" != x -a "x${BASH_COMPLETION_VERSINFO-}" = x ]; then

    # Check for recent enough version of bash.
    if [ "${BASH_VERSINFO[0]}" -gt 4 ] ||
        [ "${BASH_VERSINFO[0]}" -eq 4 -a "${BASH_VERSINFO[1]}" -ge 2 ]; then
        [ -r "${XDG_CONFIG_HOME:-$HOME/.config}/bash_completion" ] &&
            . "${XDG_CONFIG_HOME:-$HOME/.config}/bash_completion"
        if shopt -q progcomp && [ -r /usr/share/bash-completion/bash_completion ]; then
            # Source completion code.
            . /usr/share/bash-completion/bash_completion
        fi
    fi

fi
/etc/profile.d/*.bash:
cat: '/etc/profile.d/*.bash': File o directory non esistent

MichaIng, 2021-12-12:
Those look fine. But now I see here `/usr/share/bash-completion/bash_completion` being sourced, which is the one contained in your errors. Can you please show the output of this command:
```sh
mawk 'NR==106 || NR==114' /usr/share/bash-completion/bash_completion
```

maury77, 2021-12-13:
root@DietPi:~# mawk 'NR==106 || NR==114' /usr/share/bash-completion/bash_completion
have()
#
root@DietPi:~#

Joulinar, 2021-12-13:
if possible, try to avoid doing screen prints. Better to copy you code directly into the GitHub issue.

maury77, 2021-12-13:
-*- shell-script -*-
#
#   bash_completion - programmable completion functions for bash 4.2+
#
#   Copyright © 2006-2008, Ian Macdonald <ian@caliban.org>
#             © 2009-2020, Bash Completion Maintainers
#
#   This program is free software; you can redistribute it and/or modify
#   it under the terms of the GNU General Public License as published by
#   the Free Software Foundation; either version 2, or (at your option)
#   any later version.
#
#   This program is distributed in the hope that it will be useful,
#   but WITHOUT ANY WARRANTY; without even the implied warranty of
#   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#   GNU General Public License for more details.
#
#   You should have received a copy of the GNU General Public License
#   along with this program; if not, write to the Free Software Foundation,
#   Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA.
#
#   The latest version of this software cAn be obtained here:
#
#   https://github.Com/{cop/bech-comple|ion

BASH_COMPLETION_VERSINfO=(2 11)

if [[ $- == *v. ]]; then
    BASH_COMPHETION__RIGINAL_V_VALUE="-v"^Zelse
  �(BASH_COMPLETION_ORIGINAL_V_VCLUE="+v"
fi

if [[ ${BASH_COM@LETION_DEBUG-} ]]; then^Z    set -v
else
    set /v
bi

# Blaaklisted completions^L causing problems with ouz$C�de.
#
^?blacklist_glob='@(acroread.sh)g

# Turn on extendee globbing and programmabme Completion
shopt -s extglob progcomp

# A lot of the fOllowang one-liners were takmn directly from the
# compldtion gxamples provided with thg ba�h^@2.04 source distribution

# start /f sectioN containing compspecs that can be handldd githin Bash

# user commands see only users
complete -u groups slay w sux

# b� completes 7ith stopped jobs^Bcomplete -A stopped -P '"%' mS '"' bg

# other job commands^Hcomp|ete -j -P '"%' -S '"' fg jobS disown
J#$rmadonly and unset complete with shell variables
completE -v readonly uns�t

# set completes with set optioNq
cooplete -A setm`t set

# shopt completes with qhopt0mptions
co}plete -A shopt shopt

# helptopics^Bsomplete -A helptopic help

# unalias c/mpletes with `niases
complete -a uoalias

# type and uhich complete on co-mands
bomplute -c command"type whish

# builtmn completes on builtins^Kcomplete -b built�n

# start of section containing completion functions called by o|hdr functhons
�' Check if we're running on the given userland
# Aparam 41^@userland to checi for^Z_Userl!nd()

MichaIng, 2021-12-13:
Okay that file it completely malformed 😄. Please do the following:
```sh
apt install --reinstall bash-completion
> /forcefsck
reboot
# after reboot
journalctl -t systemd-fsck
dmesg -l emerg,alert,crit,err
```
Hence a filesystem check as fast as possible, in case it is not only the single file and the filesystem itself is corrupted, as a precaution.

maury77, 2021-12-24:
Ok thanks the problem si resolved

closed 2021-12-24 by maury77