MichaIng/DietPi #5058: Concurrent execution of DietPi-Software detected

kd9352, 2021-12-12:
Hey! I have been using dietpi now for 2 months and I really like it. Today after I updated to v7.9 I had a problem. Every time I go into dietpi-software, dietpi-config, I get the same error as below. This is the first time I post something on github, so I apologize if anything is missing.

WARNING: Concurrent execution of DietPi-Software detected                                                                
   │                                                                                                                          
   │ Please check if one of the following applies:                                                                            
   │  - This script already runs on another terminal/SSH session.                                                             
   │  - Currently a cron or systemd background job executes the script.                                                       
   │  - You started this script from within another DietPi program, causing a loop.                                           
   │                                                                                                                          
   │ Please assure that the concurrent execution has finished, before retrying, otherwise cancel this instance.               
   │                                                                                                                          
   │ The following info might help:                                                                                           
   │     248 root     ?        /usr/bin/vmtouch -l /boot/dietpi /boot/dietpi.txt                                              
   │    2431 root     tty1          \_ su dietpi                                                                              
   │    2432 dietpi   tty1              \_ bash                                                                               
   │    4477 root     tty1                              \_ su dietpi                                                          
   │    4478 dietpi   tty1                                  \_ bash                                                           
   │    4512 root     ?         \_ sshd: dietpi [priv]                                                                        
   │    4518 dietpi   ?             \_ sshd: dietpi@pts/0                                                                     
   │    4519 dietpi   pts/0             \_ -bash                                                                              
   │    5879 root     pts/0                             \_ su dietpi                                                          
   │    5880 dietpi   pts/0                                 \_ bash                                                           
   │    4168 filebro+ ?        /opt/filebrowser/filebrowser -d /mnt/dietpi_userdata/filebrowser/filebrowser.db                
   │    4211 root     tty42    /opt/dietpi-dashboard/dietpi-dashboard                                                         
   │    5670 root     pts/1     |   \_ /bin/bash /boot/dietpi/dietpi-software                                                 
   │    5703 root     pts/1     |       \_ /bin/bash /boot/dietpi/dietpi-software                                             
   │    5704 root     pts/1     |           \_ whiptail --title DietPi-Software --backtitle Native PC (x86_64) | IP:          
   │ 192.168.1.169 --menu  --ok-button Ok --cancel-button Exit --default-item Help! 21 80 14 Help! : Links to online

ravenclaw900, 2021-12-12:
Hello,
This seems to be an issue with DietPi-Dashboard, and the DietPi-Software process running with no arguments (e.g. `install` or `uninstall`). Running `systemctl restart dietpi-dashboard` should fix the problem, at least temporarily.

MichaIng, 2021-12-12:
Many thanks for your report.

It seems you have multiple shell sessions open concurrently, one at a locally attached screen, with a doubled nested `su` session? I suggest you login as `dietpi` user directly instead of using `su`, otherwise you pass the `root` users shell environment (variables) to the unprivileged `dietpi` user, which is a potential security leak. If you use this system only by yourself, it is okay, but do not start to get used to this bad practice. `su` is generally a bad instrument because of it's default behaviour of passing the environment over to the subshell, better use `sudo` if you need to get `root` access from an unprivileged login session. If you need to get access as unprivileged user, do not login as `root` in the first place.

On the SSH session it seems to be a `dietpi` session in the first place, what was the reason to for the `su dietpi` subshell? If you for some reason need a subshell, run `bash` instead.

And then the DietPi-Dashboard session.
@ravenclaw900 
I think the concurrent execution error happened within the DietPi-Dashboard terminal session, so there seems to be or have been a concurrent execution on either SSH or local console.

@kd9352 
Please follow ravenclaw's advice and see whether this solves the issue. Also please try to close all open shells and subshells and then on a single open login session, run and paste the output of the following commands:
```sh
ls -l /tmp
ps axf | grep '[d]ietpi-'
```

kd9352, 2021-12-12:
> Many thanks for your report.
> 
> It seems you have multiple shell sessions open concurrently, one at a locally attached screen, with a doubled nested `su` session? I suggest you login as `dietpi` user directly instead of using `su`, otherwise you pass the `root` users shell environment (variables) to the unprivileged `dietpi` user, which is a potential security leak. If you use this system only by yourself, it is okay, but do not start to get used to this bad practice. `su` is generally a bad instrument because of it's default behaviour of passing the environment over to the subshell, better use `sudo` if you need to get `root` access from an unprivileged login session. If you need to get access as unprivileged user, do not login as `root` in the first place.
> 
> On the SSH session it seems to be a `dietpi` session in the first place, what was the reason to for the `su dietpi` subshell? If you for some reason need a subshell, run `bash` instead.
> 
> And then the DietPi-Dashboard session. @ravenclaw900 I think the concurrent execution error happened within the DietPi-Dashboard terminal session, so there seems to be or has been a concurrent execution on either SSH or local console.
> 
> @kd9352 Please follow ravenclaw's advice and see whether this solves the issue. Else please try to close all open shells and subshells and then on a single open login session, run and paste the output of the following commands:
> 
> ```shell
> ls -l /tmp
> ps axf | grep '[d]ietpi-'
> ```

Yes, thanks to everyone for the help. I use dietpi for my own use such as jellyfin, nextcloud and minecraft server. And I really love dietpi dashboard as the server is with my mother and I am a student who lives somewhere else therefore dietpi dashboard is the one I needed so far. Now I have disabled the dietpi dashboard and it now works without problems. Best for me is to wait for stable version of dietpi dashboard. Thank you very much for this project.

kd9352, 2021-12-12:
Anyway, here's my output.

-rw-r--r-- 1 root     root     458 Dec 12 01:04 Connectd_192.168.1.240_web80.log
-rw-r--r-- 1 root     root     436 Dec 12 01:04 Connectdrmt365535.log
prwx------ 1 jellyfin jellyfin   0 Dec 12 01:08 clr-debug-pipe-4158-27981-in
prwx------ 1 jellyfin jellyfin   0 Dec 12 01:08 clr-debug-pipe-4158-27981-out
srw------- 1 jellyfin jellyfin   0 Dec 12 01:08 dotnet-diagnostic-4158-27981-socket
drwx------ 3 root     root      60 Dec 12 01:03 systemd-private-93acd6214b1847a9828b5ec3d3801a5b-haveged.service-gqkgvj
drwx------ 3 root     root      60 Dec 12 01:08 systemd-private-93acd6214b1847a9828b5ec3d3801a5b-redis-server.service-lrQXcj

@MichaIng

MichaIng, 2021-12-12:
Okay, then indeed it seems to have been a doubled instance in the dashboard itself somehow. @ravenclaw900 probably it can occur when doing software installs/uninstalls on the Software tab and shortly after calling `dietpi-software` in the Terminal tab, or the other way round? Similarly with the Services tab? I haven't tested how quickly they are terminated, also in case a DietPi script get's SIGKILLed, its working directory, which serves as concurrency flag as well, remains.

kd9352, 2021-12-12:
Hey! One more question, I just installed fuguhub to test. But when I try to uninstall it I get exit code 8. How can I solve this? Command: userdel bd  
                                                     │
│  - Exit code: 8                                                              │
│  - DietPi version: v7.9.3 (MichaIng/master) | HW_MODEL: 21 | HW_ARCH: 10 |   │
│ DISTRO: 6                                                                    │
│  - Image creator: DietPi Core Team                                           │
│  - Pre-image: from scratch                                                   │
│  - Error log:                                                                │
│ userdel: user bd is currently used by process 13374

Joulinar, 2021-12-12:
can you share following

`ps -ef | grep 13374`

kd9352, 2021-12-12:
> ps -ef | grep 13374

root@DietPi:/home/dietpi# ps -ef | grep 13374
bd         13374       1  0 12:52 ?        00:00:01 /home/bd/FuguHub -d -ubd -h/home/bd -r/home/bd/disk --threads 3 --sessions 100 --connections 80 --pidfile FuguHub
bd         13601   13374  0 12:52 pts/1    00:00:00 /usr/bin/rsync --daemon --no-detach --config /tmp/zzbdrsyncd.conf --address 127.0.0.1 --port 9334
root       15131   14644  0 13:03 pts/0    00:00:00 grep 13374

Joulinar, 2021-12-12:
can you try followig

```
systemctl unmask bdd
systemctl disable --now bdd
ps -ef | grep FuguHub
```

kd9352, 2021-12-12:
> can you try followig
> 
> ```
> systemctl unmask bdd
> systemctl disable --now bdd
> ps -ef | grep FuguHub
> ```
root@DietPi:/home/dietpi# systemctl unmask bdd
root@DietPi:/home/dietpi# systemctl disable --now bdd
Failed to disable unit: Unit file bdd.service does not exist.
root@DietPi:/home/dietpi# systemctl disable --now bdd
Failed to disable unit: Unit file bdd.service does not exist.
root@DietPi:/home/dietpi# ps -ef | grep FuguHub
bd         13374       1  0 12:52 ?        00:00:03 /home/bd/FuguHub -d -ubd -h/home/bd -r/home/bd/disk --threads 3 --sessions 100 --connections 80 --pidfile FuguHub
root       16342   14644  0 13:15 pts/0    00:00:00 grep FuguHub

I got error on second command.

Joulinar, 2021-12-12:
even that the service has been stopped and removed, the program keeps running. 🤔 

can you try to kill it

```
kill 13374
ps -ef | grep FuguHub
```

kd9352, 2021-12-12:
> even that the service has been stopped and removed, the program keeps running. 🤔
> 
> can you try to kill it
> 
> ```
> kill 13374
> ps -ef | grep FuguHub
> ```

Yeah its weird. :D I think its killed.

root@DietPi:/home/dietpi# kill 13374
root@DietPi:/home/dietpi# ps -ef | grep FuguHub
root       16607   14644  0 13:23 pts/0    00:00:00 grep FuguHub

Joulinar, 2021-12-12:
now this should work

`dietpi-software uninstall 161`

MichaIng, 2021-12-12:
Hmm, it is the SysV service shipped by FuguHub itself. I checked that one and it looks good, actually, I mean as good as a SysV service can look 😉.

We could create a proper systemd unit, but not sure whether it's worth it. At least there indeed was a little update (no actual new version/release) just this 4th November, so the project is not dead, but it is barely maintained.

I'll see if I we can apply a fix the one or the other way and in this turn offer that fix to the FuguHub developer, as a sneaky/indirect project status check 😎.

MichaIng, 2021-12-20:
I can't believe it. I just removed the `systemctl disable --now` workaround from Stretch, and now it turns out that at least for purge SysV services it does not stop the service either. The workaround was as I recognised that only on Stretch the `--now` does not do what it is supposed to do, stopping the service also when it was disabled already and when there is an _additional_ SysV service which systemd syncs the status with automatically. But in case there is _only_ a SysV service it even does not work on Bullseye. Will retest on Bookworm and in case do an upstream report. To fix it at least for FuguHub I'll create a systemd unit anyway.
_________
I think we need to replace the installer as well:
```sh
...
echo "Are you installing on an online VPS/cloud server or on a home computer?"
echo "Select 'yes' (y) for online, 'no' (n) for private home/business network"
read -p "(y/n)?"
if [ "$REPLY" == "y" ]; then
    export ISCLOUD=true
    echo "********** Updating, please wait.... *****"
    apt-get -q=2 update
    read -p "Remove unnecessary/conflicting packages such as Apache (y/N)?"
    if [ "$REPLY" != "y" ]; then
        echo "********** Removeing Apache, if installed. *****"
        apt-get -q=2 --purge remove apache2 
        apt-get -q=2 --purge remove apache2-utils
        apt-get -q=2 --purge remove apache2-doc
        apt-get -q=2 --purge remove apache2.2-bin
        echo "********** Removing stuff we do not need  *****"
        apt-get -q=2 --purge remove samba
        apt-get -q=2 --purge remove smbfs
        apt-get -q=2 --purge remove lynx
        apt-get -q=2 --purge remove lynx-cur
    fi
#Keeps time updated
    apt-get -q=2 install ntp

#rsync is used by the backup plugin
    apt-get -q=2 install rsync
fi
...
```
When selecting `y`, without asking it purges Apache2, the Samba server, the lynx command line browser and installs NTP... I tested it a bunch of times but never selected `y` (VPS/cloud/online) at this point, now I see I should have 😉.

MichaIng, 2021-12-20:
I removed the official installer now, which is very outdated, inconsistent between x86_64 and ARM, has a broken reinstall detection, in cases installs (with inconsistent conditions) a backup plugin which works only with a single commercial cloud provider, while weblinks in FuguHub about it are broken. Here now a fully automated, consistent and slim solution: #5102

The install count for FuguHub is very low and the project nearly unmaintained, so this is my last effort on this software option unless install count goes significantly up and/or project shows significant progress.

kd9352, 2021-12-20:
> 

Thank you so much for fixing the issues. :D

closed 2022-01-02 by MichaIng