MichaIng/DietPi #5059: Iteration v8.0 (release plan)

MichaIng, 2021-12-12:
## Day 0: Init new development cycle
To be done until: `2021-12-13`
- [ ] Create a new GitHub milestone https://github.com/MichaIng/DietPi/milestone/110
- [ ] Select issues for the new iteration in the new milestone
- [ ] Create a new section in the [DietPi Wiki](https://github.com/MichaIng/DietPi/wiki/Iteration-plans)
- [ ] A new development cycle starts with a subversion increment, or major version increment when we drop support for an old Debian version or old DietPi versions. The RC version during the development cycle is "-1", to clearly identify unstable version strings: `v7.9.3` =>` v8.0.-1`.
- [ ] Init [`CHANGELOG.md`](https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt)
- [ ] Add support to [DietPi-Survey_report](https://github.com/MichaIng/DietPi/blob/dev/.meta/dietpi-survey_report)

## First week
To be done until: `2021-12-18`
- [ ] Do a Twitter retweet (possible other platform) about the last release
- [ ] Make the iteration planning - a little list of important items that shall be addressed within this release and add them to the new GitHub milestone
- [ ] Add it to the [DietPi Wiki](https://github.com/MichaIng/DietPi/wiki/Iteration-plans)
- [ ] Share this on Twitter, Forum, GitHub, Petreon, etc. to inform about the plans and allow any user to give some feedback on it.
- [ ] Share this on Facebook (DietPi group) to inform about the plans and allow any user to give some feedback on it.

### NOTE: The below dates likely get deferred due to Christmas an new years holidays!

## 3rd week: Beta phase
To be done until: `2022-01-01`
- [ ] Merge live patches from `master` and remove in `dev`.
- [ ] Create [pull requests](https://github.com/MichaIng/DietPi-Docs/pulls) for all [required documentation changes](https://github.com/MichaIng/DietPi-Docs/issues?q=is%3Aopen+is%3Aissue+label%3A"Next+DietPi+release"):

To be done at: `2022-01-01`
- [ ] Launch the initial beta as `.0` RC version
- [ ] Advertise on GitHub
- [ ] Advertise on [DietPi Forum](https://dietpi.com/phpbb/viewforum.php?f=9)
- [ ] Advertise on DietPi-Banner [MOTD](https://dietpi.com/motd)

To be done at: `2022-01-02`
- [ ] Advertise on website slider: https://dietpi.com/
- [ ] Advertise on Twitter ([DietPi account](https://twitter.com/DietPi_))
- [ ] Advertise on Facebook ([DietPi group](https://www.facebook.com/groups/167403007229675))

To be done from: `2022-01-02` - `2021-01-06`
- [ ] Address bug reports with further beta iterations, e.g. `.1` and `.2` RC versions

## 2 days before the release
To be done until: `2022-01-06`
- [ ] Have a release PR open on the main repo which needs to be linked from the [changelog](https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt)/release notes.
- [ ] Assure that [update server version](https://github.com/MichaIng/DietPi/dev/beta/.update/version) and [DietPi-Globals default version](https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-globals) match.
- [ ] Have the docs release PR ready and reviewed, with release notes and required changes merged into dev already.
- [ ] Have a PR on the website ready to advertise the new release via website slide and have the docs PRs all ready and reviewed.

## Release
To be done at: `2022-01-08`
- [ ] Make the release
- [ ] Publish the release notes in [DietPi Docs](https://dietpi.com/docs/releases/)
- [ ] Mention the release notes in [DietPi Forum](https://dietpi.com/phpbb/viewforum.php?f=10)
- [ ] Publish on website slider: https://dietpi.com/
- [ ] Update the [MOTD](https://dietpi.com/motd)
- [ ] Update software list in [DietPi Wiki](https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list)

## Release announcements
To be done at: `2022-01-09`
- [ ] Publish on Twitter ([DietPi account](https://twitter.com/DietPi_))
- [ ] Publish on Facebook ([DietPi group](https://www.facebook.com/groups/167403007229675))
- [ ] Announce on [Hacker News](https://news.ycombinator.com/)
- [ ] Announce on [ComputerBase](https://www.computerbase.de/). Contact user [SV3N](https://www.computerbase.de/forum/members/sv3n.774722/)
- [ ] Determine if a hotfix release is needed

## Prep for the _next_ release
To be done until: `2022-01-10`
- [ ] Create a new _release plan_

ravenclaw900, 2021-12-12:
Is it increasing to v8 instead of going through v7.10, or is there a big change that the major version is increasing for?

MichaIng, 2021-12-12:
Yes the idea was to increment the major version whenever support for a Debian version is dropped, Stretch in this case. Also I couldn't resist as of the nice coincidence that 8.0 follows on 7.9 (no two digit minor version), v**8**.0 release on 0**8**.01 (at least if that release date stays despite holidays) and a new major version on a new year. Feels fresh 😄.

closed 2022-01-09 by MichaIng