MichaIng/DietPi #5060: User directory permissions and homeassistant

xmicky, 2021-12-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.8
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`  = 0
Kernel version | `uname -a`

Linux Pi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Rpi 4
- 5V Li battery pack
- SDcard used | (EG: SanDisk ultra) passes raspian speed tests

Homeassistant:
I want to comment on the read write pemissions for the dietpi user directories.
Noteably when using Homeassistant with HACS installed.
HACS is unable to install any integrations into homeassistant because of the folder permissions.
I was also unable to edit the homeassistant configuration.yaml via the samba server for the same reason.
I just set all the folder permissions for the user directories to rw.
For me, I need to be able to have read-write assess to the homeassistant user directories for headless operation.

No action needed for me.

MichaIng, 2021-12-12:
Many thanks for your report.

The Home Assistant install and data directory is of course only directly R/W accessible to the `homeassistant` user which runs the service and web interface. How did you try to install HACS? Can't it be selected from within the Home Assistant interface? That should work OOTB. If you need to install something manually to `/mnt/dietpi_userdata/homeassistant`, please do that as `root` user or via `sudo`, then assure that the service user keeps having full access to the newly added files:
```sh
sudo chown -R homeassistant:homeassistant /mnt/dietpi_userdata/homeassistant
```

Hmm, editing such a configuration file via Samba is not the way how I would do it, but via direct edit from console instead. However, if you want/need to do something like that on a regular basis, you can either setup a dedicated Samba share for the `homeassistant` user or grant `dietpi` write access:
```sh
sudo chmod -R g+rw /mnt/dietpi_userdata/homeassistant
sudo usermod -aG homeassistant dietpi
```
Or do the `chmod` for the individual files/directories only where you need write access via Samba to.

xmicky, 2021-12-12:
Hi,

For all homeassistant installations apart from homeassistant os,
HACS files have to be installed separately outside of homeassistant. Hacs
files can be installed using install from GitHub or by download and copy to
a folder in the HomeAssistant folder.

( I have a separate rpi4 with HomeAssistant os running and in HomeAssistant
os all can be done from HomeAssistant dashboard including file editing)

HACS files are written to a folder called custom_components in the
HomeAssistant user data folder. I did this with root access via the console.

In all cases Hacs integrations are installed from the HomeAssistant
dashboard. In the DietPi instance folder permissions prevent Hacs
integrations from installing.

I just changed folder permissions to get it working.

As ONLY HomeAssistant OS has ability to allow edit of HomeAssistant files -
config etc there is a need to do this. Home assistant OS has rw permissions
allows editing of files by its internal samba server.

Those using home DietPi HomeAssistant will quite probably be using a stand
alone system so only SSH and samba. Easy to samba from my phone from
anywhere!

Why do I not just use HomeAssistant OS? I'm exploring alternatives. My Asus
router keeps my file serving SSD very hot all the time. My HomeAssistant OS
rpi4 has an SSD and a samba server and that is running cool. My Asus router
has a torrent client which will only write to the attached SSD.
Homeassistant OS doesn't have a torrent client so i can't just dump the
Asus connected SSD. In diet pi I can have HomeAssistant, samba server and a
torrent client allowing me to remove the SSD from my Asus router.

My laptop doesn't come out often because the hot, high-humidity, tropical,
salt-laden air destroys the internals!

Mick

MichaIng, 2021-12-12:
Okay, so then it's clear: Move the HACS into `/mnt/dietpi_userdata/homeassistant` and then reapply R/W access for the service user:
```sh
sudo chown -R homeassistant:homeassistant /mnt/dietpi_userdata/homeassistant
```
> Home assistant OS has rw permissions
allows editing of files by its internal samba server.

Totally makes sense, as it is a OS specifically/exclusively for Home Assistant. DietPi however is a multi purpose OS with the explicit aim that you can install and run every other software concurrently. In this case it would be a bad security and stability concept when the Samba server (generally network shares) had direct write access to other programs data/config directories. For media files it is fine, but when e.g. one accidentally touches a MariaDB database file from a Windows system, database access is broken easily, leading to critical data loss and service breakage. Service directories by intention are writable to the dedicated service user only (and root, of course) so that nobody/nothing else can, by intention or accidentally, break that service. If for specific reasons it is required differently, then this must be setup manually, which then somehow implies the understanding of the UNIX permissions and the underlying basic security concept 😉.

xmicky, 2021-12-12:
ONLY Homeassistant OS has internal samba. For ALL non HomeAssistant OS
installations home assistant doesn't have the samba facilities or editing
facilities. It relies on the host OS.

MichaIng, 2021-12-12:
Samba is pretty common and pre-installed and many, even "light" OS images. Also on HA OS it is a dedicated server process, just like everywhere else, only configured to allow R/W access to the HA data directory.

Okay, I think we cleared the two permission issues/questions here. HACS install on DietPi might be something worth a tab in our documentation, what you think? https://dietpi.com/docs/software/home_automation/#home-assistant

thomasmockridge, 2022-01-05:
I would really like to install HACS on dietpi - @xmickey did you succeed in your setup? 
Did you by any chance document it somewhere?
I have installed Home Assistant but the dietpi install does not have HACS or Supervisor panel enabled by default.

Can you give more detail to "Hacs files can be installed using install from GitHub (url?) or by download (from where) and copy to a folder in the HomeAssistant folder." (/mnt/dietpi_userdata/homeassistant/xxxx)

thanks
Thomas

Joulinar, 2022-01-05:
We have a longer discussion for HA Supervised on our forum. This include some hints how to get it running.

https://dietpi.com/phpbb/viewtopic.php?t=8385

xmicky, 2022-01-05:
HACS can be installed on homeassistant in DietPi.

Just download the HACS files from Git.

Create a new folder called *custom_components* in the homeassistant folder
called *config* (you can check you have the correct location because you
should see the file *configuration.yaml) *create a folder *hacs* within
*custom_components*

use sudo to copy the files you download from git (un archive as required)
into the *hacs* folder.

You will then have to change the folder permissions for *custom_components* to
*rwxrwxrwx* to allow HACS to use them.  HACS will create new folders in
custom_components as it needs. (I presume you could also join HACS to the
group and just give group rwx permissions)

HACS will then show in HA.

The supervised part is only available for homeassitant OS. It's not needed
for other versions as these have the facilities for updating HA. In the
latest HA the sections which came under supervised menu have been moved and
supervised menu doesn't exist.

I have HA OS running on a Rpi with the HA samba server all on and SSD -
it's been super reliable.
I also have a Rpi with standard Rpi OS with HA (in Docker), Samba and
Qtorrent. Running on an SSD partitioned with a large drive for the server.

xmicky, 2022-01-05:
BTW: you can use chmod +s to set the folder permissions for config folder
and sub folders - this sets the SCIG bit in the permissions which ensures
new subfolders have same permissions. Also allows you modify
configuration.yaml (etc) easily which I do via Samba. - reference Richard
Blum: Linux command line and shell scripting bible page 431,436

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng