MichaIng/DietPi #5061: Docker and its dependants will be deselected Modules for the loaded kernel are missing.

dragonandy, 2021-12-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=7 G_DIETPI_VERSION_SUB=9 G_DIETPI_VERSION_RC=3 G_GITBRANCH='master' G_GITOWNER='MichaIng'`
- Distro version | `bullseye 0`
- Kernel version | `Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)` 
- Power supply used | (5V 3A RAVpower)
- SDcard used | (Samsung EVO Plus U3 128G)

#### Additional Information (if applicable)
- Software title | (Dokcer)
- Was the software title installed freshly or updated/migrated?  yes
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | `ae7fa101-e48a-4965-8893-b82d98537126`

#### Steps to reproduce

1. fresh install of dietpi DietPi_RPi-ARMv8-Bullseye.img 
```hash
FILE:	DietPi_RPi-ARMv8-Bullseye.img
DATE:	Sat Dec  4 17:31:56 CET 2021
MD5:	3c69e2c3735426860864f5fb1a55e9b0
SHA1:	931010db48657368b0203f1825b41016da6e81fa
SHA256:	aa6772367ac107af4705016c9cbbbefb7ce71dd8e35b651833ab7e10b4726109
```
2. run `dietpi-update` to upgrade to the latest version of dietpi
3. run `dietpi-software`  select docker and docker-compose and install 
4. prompt
 ```text
 │ [WARNING] Docker and its dependants will be deselected                                                               │
 │                                                                                                                      │
 │ Modules for the loaded kernel are missing. This is usually the case when the kernel was recently upgraded but a      │
 │ reboot is outstanding. The Docker install however will fail if it cannot load kernel modules.                        │
 │                                                                                                                      │
 │ Please do a reboot first and reselect Docker or its dependant for install.                                           │
```
5. reboot the system
6. repeat step 3 and install docker
7. still prompt me to reboot
8. doing this muiltiple times trying different combination and remove install software but still cant get docker installed

#### Expected behaviour

- docker install correctly

#### Actual behaviour

-  prompt a WARNING that keeps telling me to reboot
 ```text
 │ [WARNING] Docker and its dependants will be deselected                                                               │
 │                                                                                                                      │
 │ Modules for the loaded kernel are missing. This is usually the case when the kernel was recently upgraded but a      │
 │ reboot is outstanding. The Docker install however will fail if it cannot load kernel modules.                        │
 │                                                                                                                      │
 │ Please do a reboot first and reselect Docker or its dependant for install.                                           │
```

#### Extra details

-  I can do install docker with `bash -c "$(curl -fsSL https://get.docker.com)"` or `apt install docker.io`
-  maybe the problem #4959
-  related code maybe here https://github.com/MichaIng/DietPi/blob/0c59b07/dietpi/dietpi-software#L2182-L2194 
```bash
		# Prevent Docker install if kernel has just been upgraded and old one overwritten/purged: https://github.com/MichaIng/DietPi/issues/3126
		if [[ ${aSOFTWARE_INSTALL_STATE[162]} == 1 && ! -d /lib/modules/$(uname -r) ]]
		then
			G_WHIP_MSG '[WARNING] Docker install will be deselected
\nIt seems that your kernel has just been upgraded. The mismatch between loaded and installed kernel versions would cause a failure during Docker install.
\nPlease reselect Docker from dietpi-software after next reboot, or run:
# dietpi-software install 162'
			aSOFTWARE_INSTALL_STATE[162]=0
			# Unmark Docker dependants as well
			(( ${aSOFTWARE_INSTALL_STATE[134]} == 1 )) && aSOFTWARE_INSTALL_STATE[134]=0
			(( ${aSOFTWARE_INSTALL_STATE[185]} == 1 )) && aSOFTWARE_INSTALL_STATE[185]=0
			(( ${aSOFTWARE_INSTALL_STATE[86]} == 1 )) && aSOFTWARE_INSTALL_STATE[86]=0
		fi
```
-  the loaded kernel should be the installed kernel but I still cant install docker
```bash
root@DietPi:~# ls /lib/modules/
5.10.63-v8+
root@DietPi:~# uname -r
5.10.63-v8+
```

Joulinar, 2021-12-12:
Hi,

Many thanks for your message. We would need to have a look. Probably an issue on the kernel module detection.

Joulinar, 2021-12-12:
the line in question is this one

https://github.com/MichaIng/DietPi/blob/773e0c6e959227c17bde3a3c936e82ad057e3489/dietpi/dietpi-software#L2046

I guess there are brackets missing and it should looks like this

`if [[ ( ${aSOFTWARE_INSTALL_STATE[162]} == 1 || ${aSOFTWARE_INSTALL_STATE[134]} == 1 || ${aSOFTWARE_INSTALL_STATE[185]} == 1 || ${aSOFTWARE_INSTALL_STATE[86]} == 1 ) && ! -d /lib/modules/$(uname -r) ]]`

MichaIng, 2021-12-12:
Many thanks for reporting.

It is this inconsistent logic where the last `&&` is not followed if any of the previous conditions was true. Outside of `[[ ]]` it would be followed as intended. I remember I stumbled upon this one time already. Strange that we didn't do a single Docker install since this was merged.

Splitting it in two `[[ ]]` or the parenthesis do work. The first is easier to ship with a live patch, the second a cleaner final solution for next release, I'd say.

MichaIng, 2021-12-12:
First of all the regular fix for next release: https://github.com/MichaIng/DietPi/commit/ccccf26c

MichaIng, 2021-12-12:
Live patch merged: #5065

closed 2021-12-12 by MichaIng