MichaIng/DietPi #5062: Dietpi automated install fails - File does not exist or cannot be written by current user (/boot/dietpi.txt)

nigobo, 2021-12-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
dietpi@DietPi:~$ cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
dietpi@DietPi:~$ echo $G_DISTRO_NAME $G_RASPBIAN
bullseye 0
```
- Kernel version | `uname -a`
```
dietpi@DietPi:~$ uname -a
Linux DietPi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
dietpi@DietPi:~$ echo $G_HW_MODEL_NAME
RPi 3 Model B (armv7l)
```
- Power supply used | (EG: 5V 1A RAVpower)
```
https://www.kjell.com/se/produkter/kontor/hemmakontor/grenuttag/schneider-electric-grenuttag-med-brytare-och-usb-uttag-5-vags-svart-p67750
```
- SDcard used | (EG: SanDisk ultra)
```
SandDisk Ultra 16Gb
```

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Downloaded `DietPi_RPi-ARMv7-Bullseye.7z` from dietpi.com
2. Extracted the 7z 
3. Flashed it using `Raspberry Pi Imager`, flash and verifikation ok
4. Mounted the SD-card
5. Edited `dietpi.txt` and changed
```
AUTO_SETUP_NET_WIFI_ENABLED=1
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_INSTALL_SOFTWARE_ID=122
```
6. Inserted the SD card and booted the pi
7. Waited...
8. Connected via SSH

#### Expected behaviour

- The software (Node-RED) should be installed and be accessible from a browser

#### Actual behaviour

- Nothing seems to happen, but when I log in to the Pi using SSH I get the following error-message.
[image]
- I have tried editing the `dietpi.txt` using Sublime Text, and using VI but the results are the same.
- Most of the script seems to actually have run but not all and the installation process seems to stop here

#### Extra details

[dietpi-firstboot.log](https://github.com/MichaIng/DietPi/files/7698375/dietpi-firstboot.log)
[dietpi-ramlog.log](https://github.com/MichaIng/DietPi/files/7698376/dietpi-ramlog.log)

And **a big thanks** for all the great work you do. Dietpi is really awesome!

Joulinar, 2021-12-12:
Do you use `root` user to login?

nigobo, 2021-12-12:
@Joulinar I used `dietpi` user when I logged in

Joulinar, 2021-12-12:
Pls use `root` on first login

MichaIng, 2021-12-12:
Another good reason to remove the `dietpi` user in favour of an optional login user creation prompt _after_ first run setup has finished 😄. Either the update or installs during automated first run setup failed. When you login as `root` user, you can basically take it over an retry interactively, i.e. you get error prompts in case of a failure, where you can handle and retry the failed command.

I'll also add a root user check and a prompt to login as root user login when first run setup has not been finished yet.

nigobo, 2021-12-12:
Hi! I just made another try. This time it seemed to work. The Node-RED install was online when I tried browsing to it. And logging in as root, everything looks good. 

Unfortunately I can't figure out what I did wrong. This was the fifth attempt with what I would say the same settings, the only difference was that I logged in using `root`. But the Node-RED wasn't responding at port `1880` last time so I don't believe that it made any difference since that was the first thing I tried every time.

Will make another try later today and maybe I can shed some light on what I did wrong.

Btw, thanks for your quick replies!

MichaIng, 2021-12-12:
Okay great. When it fails again on next attempt, please do not repeat first run setup but exit to console and check the logs at:
```sh
cat /var/tmp/dietpi/logs/dietpi-update.log
cat /var/tmp/dietpi/logs/dietpi-firstrun-setup.log
```
So we can see which part actually failed.

nigobo, 2021-12-12:
Ok. Will try. The strange thing is that when it failed there were only the
two logs I posted in my first comment. There were no dietpi-update.log.

MichaIng, 2021-12-12:
> There were no dietpi-update.log.

Then the network connection check failed or timed out.

MichaIng, 2022-01-04:
Were you able to reproduce the issue?

DylanDelobel, 2022-01-07:
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=3
```
Fresh new install from 3 days ago, when I log into `root` I got the install every time with this message

[image]

MichaIng, 2022-01-07:
@DylanDelobel 
Thanks for reporting.

You can simply ignore this and continue, it does not affect the outcome. But did you also download the newest image 3 days ago? If so which one exactly? All images have been repacked actually to fix the prompt.

DylanDelobel, 2022-01-10:
@MichaIng I checked the image and your right my version is not the latest, you can probably close this issue

MichaIng, 2022-01-10:
Okay great, many thanks for the feedback.

closed 2022-01-10 by MichaIng