MichaIng/DietPi #5063: DietPi-Dashboard Migration failed running on old Port

Arghh, 2021-12-12:
Hi,

after updating to DietPi 7.9.3 today and choosing to migrate DietPi-Dashboard from Port 8088 to 5252 the Dashboard still gets served on Port 8088 although my config got changed at /opt/dietpi-dashboard:

# TCP network port
port = 5252

# TLS for HTTPS
tls = false
cert = "/path/to/cert"
key = "/path/to/key"

Reboot and Restarting the Service had no effect. Im on latest Bullseye OS Harware RasperryPi 3 Model B.

Any ideas why the settings have no effect?

Joulinar, 2021-12-12:
Can you try to reinstall dashboard pls.

Arghh, 2021-12-12:
dietpi-software reinstall 200 did the trick. Dashboard runs on 5252 und got upgraded from 0.3.0 to 0.3.2.

why not add the Dashboard as a DietPi controlled service?

Joulinar, 2021-12-12:
> why not add the Dashboard as a DietPi controlled service?

As `dietpi controlled` services will be stopped on software installation. Which is a bad idea for the Dashboard as you can trigger installation out of it as well.

@MichaIng 
I guess we need to force a reinstall on migration to ensure latest dashboard running.

MichaIng, 2021-12-12:
Hmm, alone that the setting and the comment above were present shows that this should have worked already with your version. Also the patch is only offered when the setting with port `8088` is explicitly present. Not sure if there was an intermediate bug where the port setting was not respected. @Arghh did you initially install the stable or nightly version?

Arghh, 2021-12-12:
stable. 0.3.0 was stable at the time.

ravenclaw900, 2021-12-12:
Ah, this was fixed with https://github.com/ravenclaw900/DietPi-Dashboard/commit/5965551fec41d355749c92c9c743c75b536bd815. Technically, the config file doesn't work until v0.3.2, then.

Joulinar, 2021-12-12:
In this case we would need to force the reinstall to ensure a version higher v0.3.2

MichaIng, 2021-12-12:
Ah lol, and we didn't even recognise as we always ran with defaults 😄. I'll update the patch accordingly, as well on master branch.

MichaIng, 2021-12-12:
Fixed with: https://github.com/MichaIng/DietPi/commit/c3fdfdf
And backported to master branch: https://github.com/MichaIng/DietPi/commit/efb38cc

Joulinar, 2021-12-13:
Guys not sure if it is an issue but after reinstall of DietPi dashboard, the service is not restarted as it is not controlled by DietPi. Means the port change is not taken into account 😉 

for testing I installed v0.3.0

before `dietpi-update`

```
root@DietPi4:~# cat /opt/dietpi-dashboard/config.toml
# TCP network port
port = 8088

# TLS for HTTPS
tls = false
cert = "/path/to/cert"
key = "/path/to/key"
root@DietPi4:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=375,fd=4))
tcp   LISTEN 0      128          0.0.0.0:8080      0.0.0.0:*    users:(("dietpi-dashboar",pid=525,fd=9))
tcp   LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=375,fd=5))
root@DietPi4:~# journalctl -u dietpi-dashboard.service
-- Journal begins at Mon 2021-12-13 11:07:22 CET, ends at Mon 2021-12-13 11:08:15 CET. --
Dec 13 11:07:45 DietPi4 systemd[1]: Started Web Dashboard (DietPi).
Dec 13 11:07:45 DietPi4 dietpi-dashboard[525]: 2021-12-13 11:07:45,564 INFO  [warp::server] Server::run; addr=0.0.0.0:8080
Dec 13 11:07:45 DietPi4 dietpi-dashboard[525]: 2021-12-13 11:07:45,564 INFO  [warp::server] listening on http://0.0.0.0:8080
Dec 13 11:07:46 DietPi4 dietpi-dashboard[525]: 2021-12-13 11:07:46,092 INFO  [dietpi_dashboard] Request to /ws
root@DietPi4:~# dietpi-update
```

`dietpi-update` log

```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches

[ INFO ] DietPi-Update | Current version : v7.8.2
[ INFO ] DietPi-Update | Latest version  : v7.9.3
[ INFO ] DietPi-Patch | Patching to DietPi v7.9...
[  OK  ] DietPi-Patch | sed -Ei -e s/INDEX_(LOGGING|WEBSERVER|DESKTOP|BROWSER)_CURRENT/INDEX_\1/g -e /INDEX_(LOGGING|WEBSERVER|DESKTOP|BROWSER|SSHSERVER|FILESERVER)_TARGET/d -e /INDEX_(SSHSERVER|FILESERVER)_CURRENT/d /boot/dietpi/.installed
[  OK  ] DietPi-Patch | Setting in /opt/dietpi-dashboard/config.toml adjusted: port = 5252
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[ SUB2 ] DietPi-Software > Automated reinstall
[  OK  ] DietPi-Software | Reinstalling DietPi-Dashboard (beta): Official lightweight DietPi web interface (Rust)
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[ SUB2 ] DietPi-Software > Checking for conflicts and missing inputs
[ SUB2 ] DietPi-Software > Checking for prerequisite software
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB3 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : cron
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
[  OK  ] DietPi-Software | APT update
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB2 ] DietPi-Software > Installing DietPi-Dashboard (beta): Official lightweight DietPi web interface (Rust)
[  OK  ] DietPi-Software | Checking URL: https://github.com/ravenclaw900/DietPi-Dashboard/releases/download/v0.3.2/dietpi-dashboard-aarch64
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://github.com/ravenclaw900/DietPi-Dashboard/releases/download/v0.3.2/dietpi-dashboard-aarch64 -o dietpi-dashboard-aarch64
[ INFO ] DietPi-Software | Updating file: /opt/dietpi-dashboard/dietpi-dashboard
[  OK  ] DietPi-Software | mv dietpi-dashboard-aarch64 /opt/dietpi-dashboard/dietpi-dashboard
[  OK  ] DietPi-Software | chmod +x /opt/dietpi-dashboard/dietpi-dashboard
[... 29 lines cut ...]
```

after `dietpi-update` to v7.9.3

```
root@DietPi4:~# cat /opt/dietpi-dashboard/config.toml
# TCP network port
port = 5252

# TLS for HTTPS
tls = false
cert = "/path/to/cert"
key = "/path/to/key"
root@DietPi4:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=375,fd=4))
tcp   LISTEN 0      128          0.0.0.0:8080      0.0.0.0:*    users:(("dietpi-dashboar",pid=525,fd=9))
tcp   LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=375,fd=5))
root@DietPi4:~# journalctl -u dietpi-dashboard.service
-- Journal begins at Mon 2021-12-13 11:07:22 CET, ends at Mon 2021-12-13 11:10:17 CET. --
Dec 13 11:07:45 DietPi4 systemd[1]: Started Web Dashboard (DietPi).
Dec 13 11:07:45 DietPi4 dietpi-dashboard[525]: 2021-12-13 11:07:45,564 INFO  [warp::server] Server::run; addr=0.0.0.0:8080
Dec 13 11:07:45 DietPi4 dietpi-dashboard[525]: 2021-12-13 11:07:45,564 INFO  [warp::server] listening on http://0.0.0.0:8080
Dec 13 11:07:46 DietPi4 dietpi-dashboard[525]: 2021-12-13 11:07:46,092 INFO  [dietpi_dashboard] Request to /ws
root@DietPi4:~#
```

MichaIng, 2021-12-13:
Yes that is expected, same as with SSH and VNC servers e.g. The update or reinstall could have been done via DietPi-Dashboard itself, hence we must not force a service restart, which could break things in the middle, but leave it to the user, or next restart. We could however add an info to the prompt during the upgrade about this:
```sh
NB: You need to either restart the system or the service for this change to take effect:
systemctl restart dietpi-dashboard
```

Joulinar, 2021-12-13:
I support the idea of playing a note for the user. 👍

MichaIng, 2021-12-18:
Done: https://github.com/MichaIng/DietPi/commit/936b897

closed 2021-12-18 by MichaIng