MichaIng/DietPi #5064: Backup inside zip/image on ntfs

Lycidias93, 2021-12-12:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...
No
#### Describe the solution you'd like:

- ...
Creating a backup inside a image or zip with symlinks and so on inside file in a zip archive. So we could store the backup on a ntfs hdd
#### Describe alternatives you've considered:

- ...
-
#### Additional context

- ...
-
#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2021-12-12:
We already though about and discussed this feature, but decided against it: #816
With the new backup archive feature, it seems to make more sense, though instead of using 3rd party tools to achieve this, I'd then more likely go with the setup of a Btrfs partition with native filesystem compression support so that the logic of the script can remain simple.

In any case, please use the linked issue for discussion, I'll close this one.

closed 2021-12-12 by MichaIng

Lycidias93, 2021-12-12:
The dedicated partition was the hint, I didn't consider. I will do it just like that, thank you 👍