MichaIng/DietPi #5066: Net negotiated speed goes down after some time

Fabbro03, 2021-12-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=7 G_DIETPI_VERSION_SUB=9 G_DIETPI_VERSION_RC=3 G_GITBRANCH='master' G_GITOWNER='MichaIng' G_LIVE_PATCH_STATUS[0]='applied'`
- Distro version | `buster`
- Kernel version | `Linux PoloID-server 4.14.241+ #1 SMP PREEMPT Wed Jul 28 16:55:16 UTC 2021 armv7l GNU/Linux`
- SBC model | `Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)`

#### Steps to reproduce
Simply wait

#### Expected behaviour
Constant negotiated 1Gbps

#### Actual behaviour
After some time the speed goes down to 100Mbps even in net speed benchmark don't return at 1Gbps.
Only after unplugging and replugging the cable, it renegotiates at 1Gbps

Fabbro03, 2021-12-12:
Problem posted also in the forum: [https://dietpi.com/phpbb/viewtopic.php?t=9785](url)

MichaIng, 2021-12-12:
Please keep posting issues only on one platform, else discussion, information and solutions get split across both, which slows down the investigation and makes it more difficult for others to follow, who may suffer from the same issue.

However, now you are here, so let's keep discussion here.

Can you describe how exactly you observed and then tested that the network connection speed went down persistently? Is it really the Ethernet link speed which is set down to 100 Mbps, visible via `ethtool eth0`, or does this start with and remain at 1 Gbps and only the actual transfer rate goes down?

If it is the Ethernet link speed, please paste the output of
```sh
dmesg | tail -20
```
after it dropped.

Fabbro03, 2021-12-21:
I investigated a bit. And, yes, it was about the link speed. The dmesg output showed nothing outside the times I had disconnected and reconnected the cable.
With further investigation, I noticed that the Ethernet controller was really hot to the touch. So I put an heatsink on it and the problem to date has not recurred.
In the end, I think the problem was the Ethernet controller that was thermal throttling.
Sorry for the inconvenience.

MichaIng, 2021-12-24:
Interesting that the onboard (?) Ethernet has such a thermal throttling implemented, and good to know. Glad you solved it. Happy holidays and new year 🙂.

I'll mark this issue as closed then. Feel free to reopen when required.

closed 2021-12-24 by MichaIng