MichaIng/DietPi #5067: Freeze after logging in via SSH on Pi 4 reboot

yslupdates, 2021-12-12:
- DietPi version | 7.9.2
- Distro version | debian
- Kernel version | `uname -a'
- SBC model | Raspberry Pi 4 4GB
- Power supply used | Officially branded Pi 4 usb c power
- SDcard used | 32gb samsung MB-ME32GA/AM

#### Steps to reproduce
While logged into SSH, I just type 'reboot' command.
Reboots quick, but after entering in password I get it to freeze for minutes like screenshot below
[image]

#### Expected behaviour
Should immediately go to the dietpi screen after logging in where it shows MOTD and other items

#### Actual behaviour
Freezes after entering in password, but eventually works maybe half an hour later

MichaIng, 2021-12-12:
Many thanks for your report.

I had a look through our bug report upload:
- Not sure whether it's related but the drive which contains the DietPi userdata seems to be not mounted.
- Also UFW is blocking outgoing IGMP requests. Not sure which server sends them, definitely not the SSH server, but if some service is using multicast communication in the LAN to share information, then you may want to allow them.

And here it most likely is: Check out:
```sh
journalctl -u mnt-kyrie.mount
```
You see a bunch of failed mount attempts, timing out after 90 seconds then repeating. So my guess is that on login some tasks try to access something on that mount point, probably within `/mnt/dietpi_userdata` which is pointing to it, and every single attempt to access that mount point triggers systemd,automount to try mounting that drive with a 90 seconds timeout, stacking up to the ~15 minutes you mention.

Note that `/mnt/dietpi_userdata` is expected to be present, it is actively used by Jackett, Sonarr, Radarr, Jellyfin, ... so I guess all of them are not working properly either. Hence either keep that drive attached or move userdata to a drive which remains attached. Since I do not see kernel errors, I guess it is really detached and not a hardware issue.

yslupdates, 2021-12-12:
Okay that makes a lot of sense. I do have my userdata in an 8tb external HDD via usb 3.0. I just changed my userdata to the default location and yeah it boots perfectly normal and fast now. Thanks for the tip!

Regarding the second bullet point, it's probably Jellyfin if it has something to do with casting. UFW is only allowing connections through anyone on local network and whatever connection through tun0 (via dietpi-vpn). Perhaps the requests aren't initially going through because VPN connection not connected yet? Regardless, I don't mind that issue and don't use casting.

closed 2021-12-12 by yslupdates

MichaIng, 2021-12-12:
> I do have my userdata in an 8tb external HDD via usb 3.0.

And was it really detached, or did it indeed for some other reason fail to mount? Usually having userdata (which includes the potentially large and heavy written Jellyfin transcoding cache) on an external drive is a very good idea, an SD card wouldn't stand this long when actively using Jellyfin for HD videos.

> Regarding the second bullet point, it's probably Jellyfin

Ah yes possible, broadcasting itself to clients or so.

> UFW is only allowing connections through anyone on local network

Since these are IGMP, it's naturally received by LAN members only and never passes through to WAN. But obviously its not mission critical, just keep it in mind in case some type of service/server discovery or non-HTTP protocol connections do not work.

Btw, that these systemd.autmount units hang the system for 90 seconds on each access, when the drive is not mounted, or also in case of network drives if the remote server is not available or network down: #4466
Would be much nicer if it gave up immediately if the device does not exist, just printing a meaningful error like `mount` would do and good. However, there is not really an alternative to systemd automounts for external/USB drives and network shares. Probably we can set an immediate or very short timeout for those.

yslupdates, 2021-12-13:
Drive was definitely not detached, guess failed to mount but unsure why. For now userdata is being saved on the microsd card. All good. Thanks again

Joulinar, 2021-12-13:
is the drive powered by an own PSU or just via the USB port?