MichaIng/DietPi #5069: Transmission cache size causes uneven performance

bbsixzz, 2021-12-13:
According to https://github.com/transmission/transmission/wiki/Editing-Configuration-Files the default is `cache-size-mb: 4`

Yet on a fresh install I get `cache-size-mb: 388`

The result is uneven write speeds to external USB drive from 30 MiB/s to 4 KiB/s.

I settled for `cache-size-mb: 32` and it seems to be stable and fine at 30 MiB/s.

`/var/lib/transmission-daemon/.config/transmission-daemon/settings.json`

MichaIng, 2021-12-13:
Many thanks for your report.

The cache size is set to 10% of the physical memory. A large cache especially helps to speed up seeds by holding frequently requested blocks in memory and reduce the amount of disk reads in turn, at least it should. The linked Wiki sadly does not give details.

In case of downloads, whether it is beneficial or not, depends on whether it is used in a smart way or in a brain dead way, and in case of brain dead, on the disk write speed. Brain dead would be: Write the 388 MiB in RAM full during downloads without writing anything to disk, then flush the 388 MiB to disk in one batch while pausing the download (as the cache is full). That would explain your observing, though it doesn't make sense to use a combined (!) cache like this. Smart would be to do downloads to cache and disk writes from cache to disk asynchronously, writing data to disk long before the cache is full (native filesystem cache on Linux implies batching anyway) and basically use the cache to allow continuous full speed download, regardless whether the disk can currently write that data sufficiently fast or not, e.g. when heavy reads or concurrent writes from another application are done. So in any case a larger cache should never be able to slow down downloads, but at best smoothen and speed up the download rate on slower disks and when they are used concurrently by other applications.

I read a bit through forum and GitHub issues and indeed it seems to be the brain dead cache usage. We may open an issue for some discussion on that. Especially since this cache is used for both, downloading and seeding, which is usually done concurrently, it doesn't make any sense to write it full with downloads. If there is really a minimal amount of data wanted that is written in one batch to disk, larger than what the native filesystem cache does anyway, then this should be a separate setting, so that the overall cache can keep holding data for seeding and from downloads while the latter are written to disk in smaller batches 🤔.

bbsixzz, 2021-12-13:
Thank you for the explanation!

MichaIng, 2022-12-30:
Linking a discussion about the Transmission memory cache, answering some questions about the behaviour: https://github.com/transmission/transmission/discussions/2400

MichaIng, 2023-01-01:
Solved with: https://github.com/MichaIng/DietPi/commit/9685f16
The setting is a pure download cache (there is a dedicated upload cache without a setting), in which case the default of 4 MiB should be fine, at least 10% RAM size is much too much.

closed 2023-01-01 by MichaIng