MichaIng/DietPi #5072: DietPi and tailscale

MichaIng, 2021-12-13:
### Discussed in https://github.com/MichaIng/DietPi/discussions/5071

<div type='discussions-op-text'>

<sup>Originally posted by **bbowlsbey** December 13, 2021</sup>
Having a hard time installing tailscale on Raspberry Pi 4 running DietPi.  I have tried both the 32 bit and 64 bit versions, but the the initial curl command quits with an error message that says that apt-key add has been deprecated.  Oddly enough the same curl command and install script works just fine when installing on DietPi 32bit running on a Pi Zero W.  (Cant get the Interface options I want for the Pi Zero though as I need both a USB 1 connection to a printer, and a ethernet connection to run CUPs as a print server).  

Any suggestions?

[Have tried both the <<curl -fsSL https://pkgs.tailscale.com/stable/ubuntu/focal.gpg | sudo apt-key add ->> command, which is where I get the error message that apt-key add is deprecated, and the "install with one command" curl command of <<curl -fsSL https://tailscale.com/install.sh | sh>>, which goes all the way through to the 'X of diskpace would be used.  Would you like to continue Y or n?', but then aborts with out any option to hit enter to accept the default "Y" or type yes.]  
Not sure why this installs like a charm on the current 32 bit version of DietPi running on a Pi zero.</div>

MichaIng, 2021-12-13:
@bbowlsbey
Why you use the Ubuntu key when there is a Debian key? https://pkgs.tailscale.com/stable/#debian-bullseye
Please try:
```sh
curl -sSfL 'https://pkgs.tailscale.com/stable/debian/bullseye.noarmor.gpg' -o /etc/apt/trusted.gpg.d/tailscale.gpg
apt update
```

MichaIng, 2022-02-20:
Did you succeed with the above commands?

bbowlsbey, 2022-02-20:
Sorry for the late reply.

That worked like a charm.  Connecting to a DietPi jumpbox on my remote
network now through Tailscale.

Appreciate your help.

V/R

Bryan Bowlsbey

MichaIng, 2022-02-20:
Okay great. So when we implement into `dietpi-software`, we know how. Software request tracked and voteable in discussions: #5021

I'll mark this issue hence as closed.

closed 2022-02-20 by MichaIng

bbowlsbey, 2022-02-20:
Awesome, thanks!

Appreciate all that you do with dietpi.