MichaIng/DietPi #5073: dpkg error, update from Kodi 19.1 to Kodi 19.3 fails following DietPi update to 7.9.3

raph82, 2021-12-13:
#### Required Information
- DietPi version | v7.9.3 (MichaIng/master)
- Distro version | bullseye (ID=6,RASPBIAN=0)
- Kernel version | `Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- SBC model | RPi 4 Model B (aarch64) (ID=4)
- Power supply used | Raspberry Power supply
- SDcard used | To be updated

#### Additional Information (if applicable)
- Software title | DietPi-Software/Kodi
- Was the software title installed freshly or updated/migrated? OS installed end of November 2021 with the then-current image. Following the [release notes of Dietpi 7.9](https://dietpi.com/docs/releases/#improvements-79), after upgrading I ran `dietpi-software reinstall 31` which failed. I uninstalled Kodi with dietpi-software and tried again, with this error. It keeps failing after reboots.
- Can this issue be replicated on a fresh installation of DietPi? To be updated
- Bug report ID | 07a70817-b1cc-4d94-b1d2-8e3519af71de

#### Steps to reproduce
1. `dietpi-update`
2. `dietpi-software reinstall 31`

#### Expected behaviour
- Kodi is updated to version 19.3

#### Actual behaviour
- DietPi-Software fails updating Kodi

#### Extra details
From dietpi-bugreport :
- Command        | `apt-get -qq --allow-change-held-packages install kodi`
- Exit code      | 100

#### Additional logs:
```
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 41763 files and directories currently installed.)
Preparing to unpack .../kodi_2%3a19.3-1~bullseye_all.deb ...
Unpacking kodi (2:19.3-1~bullseye) ...
dpkg: error processing archive /var/cache/apt/archives/kodi_2%3a19.3-1~bullseye_all.deb (--unpack):
 trying to overwrite '/usr/share/kodi/addons/repository.xbmc.org/addon.xml', which is also in package kodi-repository-kodi 2:19.1+dfsg2-2
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/kodi_2%3a19.3-1~bullseye_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

raph82, 2021-12-13:
This is my first bug report here.
I tried to follow the project guidelines, hope it is helpful.

MichaIng, 2021-12-13:
Many thanks for your report. It seems you installed the `kodi-repository-kodi` manually, which is not compatible and not required for Kodi on RPi (where the addons repo is included):
```sh
apt purge kodi-repository-kodi
dietpi-software reinstall 31
```

MichaIng, 2021-12-14:
The issue seems to be with the Debian package: It contains the `kodi-data` package, but the RPi package does not. So when that `kodi` and `kodi-bin` could be replace seamlessly, `kodi-data` from Debian still leaves files in place which are shipped by `kodi` from the RPi repo 😄. So at best, purging all together:
```sh
G_AGP kodi 'kodi-*'
dietpi-software reinstall 31
```

raph82, 2021-12-14:
[Previous comment](https://github.com/MichaIng/DietPi/issues/5073#issuecomment-993066768) helped to successfully install kodi. A few tweaks though:

**TL;DR**
```
G_SUDO G_AGP kodi 'kodi-*'
dietpi-software install 31
```

Step-by-step:
```
G_AGP kodi 'kodi-*'
[FAILED] Root privileges required. Please run the command with "sudo" or "G_SUDO".
sudo G_AGP kodi 'kodi-*'
sudo: G_AGP: command not found
G_SUDO G_AGP kodi 'kodi-*'
```
I didn't know G_* commands. A quick search didn't returns meaningful results for "G_SUDO". Are they DietPi specific?

```
sudo dietpi-software reinstall 31
[...]
[ INFO ] DietPi-Software | | 31: Kodi is not currently installed
[ INFO ] DietPi-Software | | Use "dietpi-software install 31" to install Kodi.
```
Kodi was uninstalled previously so I needed to adjust the command from `reinstall` to `install`

```
sudo dietpi-software install 31
```
Now Kodi 19.3 is installed! A quick run showed some bugs, like a CEC warning and the Exit Menu showing only one option. For the CEC, I remember having seen a bug report already. I'll look into it later. Thanks for your help!

I don't close the bug with this comment to allow for the investigation to go through. Close it as needed. Thanks again!

Joulinar, 2021-12-14:
> I didn't know G_* commands. A quick search didn't returns meaningful results for "G_SUDO". Are they DietPi specific?

Yes they are

MichaIng, 2021-12-14:
`G_SUDO` is missing in our docs, thanks for the hint. Global functions cannot be executed directly via `sudo`, hence another global function as `sudo` wrapper is required.

Ah yes sorry, since you uninstalled Kodi before, "reinstall" then skips it.

I recognised the CEC warning as well. This is probably something to report to Dom: https://forums.raspberrypi.com/viewtopic.php?t=323303&start=150
There are at least new CEC packages available with RPi support, so generally Kodi should be able to utilise them.

I'll mark this as closed then. Feel free to reopen when required.

closed 2021-12-14 by MichaIng

Joulinar, 2021-12-14:
@MichaIng 
There are users on the forum having black screens on Kodi reinstall now. It seems to be related to the display driver. Can you have a look to the forum posts again?