MichaIng/DietPi #5080: Clean Install Errors on Software - Log System (AUTO_SETUP_LOGGING_INDEX)

bsheeres, 2021-12-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 0

- Kernel version | `uname -a`
Linux WhyPi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 3 Model B (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2.4A (Anker 40W 5-Port USB Wall Charger)

- SDcard used | (EG: SanDisk ultra)
ScanDisk 16 GB Class 4
Pioneer 32 GB Class 10

#### Additional Information (if applicable)
Clean install.
First run as root.
Repeatable (on the only RPi I have). Close to a dozen installs with the same result.
Attempting to change the log system in the subsequent software dialog results in the same error. Ignoring will allow the install to complete. After the install changing the log system results in no error.

[image]

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

Fresh vanilla install of Bullseye ARMv7 32 bit or ARMv8 64 bit

#### Expected behaviour

Should not be seeing error

#### Actual behaviour

Error being thrown

#### Extra details

MichaIng, 2021-12-15:
Many thanks for reporting.

Indeed, since the selections are now applied immediately, but the install state file does not exist yet on first boot, this triggers errors. Fixed with: https://github.com/MichaIng/DietPi/commit/1231f41
I'll redo our images soon and pre-create the state file for those, until v8.0 has been released.

Joulinar, 2021-12-18:
@MichaIng 
did you already recreated the images with the pre-created state file?

closed 2021-12-18 by MichaIng

Joulinar, 2021-12-21:
@MichaIng 
I tested the image `DietPi_RPi-ARMv6-Bullseye` created on 17th and still got the error with the missing `/boot/dietpi/.installed`

MichaIng, 2021-12-21:
You are right, it's the 32-bit RPi images since my RPi was on the wrong Git branch `master` 🙈. Will repack them.